espressif/esp-idf #9888: Prehashed wifi psk has wifi connection but no data connection

ahmed8812, 2022-09-30:
Hi,

I found out that when when using prehashed key 64 bytes length, the WiFi has connection but I was not able to establish any data. My device is connected to the WiFi, but my mqtt connection was not able to retrieve any data from my server. I tried ping to my device through mqtt also failed

I used this PSK you can try it out
6987bfa06805c1c53884577b3cba951701b25bbe47ff1dbf890fd20b9bb193d7

Thanks

AxelLin, 2022-10-01:
Which esp-idf version ( git describe --tags ) and which SoC/module are you using?

ahmed8812, 2022-10-01:
Hi

V4.3.2
Esp32-s2-wrover

Thanks

nishanth-radja, 2022-10-06:
Hi @ahmed8812 ,I was able to ping after connection using the psk provided.
Can you pls mention the commit ID you are using?
Also is the data successful if you use a smaller PSK or passphrase?
Can we have the sniffer capture when the issue occurs with the SSID and psk used for the connection( Pls capture all the 4 way handshake so that we can decrypt the packets.)

ahmed8812, 2022-10-06:
Hi

I have not cloned the v4.3.2 from github directly, I download the file and
paste into my pc (there is certain reason for this which I will not discuss
here) , hence I cannot get the commit file

Smaller psk no problem I can connect, but with hashed key 256 bit it will
always be 64 byte right?

Wireshark is the suitable tool to sniff wifi?

AxelLin, 2022-10-06:
@ahmed8812 
I think you should try the latest release v4.3.4, or clone the latest code from git.

AxelLin, 2022-12-19:
> @ahmed8812 I think you should try the latest release v4.3.4, or clone the latest code from git.

@ahmed8812 

Any update? Do you still hit the issue with v4.3.4?
If you still hit the issue, please follow https://github.com/espressif/esp-idf/issues/9888#issuecomment-1269676057 to get log.

ahmed8812, 2022-12-19:
> If you still hit the issue, please follow #9888 (comment)

Hi

I think its due to my router. It depends on which router I used. Some
router I used I was able to use prehash while others cannot

If I found something that needs to be highlighted again if I ran these
tests again, I will be sure to let you know

Thanks

AxelLin, 2023-04-29:
> Hi I think its due to my router. It depends on which router I used. Some router I used I was able to use prehash while others cannot If I found something that needs to be highlighted again if I ran these tests again, I will be sure to let you know Thanks

Could you share the brand/model of the problematic router?

ahmed8812, 2023-04-29:
Hi

Its the Apple Airport Extreme 6th Gen

Regards

AxelLin, 2023-05-10:
Seems one of my device hit this (or similar) issue (v4.3.5-83-g33b3db35d7).
It looks like the device is successfully associated with AP, but does not get IP address.
I got WIFI_EVENT_STA_CONNECTED event but did not receive IP_EVENT_STA_GOT_IP event.

The failure log:
```
I (2084) wifi:set country: cc=US schan=1 nchan=11 policy=0

I (2091) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (2183) wifi:mode : sta (a8:03:2a:e6:c4:f4)
I (2184) wifi:enable tsf
I (4598) wifi:new:<9,2>, old:<1,0>, ap:<255,255>, sta:<9,2>, prof:1
I (5089) wifi:state: init -> auth (b0)
I (5094) wifi:state: auth -> assoc (0)
I (5112) wifi:state: assoc -> run (10)
I (8219) wifi:state: run -> init (2c0)
I (8220) wifi:new:<9,0>, old:<9,2>, ap:<255,255>, sta:<9,2>, prof:1
I (8221) wifi:new:<9,2>, old:<9,0>, ap:<255,255>, sta:<9,2>, prof:1
I (8224) wifi:state: init -> auth (b0)
I (8230) wifi:state: auth -> assoc (0)
I (8237) wifi:state: assoc -> run (10)
I (8269) wifi:connected with IGS-MESH, aid = 3, channel 9, 40D, bssid = ac:84:c6:32:1a:26
I (8270) wifi:security: WPA2-PSK, phy: bgn, rssi: -46
I (8271) wifi:pm start, type: 1

I (8345) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (23649) wifi:<ba-add>idx:0 (ifx:0, ac:84:c6:32:1a:26), tid:0, ssn:11, winSize:64
```

ps.
All other devices are working fine with the same AP, so I think it's not AP issue in this case.
However, when the device is running as AP, it's working fine and my PC can connect to this device. seems not h/w issue.
Other working device will show esp_netif_handlers: sta ip message as below.
```
I (15535) wifi:<ba-add>idx:0 (ifx:0, ac:84:c6:32:1a:26), tid:6, ssn:2, winSize:64
I (16461) esp_netif_handlers: sta ip: 192.168.1.133, mask: 255.255.255.0, gw: 192.168.1.1
```

Any hint to trace this issue?

nishanth-radja, 2024-03-01:
@AxelLin Are you still facing this issue? If yes, we need sniffer trace to check the DHCP handshake.Pls provide the SSID and password of the connection to decode the DHCP packets.

AxelLin, 2024-03-05:
@nishanth-radja 
I still hit the issue on this device even with master tree.
I'm not allow to share wireshark log in this environment.
Attached the debug build log with DHCP_DEBUG, please check if this helps:

[dhcp_debug.TXT](https://github.com/espressif/esp-idf/files/14492835/dhcp_debug.TXT)

vik-gokhale, 2024-04-23:
> Hi I think its due to my router. It depends on which router I used. Some router I used I was able to use prehash while others cannot If I found something that needs to be highlighted again if I ran these tests again, I will be sure to let you know Thanks
> […](#)

With this observation from the reporter, this issue needs to be closed.