espressif/esp-idf #9889: ESP32S3 usb_dongle  config ble-only  [bluetooth]:advertise on  Failed to register advertisement: org.bluez.Error.Failed

yuandonglp, 2022-09-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

[bluetooth]# menu advertise
[bluetooth]# clear
[bluetooth]# uuids 0x180D
[bluetooth]# appearance 833
[bluetooth]# name esp32s3
[bluetooth]# discoverable on
[bluetooth]# back
[bluetooth]:advertise on  
Failed to register advertisement: org.bluez.Error.Failed

Weijian-Espressif, 2022-10-08:
@yuandonglp  Are you using bluez host? I'm not sure what causes the bluez error, if you have bluedroid host, nimble host or controller issues, please report to us.

yuandonglp, 2022-10-08:
> @yuandonglp Are you using bluez host? I'm not sure what causes the bluez error, if you have bluedroid host, nimble host or controller issues, please report to us.

我的配置如下
esp-iot-solution/examples/usb/device/usb_dongle 
Component config->TinyUSB Stack配置中关闭了cdc串口配置，打开了Bluetooth Host Class配置，打开了ECM配置然后  Component config->Bluetooth中选择Bluetooth controller中的Bluetooth Host（Bluedroid - Dual-mode）模式，其他为默认值
然后编译程序烧写至ESP32S3中，在linux主机上hciconfig 可以看到蓝牙网卡设备，使用linux中的bluetoothctl工具来对esp32s3蓝牙进行操作，在输入advertise on打开esp32s3的蓝牙广播时报了Failed to register advertisement: org.bluez.Error.Failed 这个错误，同时也无法发现esp32s3这个蓝牙设备。

Weijian-Espressif, 2022-10-08:
我们官方提供两个 host, bluedroid 和 nimble, 使用他们遇到的问题我可以解答, 看你的描述可能是 bluez host 出现错误, 我对  bluez 也不了解. 你是从哪获取到的代码, 建议你找对应的官方渠道需求帮助.

yuandonglp, 2022-10-08:
> bluez

使用的是esp-iot-solution下的add_usb_solutions分支下的代码，这个也是官方提供的吧。bluez是蓝牙通信的协议栈，这个add_usb_solutions分支应该是支持的，因为我通过linux的bluez 可以和esp32s3进行通信，可以开启它的蓝牙搜索功能，也可以控制蓝牙设备的开启和关闭，就是无法开启esp32s3的广播功能

Weijian-Espressif, 2022-10-08:
esp-iot-solution 是其他同事开发的代码, 为了更快的解决问题, 建议您在 esp-iot-solution 路径下的 issues 提问.

yuandonglp, 2022-10-08:
好的 十分感谢

closed 2022-10-08 by yuandonglp