espressif/esp-idf #9890: lightsleep causing internal RTC to drift

kiss81, 2022-09-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### Development Kit.

Custom board (esp 32 pico d4)

### Power Supply used.

Battery

### What is the expected behavior?

I try to keep multiple esp32 boards in sync (milliseconds accuracy)
I do have an external clock (rv3028) connected to the 32k xn pin. This should be ~5ppm.
When not using light sleep or auto sleep there is no time drift when asking gettimeofday() 
I do use a led blink to indicate when the milliseconds flip from 999 to 0.

### What is the actual behavior?

When I use light sleeps (of ~950ms) instead of a vtaskdelay of 950ms) in my main code the rtc slightly drifts behind. In 4 hours this is ~200ms.

### Steps to reproduce.

    while(true)
    {
        esp_pm_lock_acquire(pmlock_handle);
        gettimeofday(&tv, NULL);
        msLeftForNextSecond = 1000 - ((tv.tv_usec / 1000) % 1000);
        //ESP_LOGI(tag, "%d", msLeftForNextSecond);
        //vTaskDelay(msLeftForNextSecond / portTICK_RATE_MS); // this works ok
        board->powerMgr.LightSleep(msLeftForNextSecond, true);
        gpio_set_level(GREENLED, 1); // enable led
         vTaskDelay(40 / portTICK_RATE_MS);
        gpio_set_level(GREENLED, 0); // disable led
        esp_pm_lock_release(pmlock_handle);
 }

### More Information.

There are lot's of settings I tried in the menuconfig that are RTC / light sleep related. I tried lot's of them, but without success so far. Will post my sdkconfig.

kiss81, 2022-09-30:
https://www.file.io/bTAM/download/ws4h3mv8QEaf

AxelLin, 2022-12-17:
> sdkconfig https://www.file.io/bTAM/download/ws4h3mv8QEaf

Above link is invalid, cannot get the sdkconfig. (There was an error retrieving your file.)

AxelLin, 2022-12-17:
Maybe this is related to https://github.com/espressif/esp-idf/issues/6687#issuecomment-1316769420

@ginkgm @KonstantinKondrashov

kiss81, 2022-12-17:
Update from my side: 
currently running v4.4.3 and I am not sure if this specific issue is still there. I didn't take into account that the rv3028 clk_out is not calibrated like it's internal time keeping. 
I have two boards running without issues now, but will test a few more the next week.
One thing to notice: some have issues when using a 1nF to GND on the 32k_xp pin. Reducing this to 220pF seems to work though. 
Will update the status on this the next week.

KonstantinKondrashov, 2023-03-14:
Hi @kiss81!
I close this issue once you do not see it anymore for v4.4.3.
Please feel free to reopen it.
Thanks.

closed 2023-03-14 by KonstantinKondrashov