espressif/esp-idf #9891: esp_bt_dev_set_device_name() not working

mbastida123, 2022-09-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

origin/release/v4.4.2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-WROOM-devkit

### Power Supply used.

USB

### What is the expected behavior?

esp_bt_dev_set_device_name() should make the ESP32 display the name.

### What is the actual behavior?

Using NRFConnect I can see the device advertising my Beacon but the name is shown as NA (Not available).

### Steps to reproduce.

```
    esp_err_t status;

	ESP_ERROR_CHECK(esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT));
    esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
    esp_bt_controller_init(&bt_cfg);
    esp_bt_controller_enable(ESP_BT_MODE_BLE);
    
    esp_bluedroid_init();
    
    
    esp_bluedroid_enable();
    
    esp_err_t err=esp_bt_dev_set_device_name("ho");
    ESP_LOGI("main", "Set name result: %s", esp_err_to_name(err));
```

I have also tried to:

```
static esp_ble_adv_data_t esp_ble_adv_data={
    .include_name=true
};
```
And then:

```
    err= esp_ble_gap_config_adv_data(&esp_ble_adv_data);
    ESP_LOGI("main", "config adv data result: %s", esp_err_to_name(err));
```
With no success.

### Debug Logs.

```plain
I (558) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (617) BTDM_INIT: BT controller compile version [5688ed5]
I (617) system_api: Base MAC address is not set
I (617) system_api: read default base MAC address from EFUSE
I (627) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (1047) main: Set name result: ESP_OK
I (1047) main: config adv data result: ESP_OK
I (1047) AltBeacon_HAL: register callback
I (61047) AltBeacon_HAL: Stop adv successfully
```

### More Information.

Everything compiles and the log doesn't show any error. 

Could it be related to this?: https://github.com/espressif/esp-idf/issues/8071

pd-esp, 2022-10-11:
Could you please try to using this **esp_ble_gap_set_device_name("ho")** instead of **esp_bt_dev_set_device_name("ho")**

mbastida123, 2022-10-11:
> Could you please try to using this **esp_ble_gap_set_device_name("ho")** instead of **esp_bt_dev_set_device_name("ho")**

Same behaviour, the code compiles and the LOG shows everything is OK but I don't see the name on the device.

```
I (584) BTDM_INIT: BT controller compile version [5688ed5]
I (584) system_api: Base MAC address is not set
I (594) system_api: read default base MAC address from EFUSE
I (594) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (1014) main: Set name result: ESP_OK
I (1014) main: config adv data result: ESP_OK
I (1024) AltBeacon_HAL: register callback
```

mbastida123, 2023-02-14:
Hi, any news on this issue?

I'm on v4.4.4 and still facing the same issue.

esp-cjh, 2023-03-24:
@mbastida123 
Yes. If you use `esp_ble_gap_config_adv_data_raw()`, you don't need to set device name. You should fill advertising and scan response data. Such as `uint8_t scan_rsp_data[] = {0x04, 0x09, 0x41, 0x42, 0x43};`, this means device name is "ABC". For detail you should refer to Bluetooth Core Spec.

Alvin1Zhang, 2023-05-24:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more updates. Thanks.

closed 2023-05-24 by Alvin1Zhang

mbastida123, 2023-06-22:
Hi @Alvin1Zhang and my apologies for not responding.

At the moment (without advertising the name) I have this:
```
typedef struct {
    uint8_t flags[3];
    uint8_t length;
    uint8_t type;
    uint16_t company_id;
    uint16_t beacon_type;
    uint8_t beacon_id[ALTBEACON_USEFUL_DATA_LEN];
    int8_t reference_rssi;
    uint8_t mfg_reserved;
}__attribute__((packed)) esp_ble_altbeacon_t;
```
That is populated as so:
```
static esp_ble_altbeacon_t altbeacon_advertising_data = {
    .flags = {0x02, 0x01, 0x06},
    .length = 0x1B,
    .type = 0xFF,
    .company_id = 0x5900,
    .beacon_type = 0xACBE,
    .beacon_id = {0xFD, 0xA5, 0x06, 0x93, 0xA4, 0xE2, 0x4F, 0xB1, 0xAF, 0xCF, 0xC6, 0xEB, 0x11, 0x11, 0x11, 0x11, 0x00, 0x00, 0x00, 0x00},
    .reference_rssi = (int8_t) 0xC3,
    .mfg_reserved = 0x33
};
```

This is converted into a simple uint8_t array and is passed to `esp_ble_gap_config_adv_data_raw()`
This works great, but now I want to add the name.

So I modify the structure to:
```
typedef struct {
    uint8_t flags[3];
    uint8_t name[6];
    uint8_t length;
    uint8_t type;
    uint16_t company_id;
    uint16_t beacon_type;
    uint8_t beacon_id[ALTBEACON_USEFUL_DATA_LEN];
    int8_t reference_rssi;
    uint8_t mfg_reserved;
}__attribute__((packed)) esp_ble_altbeacon_t;
```

and populate it:
```
static esp_ble_altbeacon_t altbeacon_advertising_data = {
    .flags = {0x02, 0x01, 0x06},
    .name = {0x04, 0x09, 'T', 'E', 'S', 'T'},
    .length = 0x1B,
    .type = 0xFF,
    .company_id = 0x5900,
    .beacon_type = 0xACBE,
    .beacon_id = {0xFD, 0xA5, 0x06, 0x93, 0xA4, 0xE2, 0x4F, 0xB1, 0xAF, 0xCF, 0xC6, 0xEB, 0x11, 0x11, 0x11, 0x11, 0x00, 0x00, 0x00, 0x00},
    .reference_rssi = (int8_t) 0xC3,
    .mfg_reserved = 0x33
};
```

I added the 0x04 and 0x09 as per your example.

However, this doesn't work. I see the BLE beacon but it seems to show a default beacon (with name ESP32 and no advertising data). As if there was an error. However, according to the ESP32 everything is correct. 

What am I doing wrong?

esp-cjh, 2023-06-25:
Hi, @mbastida123 
You should construct your data according to the advertising and scan response data format.
You can refer to Core Spec Version 5.3 Vol 3, Part C, 11

mbastida123, 2023-06-26:
> Hi, @mbastida123 You should construct your data according to the advertising and scan response data format. You can refer to Core Spec Version 5.3 Vol 3, Part C, 11

@esp-cjh thanks for the indications.

However, It's still not working. 

The ADV data is as follows:
```
static esp_ble_altbeacon_t altbeacon_advertising_data = {
    .name = {0x05, 0x09, 'T', 'E', 'S', 'T'}, 
    .flags = {0x02, 0x01, 0x06},
    .length = 0x1B,
    .type = 0xFF,
    .company_id = 0x5900,
    .beacon_type = 0xACBE,
    .beacon_id = {0x43, 0x49, 0x54, 0x00, 0x00, 0x08, 0x33, 0x06, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x4b, 0x10, 0x00, 0x4f, 0xcb, 0xee},
    .reference_rssi = (int8_t) 0xC3,
    .mfg_reserved = 0x33,
};
```
Applying what the bluetooth core spec states:
1. Length 0x05 containing type (0x09 meaning complete name) and the name "TEST"
2. Length 0x02 containing type (0x01 flags) and 0x06
3. Length 0x1B containing type (0xFF) and the AltBeacon data

By removing the point 1 (name related data) the beacon is displayed correctly. This means that point 2 and 3 are correct. However, the moment I add point 1 the ESP32 defaults to a generic beacon.

This is the raw data I'm send to `esp_ble_gap_config_adv_data_raw()`

> 05 09 54 45 53 54 02 01 06 1b ff 00 59 be ac 43
> 49 54 00 00 08 33 06 00 00 00 00 00 00 4b 10 00
> 4f cb ee c3 33 

I have checked multiple times and this raw data matches perfectly the data I'm setting on the struct.

zhp0406, 2023-06-26:
@mbastida123 The maximum length for BLE advertisement data is 31 bytes. However, the data you provided exceeds this limit with 37 bytes, so setting the advertisement data would not be successful.

Recommendation:
Move the manufacturer-specific data to the scan response data instead.

adv data : 050954455354020106
scan rsp data:1bff0059beac43495400000833060000000000004b10004fcbeec333

Using the method described above, I successfully tested the configuration.

mbastida123, 2023-06-26:
> @mbastida123 The maximum length for BLE advertisement data is 31 bytes. However, the data you provided exceeds this limit with 37 bytes, so setting the advertisement data would not be successful. [...]

Following your advice I have tried:
```
    uint8_t adv[]={0x1b, 0xff, 0x00, 0x59, 0xbe, 0xac, 0x43, 0x49, 0x54, 0x00, 0x00, 0x08, 0x33, 0x06, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x4b, 0x10, 0x00, 0x4f, 0xcb, 0xee, 0xc3, 0x33};
    ESP_ERROR_CHECK(esp_ble_gap_config_scan_rsp_data_raw(adv, 28));

    uint8_t adv2[6]={0x05, 0x09, 'T', 'E', 'S', 'T'};
    ESP_ERROR_CHECK(esp_ble_gap_config_adv_data_raw(adv2, 6));
```

Now I can see the name but not the altbeacon data.

If I invert the adv data and rsp data.
```
    uint8_t adv[]={0x1b, 0xff, 0x00, 0x59, 0xbe, 0xac, 0x43, 0x49, 0x54, 0x00, 0x00, 0x08, 0x33, 0x06, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x4b, 0x10, 0x00, 0x4f, 0xcb, 0xee, 0xc3, 0x33};
    ESP_ERROR_CHECK(esp_ble_gap_config_adv_data_raw(adv, 28));

    uint8_t adv2[6]={0x05, 0x09, 'T', 'E', 'S', 'T'};
    ESP_ERROR_CHECK(esp_ble_gap_config_scan_rsp_data_raw(adv2, 6));
```
I can see the altbeacon data but not the name.

zhp0406, 2023-06-26:
@mbastida123 
---

I have tested the code below and it works fine. You can give it a try as well.

```
static uint8_t raw_adv_data[] = {0x1b, 0xff, 0x00, 0x59, 0xbe, 0xac, 0x43, 0x49, 0x54, 0x00, 0x00, 0x08, 0x33, 0x06, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x4b, 0x10, 0x00, 0x4f, 0xcb, 0xee, 0xc3, 0x33};
static uint8_t raw_scan_rsp_data[] = {0x05, 0x09, 'T', 'E', 'S', 'T'};
```

---

```
static uint8_t raw_scan_rsp_data[] = {0x1b, 0xff, 0x00, 0x59, 0xbe, 0xac, 0x43, 0x49, 0x54, 0x00, 0x00, 0x08, 0x33, 0x06, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x4b, 0x10, 0x00, 0x4f, 0xcb, 0xee, 0xc3, 0x33};
static uint8_t raw_adv_data[] = {0x05, 0x09, 'T', 'E', 'S', 'T'};
```

BTW,

```
esp_err_t raw_scan_ret = esp_ble_gap_config_scan_rsp_data_raw(raw_scan_rsp_data, sizeof(raw_scan_rsp_data));
```

```
esp_err_t raw_adv_ret = esp_ble_gap_config_adv_data_raw(raw_adv_data, sizeof(raw_adv_data));
```

mbastida123, 2023-06-26:
> ## @mbastida123
> I have tested the code below and it works fine. You can give it a try as well.
>
> ```
> static uint8_t raw_adv_data[] = {0x1b, 0xff, 0x00, 0x59, 0xbe, 0xac, 0x43, 0x49, 0x54, 0x00, 0x00, [...]

Are you testing on IDF v4.4.4?

Because with this code:
```
    static uint8_t raw_scan_rsp_data[] = {0x1b, 0xff, 0x00, 0x59, 0xbe, 0xac, 0x43, 0x49, 0x54, 0x00, 0x00, 0x08, 0x33, 0x06, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x4b, 0x10, 0x00, 0x4f, 0xcb, 0xee, 0xc3, 0x33};
    static uint8_t raw_adv_data[] = {0x05, 0x09, 'T', 'E', 'S', 'T'};

    esp_err_t raw_scan_ret = esp_ble_gap_config_scan_rsp_data_raw(raw_scan_rsp_data, sizeof(raw_scan_rsp_data));
    esp_err_t raw_adv_ret = esp_ble_gap_config_adv_data_raw(raw_adv_data, sizeof(raw_adv_data));

    ESP_LOGI(TAG, "esp_ble_gap_config_scan_rsp_data_raw res: %s", esp_err_to_name(raw_scan_ret));
    ESP_LOGI(TAG, "esp_ble_gap_config_adv_data_raw res: %s", esp_err_to_name(raw_adv_ret));

```
I see the name but I don't see the altbeacon data.

And no error is returned:

> I (14:56:03.155) AltBeacon_HAL: esp_ble_gap_config_scan_rsp_data_raw res: ESP_OK
> I (14:56:03.156) AltBeacon_HAL: esp_ble_gap_config_adv_data_raw res: ESP_OK

mbastida123, 2023-06-26:
> @mbastida123
> 
> Try the following methods:
> 
> 1-Set adv_filter_policy to ADV_FILTER_ALLOW_SCAN_ANY_CON_ANY:
> 
> ```
> static esp_ble_adv_params_t adv_params = {
>     .adv_int_min         = 0x20,
>     .adv_int_max         = 0x40,
>     .adv_type            = ADV_TYPE_IND,
>     .own_addr_type       = BLE_ADDR_TYPE_PUBLIC,
>     .channel_map         = ADV_CHNL_ALL,
>     .adv_filter_policy   = ADV_FILTER_ALLOW_SCAN_ANY_CON_ANY,
> };
> ```
> 
> 2-Try scanning with a different device.
> 
> 3-Try using a different IDF (IoT Development Framework) version.

Found it, 

I had:
`    .adv_type           = ADV_TYPE_NONCONN_IND,`

But you had: 
`    .adv_type           = ADV_TYPE_IND,`

The problem is that now the device shows as connectable (meaning that in nrfConnect there's a Connect button). Which is not desired. Is there any alternative?

zhp0406, 2023-06-26:
@mbastida123 
```
static esp_ble_adv_params_t adv_params = {
    .adv_int_min         = 0x20,
    .adv_int_max         = 0x40,
    .adv_type            = ADV_TYPE_SCAN_IND,
    .own_addr_type       = BLE_ADDR_TYPE_PUBLIC,
    .channel_map         = ADV_CHNL_ALL,
    .adv_filter_policy   = ADV_FILTER_ALLOW_SCAN_ANY_CON_ANY,
};
```

jansonkai12, 2023-06-27:
1

zhp0406, 2023-06-27:
[image]

zhp0406, 2023-06-27:
[image]