espressif/esp-idf #9895: TinyUSB header not found

paucazou, 2022-10-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-1006-g02605f1a31

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### What is the expected behavior?

I expected to build the examples about usb without problems.

### What is the actual behavior?

Actually, tinyusb.h is not found with every example I tried.
`/save/scripts/esp-idf/examples/peripherals/usb/device/tusb_serial_device/main/tusb_serial_device_main.c:11:10: fatal error: tinyusb.h: No such file or directory
   11 | #include "tinyusb.h"
      |          ^~~~~~~~~~~
compilation terminated.
`

### Steps to reproduce.

I just entered
`idf.py build`

### Build or installation Logs.

```plain
Executing action: all (aliases: build)
Running make in directory /save/scripts/esp-idf/examples/peripherals/usb/device/tusb_serial_device/build
Executing "make -j 6 all"...
[  0%] Built target _project_elf_src
[  0%] Built target memory_ld
[  0%] Built target partition_table_bin
[  0%] Built target custom_bundle
Consolidate compiler generated dependencies of target __idf_ulp
[  0%] Performing build step for 'bootloader'
[  0%] Built target __idf_ulp
Consolidate compiler generated dependencies of target __idf_esp_serial_slave_link
[  0%] Built target __idf_esp_serial_slave_link
Consolidate compiler generated dependencies of target __idf_log
Consolidate compiler generated dependencies of target __idf_sdmmc
[  1%] Built target _project_elf_src
[  5%] Built target __idf_log
[  1%] Built target __idf_sdmmc
Consolidate compiler generated dependencies of target __idf_esp_rom
Consolidate compiler generated dependencies of target __idf_esp_https_ota
[ 12%] Built target __idf_esp_rom
[  1%] Built target __idf_esp_https_ota
Consolidate compiler generated dependencies of target __idf_esp_common
[ 14%] Built target __idf_esp_common
Consolidate compiler generated dependencies of target __idf_esp_http_server
Consolidate compiler generated dependencies of target __idf_esp_hw_support
[  2%] Built target __idf_esp_http_server
[ 24%] Built target __idf_esp_hw_support
Consolidate compiler generated dependencies of target __idf_esp_http_client
Consolidate compiler generated dependencies of target __idf_esp_system
[  3%] Built target __idf_esp_http_client
[ 26%] Built target __idf_esp_system
[... 127 lines cut ...]
Consolidate compiler generated dependencies of target __idf_cmock
Consolidate compiler generated dependencies of target __idf_protocomm
[ 95%] Built target __idf_spiffs
[ 95%] Built target __idf_cmock
Consolidate compiler generated dependencies of target __idf_fatfs
[ 97%] Built target __idf_protocomm
[ 98%] Built target __idf_fatfs
Consolidate compiler generated dependencies of target __idf_esp_local_ctrl
Consolidate compiler generated dependencies of target __idf_wifi_provisioning
[ 98%] Built target __idf_esp_local_ctrl
[ 99%] Built target __idf_wifi_provisioning
[ 99%] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/tusb_serial_device_main.c.obj
/save/scripts/esp-idf/examples/peripherals/usb/device/tusb_serial_device/main/tusb_serial_device_main.c:11:10: fatal error: tinyusb.h: No such file or directory
   11 | #include "tinyusb.h"
      |          ^~~~~~~~~~~
compilation terminated.
make[2]: *** [esp-idf/main/CMakeFiles/__idf_main.dir/build.make:76: esp-idf/main/CMakeFiles/__idf_main.dir/tusb_serial_device_main.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:4336: esp-idf/main/CMakeFiles/__idf_main.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
make failed with exit code 2, output of the command is in the /save/scripts/esp-idf/examples/peripherals/usb/device/tusb_serial_device/build/log/idf_py_stderr_output_110020 and /save/scripts/esp-idf/examples/peripherals/usb/device/tusb_serial_device/build/log/idf_py_stdout_output_110020
```

### More Information.

Maybe it's unrelated, but in the serial usb example, it is said that it is possible to access the TinyUSB stack with menuconfig, however I couldn't.

chegewara, 2022-10-02:
What is the board/target?

paucazou, 2022-10-02:
The target is Esp32-S2.

nopnop2002, 2022-10-03:
Occurs even if target is esp32.
```
$ cd esp-idf/examples/peripherals/usb/device/tusb_serial_device
$ idf.py build
../main/tusb_serial_device_main.c:11:10: fatal error: tinyusb.h: No such file or directory
   11 | #include "tinyusb.h"
```

radhikapatel15, 2022-12-02:
hello can I use ESP32-WROVER-E as my target device? I f yes then what configurations to be done in sdkconfig file

nopnop2002, 2022-12-02:
@radhikapatel15 

Usb can only use esp32s2/s3.