espressif/esp-idf #9896: WiFi disconnecting every 3 hours

se-k32, 2022-10-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

One of our clients has a problem that all ESP32 devices get disconnected precisely 3 hours after being connected to AP. The disconnect reason is WIFI_REASON_AUTH_EXPIRE (2) or WIFI_REASON_NOT_AUTHED (6). After that, devices reconnect to AP and the same thing happens again after 180 minutes. We think that the AP requests some kind of re-authentication, but ESP fails to respond to it, so AP kicks him off. Sadly we haven't been able to reproduce this issue locally just yet. Also, detailed logs are not available, because devices are stationed remotely, UDP logs are not helpful, because they get cut off when this happens. For now, we are saving the time and disconnecting reason and we figured out that this is happening every 180 minutes after connecting to AP, and the reasons are always WIFI_REASON_AUTH_EXPIRE or WIFI_REASON_NOT_AUTHED. We will try to reproduce this issue locally to get detailed logs.

Module used: ESP32 and ESP32C3 (same issue on both)
idf version: v4.4.2-138-g4c550fd79f
AP Brand: Ruckus
Authentication method: WPA2-PSK

We tried the patch from maybe a similar issue (https://github.com/espressif/esp-idf/issues/9303#issuecomment-1263950370), but sadly it did not help.

nishanth-radja, 2022-10-06:
@se-k32 can you check the group-rekey time of the Ruckus AP, Also if possible can you start the sniffer capture before the issue and see if we can capture the pkts during the issue.Also can we have the wifi config of the AP.

se-k32, 2022-10-07:
@nishanth-radja the issue in the facility seems to be resolved now. We have asked the network administrator to check again for any possible configuration on AP that could cause it and we also added this line to WiFi init code `esp_wifi_set_inactive_time(WIFI_IF_STA, 10);`. After that, there are no more disconnects. As this was an issue in production, we won't be testing there anymore, but we will try to get a similar AP to reproduce the problem. I will reopen this issue in case there won't be fix from [https://github.com/espressif/esp-idf/issues/9303] already. Thank you.

closed 2022-10-07 by se-k32