espressif/esp-idf #9897: How to increase espnow encrypted peer list more than 6

rmshub, 2022-10-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

By default maximum number of encrypted peer can be added is 6. How to increase this number and maximum possible encrypted devices

rmshub, 2022-10-05:
Any update on this request ?

rmshub, 2022-10-07:
Any info on this request ?

nachiketkukade, 2022-10-07:
@rmshub , espnow uses hardware encryption and since Station / SoftAP also use hardware encryption, sometimes even simultaneously, increasing this limit will affect other interfaces. Therefore there is no plan to increase this limit as of now. Thanks.

rmshub, 2022-10-08:
could you please update same information in documentation? In documentation mentioned that `The maximum number of paired devices is 20, and the paired encryption devices are no more than 16, the default is 6.`

nachiketkukade, 2022-10-10:
@rmshub , yes, we will make sure the documentation is updated to match the configurations with the new change. Thanks for bringing this to our attention!

Alvin1Zhang, 2023-01-12:
Thanks for reporting, fix is available at https://github.com/espressif/esp-idf/commit/d89a51299d54b3a8793872051a95d230e6818ee1, feel free to reopen.

closed 2023-01-12 by Alvin1Zhang