espressif/esp-idf #9898: ESP32 C3 issues with setting BLE Server security param.

Seref, 2022-10-02:
Hi there.

I got an ESP32 C3 and when I set up a simple BLE Server and connect my phone (Android 13 device that definitly supports BLE) the C3 spits out following errors on connection:

The security option is set with:
`esp_ble_gap_set_security_param(ESP_BLE_SM_AUTHEN_REQ_MODE, &m_authReq, sizeof(uint8_t));		  // <--- setup requested authentication mode``

using the param: ESP_LE_AUTH_REQ_SC_MITM_BOND

```
16:20:07.561 -> E (73247) BT_SMP: smp_calculate_link_key_from_long_term_key failed to update link_key. Sec Mode = 2, sm4 = 0x00
16:20:07.561 -> E (73248) BT_SMP: smp_derive_link_key_from_long_term_key failed
16:20:07.561 -> 
16:20:07.561 -> E (73256) BT_BTM: btm_proc_smp_cback received for unknown device
16:20:07.761 -> E (73453) BT_BTM: BTM_GetSecurityFlags false
16:20:07.761 -> 
16:20:07.841 -> E (73543) BT_BTM: BTM_GetSecurityFlags false
16:20:07.841 -> 
16:20:08.083 -> E (73783) BT_BTM: BTM_GetSecurityFlags false
```

When I just use ESP_LE_AUTH_BOND instead it works fine.

In addition both ESP_LE_AUTH_BOND & ESP_LE_AUTH_REQ_SC_MITM_BOND work fine on an old ESP32 (not C3)

Weijian-Espressif, 2022-10-08:
@Seref Can you test our example, /examples/bluetooth/bluedroid/ble/gatt_security_server, it works fine with my phone.Please clear the bond list on the mobile phone before testing.

Seref, 2022-10-08:
Okay that's weird, gatt_security_server seems to work and the auth mode used is ```ESP_LE_AUTH_REQ_SC_BOND``` (not MITM). It seems like the issue is isolate to the arduino-esp32-

closed 2022-10-08 by Seref