espressif/esp-idf #9902: Missing documentation of WiFi warnings / typos in warning messages / change the format and time stamp of WiFi warnings

kriegste, 2022-10-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I use IDF 4.4.2 and I get warnings like these:

> W (708) wifi:<ba-add>idx:0 (ifx:0, aa:bb:cc:dd:ee:ff), tid:0, ssn:1, winSize:64
> W (1825827) wifi:<ba-del>idx
> W (109675) wifi:sta_scan: STA is connecting, scan are not allowed!
> W (8827685) wifi:age timer already deinit
> W (1825837) wifi:hmac tx: ifx0 stop, discard

Now I wonder what they mean and if I need to worry. There is no documentation.

Also, is there in the meantime a way to have the WiFi lib display date and time in those messages? Like the rest of the IDF does. Mixing the two formats looks very strange.

BTW, my customers keep telling me I need to fix the "bad grammar" in those lines and there also seems to be a space missing between the first colon after the tag and the actual message. ;)

zhangyanjiaoesp, 2022-10-19:
> W (708) wifi:<ba-add>idx:0 (ifx:0, aa:bb:cc:dd:ee:ff), tid:0, ssn:1, winSize:64
W (1825827) wifi:<ba-del>idx

These logs do not indicate any issue. It is related to the Wi-Fi block acknowledgment mechanism. `ba-add` means the ESP32 received an add block acknowledgment request frame from the router. `ba-del` means the ESP32 received a delete block acknowledgment request frame from the router. 
Will change the level to info to avoid misunderstanding.

> W (109675) wifi:sta_scan: STA is connecting, scan are not allowed!

This log is to tell the user do not scan when connecting to an AP.

> W (8827685) wifi:age timer already deinit
W (1825837) wifi:hmac tx: ifx0 stop, discard
These logs don't need to worry. Will change the level to debug to avoid misunderstanding.

Alvin1Zhang, 2023-01-12:
Thanks for reporting and sharing the updates, sorry for slow turnaround, fix on master branch is available at https://github.com/espressif/esp-idf/commit/e6c57c0d14ab134be77fc5f7f477395b81cdeacb and on release/4.4 is https://github.com/espressif/esp-idf/commit/1c0d0e505d9fc50045bce4625a8c95aaf55af362. Feel free to reopen.

closed 2023-01-12 by Alvin1Zhang

kriegste, 2023-01-28:
I think the error

> E (6345864) wifi:Set status to INIT

needs better explanation, too. It seems to occur when e.g. the WiFi protocols are limited to 11B and the router does not support it. Am I right? Anyways, there needs to be some kind of documentation.

zhangyanjiaoesp, 2023-01-29:
@kriegste The `E (6345864) wifi:Set status to INIT` log will be printed when the association fail.

kriegste, 2023-01-29:
Then why not display "association failed"? For the past days I spent too much time to find out how and where to "Set status INIT"...

zhangyanjiaoesp, 2023-01-29:
This info shows the status will change from assoc to init.

kriegste, 2023-01-29:
I eventually found out what the problem was: I set protocol to 11B-only and my router does not support it.
Maybe you can change the message to "Association failed. Setting status to init." or just "Association failed". It would be less misleading.