espressif/esp-idf #9907: ESP console command help text with newline does not print as desired

ph2355, 2022-10-04:
I registered some commands with esp_console_cmd_register. 
I want to print the help text with some newlines.
When I invoke the help command, the line wrapping in the help text of each command seems to not work correctly.

Example:
This help text is set to this in code:
`"Reset to Factory Settings\r\nThis resets the accessory to factory settings and reboots it."`

Output I get:
  Reset to Factory Settings
This resets the accessory to factory settings and
  reboots it. All pairing information, network information and accessory speci

ph2355, 2022-10-04:
Will do, thanks.

tomghuang, 2022-10-04:
Hi @ph2355 ,

If you want to split the long lines in your own way, you should use the `arg_rem` function. For example, in the `ls.c` example, you can see the following code snippet:

```c
timesty = arg_str0(NULL, "time-style","STYLE", "show times using style STYLE:"),
          arg_rem (NULL,                       "  full-iso, long-iso, iso, locale, +FORMAT"),
          arg_rem (NULL,                       "FORMAT is interpreted like `date'; if FORMAT is"),
          arg_rem (NULL,                       "FORMAT1<newline>FORMAT2, FORMAT1 applies to"),
          arg_rem (NULL,                       "non-recent files and FORMAT2 to recent files;"),
          arg_rem (NULL,                       "if STYLE is prefixed with `posix-', STYLE"),
          arg_rem (NULL,                       "takes effect only outside the POSIX locale"),
```

When you run `./ls --help`, you will see the following output:

```
  --time-style=STYLE        show times using style STYLE:
                              full-iso, long-iso, iso, locale, +FORMAT
                            FORMAT is interpreted like `date'; if FORMAT is
                            FORMAT1<newline>FORMAT2, FORMAT1 applies to
                            non-recent files and FORMAT2 to recent files;
                            if STYLE is prefixed with `posix-', STYLE
                            takes effect only outside the POSIX locale
```

ph2355, 2022-10-10:
Hi, this issue has been resolved in argtable3 through this [commit](https://github.com/argtable/argtable3/commit/7893b37ffa0dedce2ac720aab3a34f46cd363f32)

This commit is not a release, so I don't know if you integrate non-release commits.
Just so you know, that in the next release, this issue will probably be resolved.

closed 2022-10-28 by espressif-bot