espressif/esp-idf #9908: LWIP_TCPIP_CORE_LOCKING in esp_netif

ifedotov1984, 2022-10-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I dont understand core locking procedure when LWIP_TCPIP_CORE_LOCKING=1.

Function in esp_netif_lwip.c:

`
#define _RUN_IN_LWIP_TASK_IF_SUPPORTED(function, netif, param) \
{                                                              \
    if (_IS_NETIF_ANY_POINT2POINT_TYPE(netif)) {               \
        return ESP_ERR_NOT_SUPPORTED;                          \
    }                                                          \
    return esp_netif_lwip_ipc_call(function, netif, (void *)(param)); \
}

static inline esp_err_t esp_netif_lwip_ipc_call(esp_netif_api_fn fn, esp_netif_t *netif, void *data)
{
    esp_netif_api_msg_t msg = {
            .esp_netif = netif,
            .data = data,
            .api_fn = fn
    };
#if !LWIP_TCPIP_CORE_LOCKING
    if (tcpip_initialized && g_lwip_task != xTaskGetCurrentTaskHandle()) {
        ESP_LOGD(TAG, "check: remote, if=%p fn=%p\n", netif, fn);
        sys_arch_sem_wait(&api_lock_sem, 0);
        tcpip_send_msg_wait_sem((tcpip_callback_fn)esp_netif_api_cb, &msg, &api_sync_sem);
        sys_sem_signal(&api_lock_sem);
        return msg.ret;
    }
#endif /* !LWIP_TCPIP_CORE_LOCKING */
    ESP_LOGD(TAG, "check: local, if=%p fn=%p\n", netif, fn);
    return fn(&msg);
}`

When LWIP_TCPIP_CORE_LOCKING=0 then, if g_lwip_task != xTaskGetCurrentTaskHandle()  function calls tcpip_send_msg_wait_sem. And all is OK (thread safety is provided by executing the function on the LWIP thread)

When LWIP_TCPIP_CORE_LOCKING=1 - just calls  fn(&msg) without LOCK_TCPIP_CORE() and UNLOCK_TCPIP_CORE()!!!
So, _RUN_IN_LWIP_TASK_IF_SUPPORTED does not provide thread safety in this case...

david-cermak, 2022-10-21:
@ifedotov1984 if `LWIP_TCPIP_CORE_LOCKING` is enabled, than there's no need to perform any additional thread safety operations to call lwip API. The lwIP itself uses it's own locking mechanism to assure thread safety.

ifedotov1984, 2022-10-21:
> @ifedotov1984 if `LWIP_TCPIP_CORE_LOCKING` is enabled, than there's no need to perform any additional thread safety operations to call lwip API. The lwIP itself uses it's own locking mechanism to assure thread safety.

for example:

static esp_err_t esp_netif_update_default_netif(esp_netif_t *esp_netif, esp_netif_action_t action)
{
    return esp_netif_lwip_ipc_call(esp_netif_update_default_netif_lwip, esp_netif, (void*)action);
}

in _esp_netif_update_default_netif_lwip_ calls _esp_netif_set_default_netif_internal_

in
```
static void esp_netif_set_default_netif_internal(esp_netif_t *esp_netif)
{
    if (_IS_NETIF_POINT2POINT_TYPE(esp_netif, PPP_LWIP_NETIF)) {
#if CONFIG_PPP_SUPPORT
        esp_netif_ppp_set_default_netif(esp_netif->netif_handle);
#endif
    } else {
        netif_set_default(esp_netif->lwip_netif);
    }
}
```

there are calling internal lwip function _netif_set_default_

```
void
netif_set_default(struct netif *netif)
{
  LWIP_ASSERT_CORE_LOCKED();

  if (netif == NULL) {
    /* remove default route */
    mib2_remove_route_ip4(1, netif);
  } else {
    /* install default route */
    mib2_add_route_ip4(1, netif);
  }
  netif_default = netif;
  LWIP_DEBUGF(NETIF_DEBUG, ("netif: setting default interface %c%c\n",
                            netif ? netif->name[0] : '\'', netif ? netif->name[1] : '\''));
}

```
LWIP_ASSERT_CORE_LOCKED in esp idf defined as empty....

From LWIP:
/**
 * Macro/function to check whether lwIP's threading/locking
 * requirements are satisfied during current function call.
 * This macro usually calls a function that is implemented in the OS-dependent
 * sys layer and performs the following checks:
 * - Not in ISR (this should be checked for NO_SYS==1, too!)
 * - If @ref LWIP_TCPIP_CORE_LOCKING = 1: TCPIP core lock is held
 * - If @ref LWIP_TCPIP_CORE_LOCKING = 0: function is called from TCPIP thread
 * @see @ref multithreading
 */

So, _netif_set_default_ must be called in LORE LOCKED state or from lwip task, but it is called from esp_netif_update_default_netif_lwip (not from lwip task) without mutexes LOCK_TCPIP_CORE() and UNLOCK_TCPIP_CORE().

It is unsafe (it may be mutex LOCK_TCPIP_CORE()  locked from lwip task, and in same time we call function _netif_set_default_ from other task without taking mutex LOCK_TCPIP_CORE)

david-cermak, 2022-10-22:
I'm sorry, you're right -- we should define `sys_mark_tcpip_thread()` and `sys_check_core_locking()`! Thanks for explaining and reporting this problem.

I can also see that this option is broken on the current master and vfs `select()` functionality is not supported. Will fix. Thanks again for the report!

AxelLin, 2022-12-23:
@david-cermak Any update? (Since this has been in reviewing state for a Month)

ifedotov1984, 2022-12-24:
> @david-cermak Any update? (Since this has been in reviewing state for a Month)

i made some changes to work that.
But, because of DNS sources change (i change dns for support 2 dns for each netif, not global dns servers as default), file cant be just replaced.

Mainly changed functions:
esp_netif_api_cb
esp_netif_lwip_ipc_call

also there was bugs in calling locked functions from another locked functions
(for example esp_netif_update_default_netif form esp_netif_start_api) 
Also, i did not add sys_check_core_locking. Core locking in my code  just controled by myself

[esp_netif_lwip.zip](https://github.com/espressif/esp-idf/files/10298955/esp_netif_lwip.zip)

david-cermak, 2023-01-02:
> @david-cermak Any update? (Since this has been in reviewing state for a Month)

I'm sorry for such a slow turnaround, it is indeed still under review as the change involves multiple components (lwip, sntp, openthread) updating docs and adding new tests.

Thanks for providing your solution @ifedotov1984 !

I can also provide my patch before we merge it (if needed). My update also involves `sys_check_core_locking()` and thus is applicable to non-core-locking config (default) as well. This would be useful for validating thread safety in your apps and projects.

closed 2023-01-19 by espressif-bot

AxelLin, 2023-03-09:
> I'm sorry, you're right -- we should define `sys_mark_tcpip_thread()` and `sys_check_core_locking()`! Thanks for explaining and reporting this problem.
>
> I can also see that this option is broken on [...]

Is this fix available on v5.0 branch?

ifedotov1984, 2023-03-09:
> Is this fix available on v5.0 branch?

On master branch, not last release 5.0.1