espressif/esp-idf #9909: esp_wifi_restore() adds new keys to NVS

NoNullptr, 2022-10-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### Development Kit.

ESP32-S3-MINI-1-N8

### Power Supply used.

USB

### What is the expected behavior?

Calling esp_wifi_restore() is expected to restore persistent NVS settings to default values, i.e., it is expected to erase keys that were set by calling the documented functions.

### What is the actual behavior?

esp_wifi_restore() adds keys to NVS, including keys that are not set by calling documented functions (e.g. "sta.pmk", "sta.phym") and keys that seem to belong to older esp-idf versions ("auto.conn").

### Steps to reproduce.

1. Step: erase_flash
2. Step: set configuration
3. Step: call esp_wifi_restore

Compare NVS data after step 2 and 3.

nishanth-radja, 2022-10-10:
@NoNullptr can we have the application code which your are using to see this bug,Also can you mention the commit ID you are using.

mhdong, 2022-10-25:
hi @NoNullptr esp_wifi_restore() will restore persistent NVS settings to default values. "sta.pmk", "sta.phym" and "auto.conn" are setting to default values when erase_flash is calling.

NoNullptr, 2022-10-26:
> hi @NoNullptr esp_wifi_restore() will restore persistent NVS settings to default values. "sta.pmk", "sta.phym" and "auto.conn" are setting to default values when erase_flash is calling.

I expect esp_wifi_restore() to remove the key-value pairs from NVS that were previously written, instead of explicitly writing defaults that are hard-coded in the codebase already. This would be more robust in case these keys or the default values change in the future.

"auto.conn" specifically seems to be an obsolete key from an older ESP-IDF (it was used in v3.3, but deprecated and then removed, see esp_wifi_set_auto_connect). I don't know if the internal WiFi code still uses this key, but in v4.4.2 only esp_wifi_restore() seems to write it – normal setup and connection doesn't.

NoNullptr, 2022-10-26:
> @NoNullptr can we have the application code which your are using to see this bug,Also can you mention the commit ID you are using.

Here is a simple example that illustrates the behaviour. It does the following:

1. erase all net80211 NVS keys
2. connect as STA to WiFi
3. print all net80211 NVS keys
4. call esp_wifi_restore()
5. print all net80211 NVS keys again

On v4.4.2 the behavior is currently this:
- After successfully connecting to WiFi, there are 18 keys on NVS.
- After calling esp_wifi_restore, there are 40 keys on NVS, including keys for AP, which wasn't set or started, and keys that seem to belong to v3 ESP-IDF ("auto.conn").

When changing the configuration, only the changed parameters are stored on NVS, for example the key "sta.phybw" is only stored when esp_wifi_set_bandwidth() was called. However, esp_wifi_restore() stores all default values (and some more) on NVS, instead of erasing them.

```
#include <esp_event.h>
#include <esp_log.h>
#include <esp_netif.h>
#include <esp_wifi.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <nvs_flash.h>

const char TAG[] = "test";

wifi_config_t wifi_config_sta = {
    .sta = {
        .ssid = "",
        .password = "",
    }
};

void dump_nvs_net80211()
{
    nvs_entry_info_t info;
    nvs_iterator_t it;

    it = nvs_entry_find("nvs", "nvs.net80211", NVS_TYPE_ANY);
    while (it != NULL) {
        nvs_entry_info(it, &info);
        printf("nvs.net80211 has key \"%s\"\n", info.key);
        it = nvs_entry_next(it);
    }
}

void erase_nvs_net80211()
{
    nvs_handle_t handle;
    nvs_entry_info_t info;
    nvs_iterator_t it;

    ESP_ERROR_CHECK(nvs_open("nvs.net80211", NVS_READWRITE, &handle));
    it = nvs_entry_find("nvs", "nvs.net80211", NVS_TYPE_ANY);
    while (it != NULL) {
        nvs_entry_info(it, &info);
        ESP_ERROR_CHECK(nvs_erase_key(handle, info.key));
        it = nvs_entry_next(it);
    }
    ESP_ERROR_CHECK(nvs_commit(handle));
    nvs_close(handle);
}

void app_main(void)
{
    ESP_ERROR_CHECK(nvs_flash_init());
    erase_nvs_net80211();

    ESP_ERROR_CHECK(esp_netif_init());
    ESP_ERROR_CHECK(esp_event_loop_create_default());

    esp_netif_t *netif_sta = esp_netif_create_default_wifi_sta();
    wifi_init_config_t wifi_init_config = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&wifi_init_config));

    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
    ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_STA, &wifi_config_sta));

    ESP_ERROR_CHECK(esp_wifi_start());
    vTaskDelay(pdMS_TO_TICKS(5000));

    dump_nvs_net80211();
    vTaskDelay(pdMS_TO_TICKS(500));

    ESP_ERROR_CHECK(esp_wifi_restore());
    vTaskDelay(pdMS_TO_TICKS(5000));

    dump_nvs_net80211();
    vTaskDelay(pdMS_TO_TICKS(500));
}
```

mhdong, 2022-11-01:
hi @NoNullptr  Yes, esp_wifi_restore() stores all default values on NVS, instead of erasing them. When changing the configuration, only the changed parameters are stored on NVS, the rest will not be changed or erased.

NoNullptr, 2022-11-02:
> hi @NoNullptr Yes, esp_wifi_restore() stores all default values on NVS, instead of erasing them. When changing the configuration, only the changed parameters are stored on NVS, the rest will not be changed or erased.

I described that already in the bug report. It is reasonable to expect esp_wifi_restore() to erase the current configuration from NVS, not to store new values (even if they're the internal defaults, and in particular not values that seem to be part of an older ESP-IDF version).

mhdong, 2022-11-02:
`/**
 * @brief  Restore WiFi stack persistent settings to default values
 *
 * This function will reset settings made using the following APIs:
 * - esp_wifi_set_bandwidth,
 * - esp_wifi_set_protocol,
 * - esp_wifi_set_config related
 * - esp_wifi_set_mode
 *
 * @return
 *    - ESP_OK: succeed
 *    - ESP_ERR_WIFI_NOT_INIT: WiFi is not initialized by esp_wifi_init
 */
esp_err_t esp_wifi_restore(void);`
hi @NoNullptr esp_wifi_restore() just restore wifi nvs setting to default values.

podaen, 2022-11-07:
I want to make prov work using wifi as multiple instance, but it doesn't seem possible to do it like this also because of the nvs.

```
    WiFiProvClass* pWiFiProv= nullptr;
    WiFiClass* pWiFi= nullptr;
    uint32_t cntNeworks = 0;
```
I listed all storages from the documentation.

out	https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html#_CPPv417esp_wifi_get_modeP11wifi_mode_t
out	https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html#_CPPv416esp_wifi_restorev
out	https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html#_CPPv419esp_wifi_scan_startPK18wifi_scan_config_tb
out	https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html#_CPPv424esp_wifi_scan_get_ap_numP8uint16_t
out	https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html#_CPPv428esp_wifi_scan_get_ap_recordsP8uint16_tP16wifi_ap_record_t
out	https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html#_CPPv415esp_wifi_get_psP14wifi_ps_type_t
out	https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html#_CPPv421esp_wifi_get_protocol16wifi_interface_tP7uint8_t
out	https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html#_CPPv422esp_wifi_get_bandwidth16wifi_interface_tP16wifi_bandwidth_t
out	https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html#_CPPv420esp_wifi_get_channelP7uint8_tP18wifi_second_chan_t
flash	https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html#_CPPv420esp_wifi_set_countryPK14wifi_country_t
out	https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html#_CPPv416esp_wifi_get_mac16wifi_interface_tAL6E_7uint8_t
out	https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html#_CPPv424esp_wifi_get_promiscuousPb
out	https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html#_CPPv431esp_wifi_get_promiscuous_filterP25wifi_promiscuous_filter_t
out	https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html#_CPPv436esp_wifi_get_promiscuous_ctrl_filterP25wifi_promiscuous_filter_t
out	https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html#_CPPv423esp_wifi_ap_get_sta_aidAL6E_K7uint8_tP8uint16_t
flash	https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html#_CPPv425esp_wifi_set_country_codePKcb
out	https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html#_CPPv420esp_wifi_get_countryP14wifi_country_t

these below doesn't mentioned that has been stored. Maybe it happens when the wifi starts...

https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html#_CPPv417esp_wifi_set_mode11wifi_mode_t
https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html#_CPPv421esp_wifi_set_protocol16wifi_interface_t7uint8_t
...

They are stored in to the nvs to reboot faster and to minimize the write cycles on the memory because the are limited. Every AP or STA does that.

podaen, 2022-11-07:
@kriegste You might be interested in this too.

jack0c, 2023-06-02:
master: ce1ab731be47fa2d509739b2ce7f0244ce836950
v4.4: 53c186af263ac76fb0a45e6f63037c43a86c3736
v4.3: 53c186af263ac76fb0a45e6f63037c43a86c3736

jack0c, 2023-06-02:
Close this since it is not a bug.

closed 2023-06-02 by jack0c