espressif/esp-idf #9914: I can't start esp-idf even after multiple reinstalls.

soulk87, 2022-10-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

You want to run esp-idf .

### What is the actual behavior?

I can run esp-idf. Please check below.
[image]

### Steps to reproduce.

1. install ESP-IDF via using "esp-idf-tools-setup-offline-4.4.2.exe"
2. Run ESP-IDF 4.4 CMD in windows desktop folder.

( I reinstalled ESP-IDF many times but It is not work)

### Build or installation Logs.

```plain
When I Run install.bat.
Installing ESP-IDF tools
Current system platform: win64
Selected targets are: esp32s2, esp32c3, esp32, esp32s3, esp32h2
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, xtensa-esp32s3-elf, riscv32-esp-elf, esp32ulp-elf, esp32s2ulp-elf, cmake, openocd-esp32, ninja, idf-exe, ccache, dfu-util
WARNING: tool xtensa-esp32-elf version esp-2021r2-patch3-8.4.0 is installed, but has reported version unknown
Installing xtensa-esp32-elf@esp-2021r2-patch3-8.4.0
file xtensa-esp32-elf-gcc8_4_0-esp-2021r2-patch3-win64.zip is already downloaded
WARNING: destination path already exists, removing
Extracting C:\Espressif\dist\xtensa-esp32-elf-gcc8_4_0-esp-2021r2-patch3-win64.zip to C:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0
WARNING: tool xtensa-esp32s2-elf version esp-2021r2-patch3-8.4.0 is installed, but has reported version unknown
Installing xtensa-esp32s2-elf@esp-2021r2-patch3-8.4.0
file xtensa-esp32s2-elf-gcc8_4_0-esp-2021r2-patch3-win64.zip is already downloaded
WARNING: destination path already exists, removing
Extracting C:\Espressif\dist\xtensa-esp32s2-elf-gcc8_4_0-esp-2021r2-patch3-win64.zip to C:\Espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch3-8.4.0
WARNING: tool xtensa-esp32s3-elf version esp-2021r2-patch3-8.4.0 is installed, but has reported version unknown
Installing xtensa-esp32s3-elf@esp-2021r2-patch3-8.4.0
file xtensa-esp32s3-elf-gcc8_4_0-esp-2021r2-patch3-win64.zip is already downloaded
WARNING: destination path already exists, removing
Extracting C:\Espressif\dist\xtensa-esp32s3-elf-gcc8_4_0-esp-2021r2-patch3-win64.zip to C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch3-8.4.0
WARNING: tool riscv32-esp-elf version esp-2021r2-patch3-8.4.0 is installed, but has reported version unknown
Installing riscv32-esp-elf@esp-2021r2-patch3-8.4.0
file riscv32-esp-elf-gcc8_4_0-esp-2021r2-patch3-win64.zip is already downloaded
WARNING: destination path already exists, removing
Extracting C:\Espressif\dist\riscv32-esp-elf-gcc8_4_0-esp-2021r2-patch3-win64.zip to C:\Espressif\tools\riscv32-esp-elf\esp-2021r2-patch3-8.4.0
Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping esp32s2ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping cmake@3.23.1 (already installed)
Skipping openocd-esp32@v0.11.0-esp32-20220411 (already installed)
[... 63 lines cut ...]
  Attempting uninstall: esp-windows-curses
    Found existing installation: esp-windows-curses 0.1
    Uninstalling esp-windows-curses-0.1:
      Successfully uninstalled esp-windows-curses-0.1
Successfully installed esp-windows-curses-0.1
WARNING: You are using pip version 20.3.3; however, version 22.2.2 is available.
You should consider upgrading via the 'C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe -m pip install --upgrade pip' command.
All done! You can now run:
   export.bat

After I run exprot.bat.
Setting IDF_PATH: C:\Espressif\frameworks\esp-idf-v4.4.2

Adding ESP-IDF tools to PATH...
WARNING: tool xtensa-esp32-elf version esp-2021r2-patch3-8.4.0 is installed, but has reported version unknown
ERROR: tool xtensa-esp32-elf has no installed versions. Please run 'C:\Espressif\frameworks\esp-idf-v4.4.2\install.bat' to install it.
WARNING: tool xtensa-esp32s2-elf version esp-2021r2-patch3-8.4.0 is installed, but has reported version unknown
ERROR: tool xtensa-esp32s2-elf has no installed versions. Please run 'C:\Espressif\frameworks\esp-idf-v4.4.2\install.bat' to install it.
WARNING: tool xtensa-esp32s3-elf version esp-2021r2-patch3-8.4.0 is installed, but has reported version unknown
ERROR: tool xtensa-esp32s3-elf has no installed versions. Please run 'C:\Espressif\frameworks\esp-idf-v4.4.2\install.bat' to install it.
WARNING: tool riscv32-esp-elf version esp-2021r2-patch3-8.4.0 is installed, but has reported version unknown
ERROR: tool riscv32-esp-elf has no installed versions. Please run 'C:\Espressif\frameworks\esp-idf-v4.4.2\install.bat' to install it.
WARNING: tool ccache version 4.3 is installed, but has reported version unknown
ERROR: tool ccache has no installed versions. Please run 'C:\Espressif\frameworks\esp-idf-v4.4.2\install.bat' to install it.
```

dobairoland, 2022-10-05:
Could you please try to run in your CMD the following command and share the output?

```
C:\Espressif\tools\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe --version
```

soulk87, 2022-10-05:
Hi @dobairoland 
Thank you for your fast response.
Here is the output.
C:\Espressif\tools>C:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe --version
xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2021r2-patch3) 8.4.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

I change the command C:\Espressif\tools\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe --version
to C:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe --version
(I have different dir path.)

Thank you

dobairoland, 2022-10-05:
Ok, thanks for the additional information. This looks good, so the issue is about something else.

It is interesting that the logs have proper backslashes (`\`) but in the screenshot there are weird characters instead. I wonder if this might be caused by improper Unicode setup.  @georgik Have you seen anything like this before?

georgik, 2022-10-06:
@soulk87 Please, can you try the PowerShell launcher instead of the CMD one?
The installer should deploy the launcher on the Desktop. 
Please try in PowerShell: `. .\export.ps1`

soulk87, 2022-10-06:
Hi @georgik 
Thank you for your suggest.
But the PowerShell has same problem...

georgik, 2022-10-06:
@soulk87 Please type: 
```idf-env config get```
This should return the configuration of installed ESP-IDF environments

soulk87, 2022-10-06:
@georgik 
Here is my configuration.
C:\Espressif\frameworks\esp-idf-v4.4.2>idf-env config get
{
    "$schema": "http://json-schema.org/schema#",
    "$id": "http://dl.espressif.com/dl/schemas/esp_idf",
    "_comment": "Configuration file for ESP-IDF Eclipse plugin.",
    "_warning": "Use / or \\ when specifying path. Single backslash is not allowed by JSON format.",
    "gitPath": "C:/Espressif/tools/idf-git/2.34.2/cmd/git.exe",
    "idfToolsPath": "C:/Espressif",
    "idfSelectedId": "esp-idf-e91d384503485fbb54f6ce3d11e841fe",
    "idfInstalled": {
        "esp-idf-e91d384503485fbb54f6ce3d11e841fe": {
            "version": "4.4",
            "python": "C:/Espressif/python_env/idf4.4_py3.8_env/Scripts/python.exe",
            "path": "C:/Espressif/frameworks/esp-idf-v4.4.2/"
        }
    }
}

georgik, 2022-10-06:
Thank you @soulk87 for the diagnostic information. I just did the installation into plain new Windows Sandbox on exactly same path and exact same config. The installation works.

Which antivirus do you have on the machine? 

Windows Defender should not have any issues, but some others may cause trouble. Basically, the error message is that the tool is not able to execute binary to get the version number, so it tries to reinstall it.

georgik, 2022-10-06:
@soulk87  Please try checking the logs from those security tools. They may delay or refuse the execution of the binary, which can cause trouble reading the version number. This might be often caused by the fact that tools are built using MinGW, which produces binary with a similar file signature to some viruses.

You can add exclusions to Windows defender using the command:
```
idf-env antivirus exclusion add --all
```
Further commands for Windows Defender exclusion are here: https://github.com/espressif/idf-env#working-with-antivirus

You can try the installation on the same machine using Windows Sandbox. If it works in the Sandbox, it indicates a problem with the security tool configuration.

closed 2023-06-29 by dobairoland