espressif/esp-idf #9915: Crash at file appending on SPIFFS

jrHernandezSio, 2022-10-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

Sometimes, I am having a crash when starting my device due to problems with the SPIFFS (apparently).

The problem comes when receiving data and writitng to a file in a SPIFFS partition.

When receiving data via HTTP a dynamic buffer is created based on the content-lenght header included in HTTP response.

`file_buffer = (uint8_t*) malloc(sizeof(uint8_t) * (content_len + 1));`

Then the full buffer is appended to a file.

```
DBG_print("Adding packet to file...\n");
FILE* f = fopen(file_name, "a");
if (f == NULL) 
{
    DBG_print("Failed to OPEN file\n");
    addOk = false;
}
...
```

Anyone can help on which can be the root cause.

```
-DEBUG-: [1664953447,560 s] Adding packet to file... 

abort() was called at PC 0x40082a79 on core 0

Backtrace:0x400923fb:0x3ffbab70 0x40092b6d:0x3ffbab90 0x40098fce:0x3ffbabb0 0x40082a79:0x3ffbac20 0x400843f5:0x3ffbac40 0x400852b9:0x3ffbac60 0x400852c5:0x3ffbac80 0x4008fe81:0x3ffbaca0 0x400850ad:0x3ffbacc0 0x40151d09:0x3ffbad00 0x40124207:0x3ffbad30 0x40127d43:0x3ffbad60 0x40125973:0x3ffbad90 0x4012714a:0x3ffbadd0 0x40124888:0x3ffbae10 0x40123fb9:0x3ffbae40 0x401523a6:0x3ffbae60 0x401a5520:0x3ffbae80 0x401a55c9:0x3ffbaeb0 0x400df034:0x3ffbaed0 0x400df186:0x3ffbaf00 0x400df28e:0x3ffbaf20 0x400df44d:0x3ffbaf60 0x400da6c9:0x3ffbafa0 0x400d4cab:0x3ffbafc0 0x400d5d75:0x3ffbafe0 0x400d6a58:0x3ffbb000 0x400d82bf:0x3ffbb030 0x401c6573:0x3ffbb170 0x40095ead:0x3ffbb190
  #0  0x400923fb:0x3ffbab70 in panic_abort at C:\Users\jr.hernandez\.platformio\packages\framework-espidf@3.40300.0\components\esp_system/panic.c:356
  #1  0x40092b6d:0x3ffbab90 in esp_system_abort at C:\Users\jr.hernandez\.platformio\packages\framework-espidf@3.40300.0\components\esp_system/system_api.c:112
  #2  0x40098fce:0x3ffbabb0 in abort at C:\Users\jr.hernandez\.platformio\packages\framework-espidf@3.40300.0\components\newlib/abort.c:46
  #3  0x40082a79:0x3ffbac20 in esp_intr_noniram_disable at C:\Users\jr.hernandez\.platformio\packages\framework-espidf@3.40300.0\components\esp_system/intr_alloc.c:802 (discriminator 1)
  #4  0x400843f5:0x3ffbac40 in spi_flash_disable_interrupts_caches_and_other_cpu at C:\Users\jr.hernandez\.platformio\packages\framework-espidf@3.40300.0\components\spi_flash/cache_utils.c:259
  #5  0x400852b9:0x3ffbac60 in cache_disable at C:\Users\jr.hernandez\.platformio\packages\framework-espidf@3.40300.0\components\spi_flash/spi_flash_os_func_app.c:78
  #6  0x400852c5:0x3ffbac80 in spi1_start at C:\Users\jr.hernandez\.platformio\packages\framework-espidf@3.40300.0\components\spi_flash/spi_flash_os_func_app.c:107
  #7  0x4008fe81:0x3ffbaca0 in spiflash_start_default at C:\Users\jr.hernandez\.platformio\packages\framework-espidf@3.40300.0\components\spi_flash/esp_flash_api.c:110
  #8  0x400850ad:0x3ffbacc0 in esp_flash_read at C:\Users\jr.hernandez\.platformio\packages\framework-espidf@3.40300.0\components\spi_flash/esp_flash_api.c:652
  #9  0x40151d09:0x3ffbad00 in esp_partition_read at C:\Users\jr.hernandez\.platformio\packages\framework-espidf@3.40300.0\components\spi_flash/partition.c:347
  #10 0x40124207:0x3ffbad30 in spiffs_api_read at C:\Users\jr.hernandez\.platformio\packages\framework-espidf@3.40300.0\components\spiffs/spiffs_api.c:36
  #11 0x40127d43:0x3ffbad60 in spiffs_phys_rd at C:\Users\jr.hernandez\.platformio\packages\framework-espidf@3.40300.0\components\spiffs\spiffs\src/spiffs_cache.c:161
  #12 0x40125973:0x3ffbad90 in spiffs_obj_lu_find_entry_visitor at C:\Users\jr.hernandez\.platformio\packages\framework-espidf@3.40300.0\components\spiffs\spiffs\src/spiffs_nucleus.c:166
  #13 0x4012714a:0x3ffbadd0 in spiffs_object_find_object_index_header_by_name at C:\Users\jr.hernandez\.platformio\packages\framework-espidf@3.40300.0\components\spiffs\spiffs\src/spiffs_nucleus.c:1694
  #14 0x40124888:0x3ffbae10 in SPIFFS_open at C:\Users\jr.hernandez\.platformio\packages\framework-espidf@3.40300.0\components\spiffs\spiffs\src/spiffs_hydrogen.c:230 (discriminator 2)
  #15 0x40123fb9:0x3ffbae40 in vfs_spiffs_open at C:\Users\jr.hernandez\.platformio\packages\framework-espidf@3.40300.0\components\spiffs/esp_spiffs.c:477
  #16 0x401523a6:0x3ffbae60 in esp_vfs_open at C:\Users\jr.hernandez\.platformio\packages\framework-espidf@3.40300.0\components\vfs/vfs.c:391 (discriminator 3)
  #17 0x401a5520:0x3ffbae80 in _fopen_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fopen.c:129
  #18 0x401a55c9:0x3ffbaeb0 in fopen at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fopen.c:168
  #19 0x400df034:0x3ffbaed0 in add_packet_to_file_avd at lib_prj/BG95/bg95.c:1354
  #20 0x400df186:0x3ffbaf00 in FILE_parse_append at lib_prj/BG95/bg95.c:1512
  #21 0x400df28e:0x3ffbaf20 in bg95_rx_http_to_file at lib_prj/BG95/bg95.c:1621
  #22 0x400df44d:0x3ffbaf60 in BG95_transmit_http_get at lib_prj/BG95/bg95.c:1739
  #23 0x400da6c9:0x3ffbafa0 in SEND_get_barcodes_to_file at src/send_manager.c:2271
  #24 0x400d4cab:0x3ffbafc0 in HD_rx_barcodes_to_file at src/main.c:979
  #25 0x400d5d75:0x3ffbafe0 in SYSTEM_notify at src/main.c:1730
  #26 0x400d6a58:0x3ffbb000 in SYSTEM_setup at src/main.c:1933
  #27 0x400d82bf:0x3ffbb030 in app_main at src/main.c:6850
  #28 0x401c6573:0x3ffbb170 in main_task at C:\Users\jr.hernandez\.platformio\packages\framework-espidf@3.40300.0\components\freertos\port/port_common.c:133 (discriminator 2)      
  #29 0x40095ead:0x3ffbb190 in vPortTaskWrapper at C:\Users\jr.hernandez\.platformio\packages\framework-espidf@3.40300.0\components\freertos\port\xtensa/port.c:168

ELF file SHA256: 3639363134373436

Rebooting...

Boot!...
 ```

Regards, Ramon.

jrHernandezSio, 2022-10-06:
Hi,

So sorry. I  missed thise details yesterday.

- framework-espidf @ 3.40300.0 (4.3.0)
- HARDWARE: ESP32 240MHz, 320KB RAM, 15.62MB Flash

jrHernandezSio, 2022-10-06:
Hi,

Updated to framework-espidf @ 3.40401.0 (4.4.1)

Same result, it crashed, but I think I know the root cause. It seems to be some unexpected values when recieving via HTTP that makes a buffer overflow somewhere.

Regards, Ramon.

rrtandler, 2022-11-23:
Hi Ramon,

Based on your observation of unexpected values when recieving via HTTP that potentially makes a buffer overflow, could you provide us with more details of your investigation so far? 

It was already mentioned earlier, it would help us if you could publish a code snippet causing the trouble.

rrtandler, 2023-02-06:
Hi Ramon,

It looks like there is no new information available for a while, so I am closing the issue without resolution. If you would need to reiterate on this topic, please provide details and reopen the issue.

Radek

closed 2023-02-06 by rrtandler