espressif/esp-idf #9916: ISSUE REGARDING BLUETOOTH RECONNECTION

beginner140BB, 2022-10-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

- 1. connect to esp32 via bluetooth from android phone.
- 2. Checking sending and receiving data.
- 3. diconnected using android phone.
- 4.then again  try to connect to the android phone but not able to connect.

...
[image]

don't know what is happenning with the events
need to also know what are the basic sdk configuration during bluetooth connection in when using espressif ide
@nkolban

vik-gokhale, 2022-10-06:
hi @beginner140BB,
We were thinking of looking at the issue in a detailed way...what is the reason you closed the issue?

beginner140BB, 2022-10-06:
> hi @beginner140BB, We were thinking of looking at the issue in a detailed way...what is the reason you closed the issue? Just curious.
> 
> Regards, Vikram
Hey @vik-gokhale 

Currently the issue has been solved on **example code** by updating to latest verion of ESP-IDF v4.4.2.
There are some bug fixes reported on ESP-IDF v4.4.2 so i think that solves the issue of bluetooth reconnection.

Need to check will there be any issues when integrating to project.

vik-gokhale, 2022-10-07:
hi @beginner140BB,
busy with more important things and looking at custom code is not something we do.
If we get enough time, we may help. But that is less probable.
If example solves the problem with custom code, please use the code from examples.

beginner140BB, 2022-10-07:
> If example solves the problem with custom code, please use the code from examples.

actually need to know what is happening in technical point of view when changing this to 1 connection after disconnection it is not reconnecting but able to pair and unpair the bluetooth.

need to know how we can configure the sdkconfig for this by enabling one device should be connected to bluetooth

[image]

does anyone expertise in playing with sdkconfig 😊🤷‍♀️

closed 2022-11-23 by beginner140BB

alighorbani69, 2024-03-30:
Hi 
I had the same issue, the same error:
BT_APPL: bta_jv_add_rfc_port failed to create new listen port

it was not reconnected, but after changing the CONFIG_BT_ACL_CONNECTIONS from 1 to 2 the issue fixed