espressif/esp-idf #9918: MQTT5 example failure with esp-tls

ChrisJones79, 2022-10-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-1006-g02605f1a31

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom Board | ACEIRMC D1 Mini NodeMCU ESP32 ESP-WROOM-32

### Power Supply used.

USB

### What is the expected behavior?

Connect to the MQTT event, execute example pub/sub actions.

### What is the actual behavior?

From the command prompt, idf.py monitor:

Leaving...
Hard resetting via RTS pin...
Executing action: monitor
Running idf_monitor in directory c:\espressif\frameworks\esp-idf-master\examples\protocols\mqtt5
Executing "C:\Espressif\python_env\idf5.1_py3.8_env\Scripts\python.exe C:\Espressif\frameworks\esp-idf-master\tools/idf_monitor.py -p COM15 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 --revision 0 c:\espressif\frameworks\esp-idf-master\examples\protocols\mqtt5\build\mqtt5.elf --force-color -m 'C:\Espressif\python_env\idf5.1_py3.8_env\Scripts\python.exe' 'C:\Espressif\frameworks\esp-idf-master\tools\idf.py' '-p' 'COM15'"...
←[0;33m--- WARNING: GDB cannot open serial ports accessed as COMx←[0m
←[0;33m--- Using \\.\COM15 instead...←[0m
←[0;33m--- idf_monitor on \\.\COM15 115200 ---←[0m
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
I (1�ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7032
load:0x40078000,len:15376
load:0x40080400,len:3816
0x40080400: _init at ??:?

entry 0x40080648
I (27) boot: ESP-IDF v5.1-dev-1006-g02605f1a31-dirty 2nd stage bootloader
I (27) boot: compile time Oct  5 2022 12:58:07
I (29) boot: chip revision: V100
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 2MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
[... 290 lines cut ...]

I (3972) wifi:AP's beacon interval = 102400 us, DTIM period = 3
I (5612) example_connect: Got IPv6 event: Interface "example_netif_sta" address: fe80:0000:0000:0000:96b5:55ff:fe27:05d8, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (5612) esp_netif_handlers: example_netif_sta ip: 192.168.0.102, mask: 255.255.255.0, gw: 192.168.0.1
I (5622) example_connect: Got IPv4 event: Interface "example_netif_sta" address: 192.168.0.102
I (5632) example_common: Connected to example_netif_sta
I (5642) example_common: - IPv4 address: 192.168.0.102,
I (5642) example_common: - IPv6 address: fe80:0000:0000:0000:96b5:55ff:fe27:05d8, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (5662) MQTT5_EXAMPLE: Other event id:7
W (5672) wifi:<ba-add>idx:0 (ifx:0, 50:0f:f5:de:da:e1), tid:0, ssn:4, winSize:64
W (6232) wifi:<ba-add>idx:1 (ifx:0, 50:0f:f5:de:da:e1), tid:2, ssn:0, winSize:64
E (16242) esp-tls: [sock=54] select() timeout
E (16242) transport_base: Failed to open a new connection: 32774
E (16242) mqtt_client: Error transport connect
I (16242) MQTT5_EXAMPLE: MQTT_EVENT_ERROR
I (16252) MQTT5_EXAMPLE: MQTT5 return code is 0
E (16252) MQTT5_EXAMPLE: Last error reported from esp-tls: 0x8006
I (16262) MQTT5_EXAMPLE: Last errno string (Success)
I (16262) MQTT5_EXAMPLE: MQTT_EVENT_DISCONNECTED

Done

C:\Espressif\frameworks\esp-idf-master\examples\protocols\mqtt5>git describe --tags
v5.1-dev-1006-g02605f1a31
```

meltdown03, 2022-10-09:
I can confirm the same error on the ESP32-S3 Dev Module as well.

ESP-YJM, 2022-10-10:
@ChrisJones79 Could you please try other broker URL, such as `test.mosquitto.org`. The origin broker URL `mqtt.eclipseprojects.io` is not available.

ChrisJones79, 2022-10-10:
I will try and get back to you soon.

meltdown03, 2022-10-10:
Thanks @ESP-YJM, that URL works for me

ChrisJones79, 2022-10-10:
@ESP-YJM yes, it works for me too.

closed 2022-10-10 by ChrisJones79

ChrisJones79, 2022-10-11:
You may close this. Thank you Neal.