espressif/esp-idf #9919: ESP32 I2C: gaps (about 19 ms) between transactions at 100hz Freertos tickrate

mitchellcairns, 2022-10-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

For my setup, I'm using an official ESP-32 WROOM devkit. 240Mhz set for the clock speed, though changing this setting did not have any impact.

I'm on branch 4.4 release (which supposedly has a fix listed in this commit https://github.com/espressif/esp-idf/commit/cb411a81a2029d7b68926520d6e1ea0590b04988 )

I do not believe that patch properly addressed the issue and more inspection is necessary. 

Changing the 100hz tickrate for Freertos to 1000hz is a band-aid on the issue in my estimation. If this is the expected route to go, let me know and I can close the issue. 

Here's the gaps at 100hz
[image]

Gaps at 1000hz tickrate
[image]

mitchellcairns, 2022-10-05:
Furthermore, this is the relevant code minus the driver setup which has the I2C line running at 400kHz

[image]

I perform two I2C transactions one right after the other to demonstrate that the gap is intrinsically linked to the Freertos tick rate.

o-marshmallow, 2022-10-08:
Hi @mitchellcairns ,

I just tested on my side, with `release/v4.4` branch, commit 9269a536ac123677b165ff531369dfd1338ef103, and I cannot reproduce your issue. I am also writing 8 bytes to a device in a `while(1)` loop, here is what I get:
[image]

[image]

I kept the default FreeRTOS configuration of 100Hz, and set the CPU clock to 240MHz. My I2C bus is set to 400KHz with 2kOhm pull-up resistors (internal pull-up disabled).

I tested with the approach you showed (`i2c_master_start`, `i2c_master_write`, etc...) and the simplest approach using `i2c_write_to_device`. In both cases, the delay between transactions is less than 50us.

mitchellcairns, 2022-10-08:
> Hi @mitchellcairns ,
> 
> I just tested on my side on an ESP32-WROOM-32UE, with `release/v4.4` branch, commit [9269a53](https://github.com/espressif/esp-idf/commit/9269a536ac123677b165ff531369dfd1338ef103), and I cannot reproduce your issue. I am also writing 8 bytes to a device in a `while(1)` loop, here is what I get: [image]
> 
> [image]
> 
> I kept the default FreeRTOS configuration of 100Hz, and set the CPU clock to 240MHz. My I2C bus is set to 400KHz with 2kOhm pull-up resistors (internal pull-up disabled).
> 
> I tested with the approach you showed (`i2c_master_start`, `i2c_master_write`, etc...) and the simplest approach using `i2c_write_to_device`. In both cases, the delay between transactions is less than 50us.
> 
> Can you have a try with the following:
> 
> ```
>     while (1) {
>         i2c_master_write_to_device(I2C_NUM_0, USB_I2C_ADDR, output_buffer, 8, 1000/portTICK_PERIOD_MS);
>     }
> ```
> 
> What about your I2C bus setup? Are you using internal pull-ups or external ones? If external ones, which value?

Thanks for the attempt to replicate. I'm using external pull-ups with internal disabled. Using 1k pull-ups. I don't think that's related as the actual signal is getting through just fine and the clock speed is accurate to what I'm setting; just getting that large delay between commands.

o-marshmallow, 2022-10-09:
@mitchellcairns I have just tried with an ESP32-WROOM-32, same result.

Can you have a try on master to check whether you also encounter this issue? Can you try with another device?

mitchellcairns, 2022-10-09:
> @mitchellcairns I have just tried with an ESP32-WROOM-32, same result, it's a success with less than 30us between transactions
> 
> Can you have a try on master to check whether you also encounter this issue? Can you try with another device?

I redownloaded everything once more and now I'm getting 500us timings between commands. This is much more reasonable. I'm not sure what changed, but it seems that the issue is resolved for now.

closed 2022-10-09 by mitchellcairns