espressif/esp-idf #9921: esp_bt_controller_deinit() function causes assert isssue while using BLE5.0 extended scanning

AishwaryaK24, 2022-10-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello Team,

I am currently using ESP32C3 and esp-idf version v4.3.2 (also tried using v4.4.2). I am using the ble5.0 on ESP in controller only mode and VHCI with transportation medium as the socket communication. 

I am sending a series of commands to set the extended scan parameters and start extended scanning with duplicate filter turned ON. I am successfully getting the adv reports and able to sent the peripheral list to the Host. However during the scanning process, if the user closes the socket, it is implemented such that the ESP immediately executes the de-initialization process (**esp_bt_controller_disable() followed by esp_bt_controller_deinit()**)for the controller. The **esp_bt_controller_disable()** function gets executed successfully however the deinit function gives the following error
**assert llm_scan.c 1489, param 00000000 00000000**

Also, I tried sending manually the ext scan disable command frame to the controller before the deinit process and the error disappears. 

Note: This error does not occur when using the legacy scan settings or when using the extended scan with duplicate filter turned OFF. It is encountered only during extended scan with duplicate filter turned ON.

The issue is that the de-initialization can be called at any instance and hence it would be really helpful if someone could point me towards an ideal process for de-initializing the controller (irrespective of the controller state) and the significance of the assert and the  **param** being displayed.  . 

Thank you.

Weijian-Espressif, 2022-10-08:
@AishwaryaK24 We have reproduced the problem, we will fix it as soon as possible and provide you with the fix code

AishwaryaK24, 2022-10-11:
@Weijian-Espressif
Noted. 
Thank you so much for your prompt response. Waiting for the fix.

Weijian-Espressif, 2022-10-11:
[libbtdm_app.a.zip](https://github.com/espressif/esp-idf/files/9750618/libbtdm_app.a.zip)

I have fixed this problem, please test, by the way, it is recommended to stop all events of bluetooth when releasing bluetooth @AishwaryaK24

AishwaryaK24, 2022-10-11:
@Weijian-Espressif
Thank you so much for the updated library. Will test with it and let you know asap.
Also regarding stopping all the events of bluetooth, I use esp_bt_controller_disable() before calling the esp_bt_controller_deinit() function. Also, I tried to reset the controller by manually sending the HCI Reset frame to the controller. However,  it did not resolve my issue. 
Could you please suggest a way to stop all the ongoing ble events and then initiate de-initialization?

Weijian-Espressif, 2022-10-11:
[libbtdm_app.a.zip](https://github.com/espressif/esp-idf/files/9752080/libbtdm_app.a.zip)
sorry, please retry again

Weijian-Espressif, 2022-10-11:
@AishwaryaK24 I mean when you release bluetooth, if there is an event running, you should stop the event first. For example, if you are scanning, you should stop scanning, and then release bluetooth.

AishwaryaK24, 2022-10-11:
@Weijian-Espressif 
Thank you for the updated library. I tested the updated library and it does not give compilation error. 
However it returns error code 0x101 for esp_bt_controller_init(&cfg), where 
cfg =  BT_CONTROLLER_INIT_CONFIG_DEFAULT();
Also, on the console it gives the below log
`Config struct mismatch: magic=5a5aa5a5, ver=02112280`
I am currently using esp-idf v4.3.2. Does the version of esp-idf used affect this?

Weijian-Espressif, 2022-10-11:
Can you  provide me the IDF commit ?

AishwaryaK24, 2022-10-11:
Understood. Will try with the latest v4.3.
Btw, it would be really helpful if you could please share with me the esp-idf version used by you for the provided library?

Weijian-Espressif, 2022-10-11:
The version of the test lib we provide is the latest lib corresponding to the current release branch

AishwaryaK24, 2022-10-13:
Understood. Will check with the latest version. 
Thank you for the support.

AishwaryaK24, 2022-10-20:
Hello @Weijian-Espressif 
Sorry for the delay.
I checked the updated library provided by you, with the latest release for v4.3 i.e. v4.3.4.
The config struct mismatch error did not occur, however, the original error **assert llm_scan.c 1489, param 00000000 00000000** is getting displayed again.
Could you please help with this?
Thank you.

Weijian-Espressif, 2022-10-20:
@AishwaryaK24  Please provide the complete log, I want to see the commit information of lib.

AishwaryaK24, 2022-10-20:
@Weijian-Espressif I am attaching the log below. 
Please let me know if you need any other information.
[ble_log.txt](https://github.com/espressif/esp-idf/files/9827144/ble_log.txt)

Weijian-Espressif, 2022-10-20:
@AishwaryaK24  Please check whether you are using the lib provided by me. According to the log information, the commit information of the lib is incorrect

AishwaryaK24, 2022-10-20:
@Weijian-Espressif 
I am replacing the existing library at **/esp-idf/components/bt/controller/lib_esp32c3_family/esp32c3** with the library provided by you. Then executing idf.py fullclean and idf.py build.
It will be helpful if you could let me know the correct process if the above is incorrect.
Thank you.

Weijian-Espressif, 2022-10-20:
@AishwaryaK24  
[image]

There is no problem with your steps, I just re-tested the lib in the attachment, the commit information is correct, and there is no problem with the lib.

Do you have multiple IDF projects? Is it possible that the IDF PATH points to other IDF projects?

AishwaryaK24, 2022-10-21:
@Weijian-Espressif 
Thank you for the suggestion. Yes, the issue was the IDF_PATH setting. 
However, after I changed the IDF_PATH to current directory(v4.3.4) and built the code, for some reason it is not connecting to wifi (using the original btdm lib of v4.3.4). I tried the same code with the current production code idf version(v4.3.2), and it was able to connect to wifi. 
Due to this I am not able to connect to cloud for testing ble. 
I am attaching the failed log file, will it be possible for you to suggest some solution?
Sorry for the inconvenience.
Thank you.

[failed_wifi_connection.txt](https://github.com/espressif/esp-idf/files/9836015/failed_wifi_connection.txt)

Weijian-Espressif, 2022-10-21:
@AishwaryaK24 I'm not sure what's going on with wifi, you can use the latest IDF like master, v5.0 or 4.4

AishwaryaK24, 2022-10-21:
Understood. Give me some time to check and I will get back to you.
Thank you.

Alvin1Zhang, 2023-05-24:
Thanks for reporting, fix is available at https://github.com/espressif/esp-idf/commit/4a3319d381789a11f463d8062934e384006de97b, and already back ported to v5.0, v4.4 and v4.3. Feel free to reopen.

closed 2023-05-24 by Alvin1Zhang