espressif/esp-idf #9922: Is there any way to suspend and resume esp modbus slave task?

Rbhagwat62, 2022-10-06:
### Is your feature request related to a problem?

Hello,

Currently I am working on the non WiFi Modbus implementation on ESP32. I want to discuss regarding the code implementation for my requirement.

My project has two UARTs, one is connected to Modbus Master device and other is connected to the Modbus Slave device.

Master device functionality is to keep poling holding registers from the slave device and display its value on local display. It also write holding register values to the Slave device.

Cloud part will going to add later where slave device's all register values are updated to cloud and any value updated from cloud is write on the slave device.

On ESP I have started with esp-modbus component, ESP act as master for slave device and slave for master device. On ESP I have implemented master task which has polling scheduler for periodic reading of slave device registers and store it into uint16 type array. Slave task is used to send response master device for that same uint16 array is passed to mbc_slave_set_descriptor API. The uint16 array is basically cache values of the slave device.

Currently, all parameters from slave device keep updating in the array as well master device also gets values. So for reading functionality for master device everything is working fine.

But when master device is writing any Parameter, slave task is simply writes value in array and sends response back to master device. Master device then start polling again for reading holding registers. On ESP, write event is detected by checking mbc_slave_check_event API. My requirement is to write values to slave device through master task and pass its response back to master device.

### Describe the solution you'd like.

Is there any way whenever write request receive from Master device, I can pause the slave task, write data to salve device through master task, after completing write on slave device send response back to Master device and finally resume the slave task to get read requests from master device.

I check the slave source code, in mbc_serial_slave_start function eMBInit is called which has call backs for pvMBFrameStartCur, pvMBFrameStopCur functions used to resume or suspend the vUartTask for Modbus UART . These callbacks are called using eMBEnable and eMBDisable. I thought of use these API to stop Uart task for slave task and resume as per my requirement. But these are private API and not sure if it is the correct way.

### Describe alternatives you've considered.

Alternately, Instead of using slave from esp-modbus I can implement low level UART APIs to receive request and send response back to master device. In this case I need to implement all Modbus parsing and error handling.

Suggest me if any better option I can implement.

mlotorto, 2024-03-22:
Hello

Did you find a solution or workaround for this issue?
Thank you

Rbhagwat62, 2024-05-02:
@alisitsyn : Thanks for your Reply.

I already write my own functions for Modbus slave implementation. In future if required I will get back to you.

closed 2024-05-02 by Rbhagwat62

alisitsyn, 2024-05-02:
@Rbhagwat62,

Thank you for the update. Unfortunately the v1.0.x will not be updated with the new features and I can not change the public interface of it. So, please update it on your side using the example from v2.0.x. Feel free to reopen the issue or ask the questions at any time. Good luck!