espressif/esp-idf #9923: ISP read data only in byte multiples?

EspritElf, 2022-10-06:
I'm collecting  10-bit AD data through the ISP bus on the ESP32 and I noticed that if I set length = rxlength = 10 bits, I only receive the 8 most significant bits into rx_data[0] while rx_data[1] is equal to 0x00.

If I set length = rxlength = 7 bits, then both rx_data[0] and rx_data[1] are equal to 0x00, no data is received into the registers. 

Only when I set length = rxlength = 16 bits do I get all 10 bits of data. I verified all three conditions using an o-scope to check the data traffic. 

Is this the way the SPI controller functions or am I missing a setting?

Maksons, 2022-10-16:
Small question, do you initialized SPI Host with DMA or without?
I had similar problem described in #8636 and also some other reports about SPI RX problem with DMA enabled.
And also there is provided pull request with addition SPI transaction flag (to disable DMA) in #9069 and if possible could you give a try to it ?

EspritElf, 2022-10-16:
I read your issue report and yes they are very similar.  I initialize the SPI Host with the flag bit SPI_DMA_CH_AUTO set.  I will use the pull and see if that works with the SPI_TRANS_DONT_DMA transaction flag bit set. It'll be later this week before I can get to it as I have my head wrapped around some Verilog right now and I need to set up my test station to properly test it.

I really wanted to use DMA to access the AD chip I'm using but unfortunately I can find no way to deassert the CS line in between DMA sample reads, which is required for the AD to initiate the next sample. Using polling transmit I am only able to sustain about 2k samples per second from the AD.

EspritElf, 2022-10-23:
Maksons,
Sorry for the delay. I was able to get back to this a couple days ago and unfortunately my Git-Fu is not very strong. I was using esp-idf v 4.4.2 and I didn't know how to checkout your PR and merge it with 4.4.2. I decided to move to 5.0 and attempt to merge the PR with that version but it took me a few days to make my code compliant with 5.0, the gpTimer changes were especially painful. Anyway, I finally got that done but before I did that I tried SPI transactions with the SPI_DMA_DISABLED flag set using v 4.4.2. Setting that the SPI Controller still would only transfer full bytes, not just the 10 bits I need from the AD.

I just tried the transfer with esp-idf 5.0 with and without SPI_DMA_DISABLED. When it is not set I am unable to transfer other than byte multiples, but when I do set SPI_DMA_DISABLED I am finally able to transfer the 10 (and only 10) bits of the AD sample. It looks like my problem is solved by 5.0 as long as SPI_DMA_DISABLED is set.

I would try PR 9069 but I am frankly exhausted from this problem and am glad it is done. I think the only case I could try would be to not set SPI_DMA_DISABLED and then set SPI_TRANS_DONT_DMA, but I think it would only accomplish the same thing and wouldn't say much about the validity of 9069.

Best of luck on your issues with the SPI Controller.

EspritElf, 2022-10-23:
Forgot to add, thanks for alerting me about the issue with DMA and SPI, I would never have thought that was the problem. I spent hours reading all the Rxdata registers and my O-Scope for any clues as to how the data magically disappears between the O-Scope and my SW.