espressif/esp-idf #9924: ESP32S3 uses QSPI to send data causing length information to be overwritten

mmMicky, 2022-10-06:


mmMicky, 2022-10-06:
It seems normal that it sends a small amount of data. In the example, the length is 194 x 110 x 2 x 16=682,880(bit). Only a small number of messages are sent.

Maksons, 2022-10-16:
Looks like some 16KiB transaction limitation ?
150*109 = 16350 (< 16384)
150*110 = 16500 (> 16384) , 116 bytes over , maybe only those got sent ?
From ESP-IDF SPI Master:
"Maximum transfer size, in bytes. Defaults to 4092 if 0 when DMA enabled..."
may be it got *4 because of QSPI type of transaction

Forairaaaaa, 2023-04-09:
Hi, I meet the same issue, when my data is more than `32,000 bytes` like `34,000` big, it'll only send `2,000` byte. But it's normal when I send `34,000` bytes one by one

dj140, 2024-07-16:
I get the same issue too but it seems to behave a little differently, when I transfer a large amount of data, the QSPI bus will stall every 32768 dat and an error will occur when the transmitted data is greater than 32768 * 8 bytes.

 `E (1627) lcd_panel.io.spi: panel_io_spi_tx_color(387): spi transmit (queue) color failed`

is esp32s3 max dma transfer size 32768 bytes?