espressif/esp-idf #9928: Discovery delegate crashes OpenThread Border Router if the name doesn't exist

fcgdam, 2022-10-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-1006-g02605f1a31

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### Development Kit.

ESP32-WROOM DEVKit as OTBR, NRF52840 as OT node

### Power Supply used.

USB

### What is the expected behavior?

When according to documentation, if using Discovery Delegation, the DNS server set on the node is the address of the border router, and a resolve query is made for **default.service.arpa.** then either a answer is returned or (at least) a not found is returned. 

### What is the actual behavior?

If doing a DNS resolve query to find a published service and the service exists, then it answers correctly:
```
uart:~$ ot dns browse _test._udp.default.service.arpa
DNS browse response for _test._udp.default.service.arpa.
testhost
    Port:12345, Priority:0, Weight:0, TTL:120
    Host:arboria-3.default.service.arpa.
    HostAddress:fd11:1111:1122:2222:c2d8:7ba9:9e6d:5d3f TTL:120
    TXT:[test=31, dn=616162626262] TTL:120

Done
uart:~$ 
```
Now doing a DNS resolve query for the returned hostname **arboria-3.default.service.arpa.**:
```
uart:~$ ot dns resolve arboria-3.default.service.arpa.
DNS response for arboria-3.default.service.arpa. - fd11:1111:1122:2222:c2d8:7ba9:9e6d:5d3f TTL:120 
Done
uart:~$ 
```
But, if, for example we try to resolve **arboria-2.default.service.arpa.** which does not exist, the Border Router crashes and reboots.

```
uart:~$ ot dns resolve arboria-2.default.service.arpa.
... lot's of messages, since the BR rebooted.
Error 28: ResponseTimeout
```

### Steps to reproduce.

1) On a Linux machine on the same local LAN segment of the BR publish a service:
```
avahi-publish-service testhost _test._udp 12345 test=1 dn="aabbbb"
```
2) On the BR find the IP addresses:
```
> ipaddr

fd2a:abcd:31a3:ab09:0:ff:fe00:fc11
fdbf:fcc1:5c19:1:c76:ed57:ee88:4053
fd2a:abcd:31a3:ab09:0:ff:fe00:fc38
fd2a:abcd:31a3:ab09:0:ff:fe00:fc10
fd2a:abcd:31a3:ab09:0:ff:fe00:1c00
fd2a:abcd:31a3:ab09:14c1:9197:6902:597c
fe80:0:0:0:f8b9:23b0:cbd1:3331
Done
```

3) Use one of the global addresses to configure as the DNS server on the OT node:
```
uart:~$ ot dns config fd2a:abcd:31a3:ab09:14c1:9197:6902:597c
Done
uart:~$ ot dns config
Server: [fd2a:abcd:31a3:ab09:14c1:9197:6902:597c]:53
ResponseTimeout: 6000 ms
MaxTxAttempts: 3
RecursionDesired: yes
Done
uart:~$ 
```

4) Find the hostname running the published service for the published service:
```
uart:~$ ot dns browse _test._udp.default.service.arpa
DNS browse response for _test._udp.default.service.arpa.
testhost
    Port:12345, Priority:0, Weight:0, TTL:120
    Host:arboria-3.default.service.arpa.
    HostAddress:fd11:1111:1122:2222:c2d8:7ba9:9e6d:5d3f TTL:120
    TXT:[test=31, dn=616162626262] TTL:120

Done
uart:~$ 
```

5) Query for the reported hostname:
```
uart:~$ ot dns resolve arboria-3.default.service.arpa.
DNS response for arboria-3.default.service.arpa. - fd11:1111:1122:2222:c2d8:7ba9:9e6d:5d3f TTL:120 
Done
uart:~$ 
```

6) Make the BR crash:
```
uart:~$ ot dns resolve INVALID_SOMETHING.default.service.arpa.
uart:~$ 
```

### Debug Logs.

```plain
> E (494105) OPENTHREAD: Assert failed at /IDF/components/openthread/openthread/src/core/net/dnssd_server.cpp:1017
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400014fd  PS      : 0x00060730  A0      : 0x80090d72  A1      : 0x3ffd8f60  
A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffd8f30  
A10     : 0x00000028  A11     : 0x00000001  A12     : 0x0000000a  A13     : 0x0000000c  
A14     : 0x00000000  A15     : 0xffffffff  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400014fa:0x3ffd8f60 0x40090d6f:0x3ffd8f70 0x401a2219:0x3ffd9090 0x401487b1:0x3ffd90c0 0x401b72c9:0x3ffd90e0 0x401a6e7d:0x3ffd9100 0x401a2c27:0x3ffd9140 0x401a1a6d:0x3ffd9170 0x400d8fd8:0x3ffd91c0 0x4008d3e5:0x3ffd9250

ELF file SHA256: 3c9447ff56e57383

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7024
load:0x40078000,len:15376
load:0x40080400,len:3816
entry 0x40080648
I (27) boot: ESP-IDF v5.1-dev-1006-g02605f1a31 2nd stage bootloader
```

chshu, 2022-10-08:
@fcgdam The bug has been fixed internally, will update once it's synced to Github.

chshu, 2022-10-08:
@fcgdam Could you pull the latest IDF mater branch and verify the fix? It's included in the commit 1d826c7a.

fcgdam, 2022-10-10:
Good morning:

From the preliminary tests that I've done, at least the cases that led to the BR to reboot no longer happens.
I've tested with valid and invalid queries and the BR never crashed. So it seems to be solved.

closed 2022-10-10 by fcgdam