espressif/esp-idf #9930: Request BT bss and data segments be released on C3 and others as they are on the ESP32

stevegrau-ayla, 2022-10-06:
Can the data and bss segments of the BT stacks be released on the C3 and other chips as they are with ESP32? I understand that controller memory is not releasable on all chips. However, it seems like it should still be possible to release the data and bss segments, as is done on the ESP32.

See "if (mode == ESP_BT_MODE_BTDM)" cases in ESP32 implementations of esp_bt_controller_mem_release and esp_bt_mem_release.

https://github.com/espressif/esp-idf/blob/02605f1a31be4233b97e83a0460bb458f277dfea/components/bt/controller/esp32c3/bt.c#L886

stevegrau-ayla, 2022-10-06:
Looking at the map _bt_data is a lot smaller on the C3 than on ESP32. Looks like 288 bytes can be released on the C3 verses more than 4K on the ESP32, so not a huge amount to be gained but not completely insignificant either.

` 0x000000003fc958a4 _bt_data_start = ABSOLUTE (.)
 0x000000003fc95964 _bt_data_end = ABSOLUTE (.)
 0x000000003fc95964 _btdm_data_start = ABSOLUTE (.)
 0x000000003fc959c4 _btdm_data_end = ABSOLUTE (.)	288 bytes

 0x000000003fc9646b _nimble_data_start = ABSOLUTE (.)
 0x000000003fc9646c _nimble_data_end = ABSOLUTE (.)	1 byte

 0x000000003fca2e50 _bt_bss_start = ABSOLUTE (.)
 0x000000003fca2e50 _bt_bss_end = ABSOLUTE (.)
 0x000000003fca2e50 _nimble_bss_start = ABSOLUTE (.)
 0x000000003fca2e50 _nimble_bss_end = ABSOLUTE (.)
 0x000000003fca2e50 _btdm_bss_start = ABSOLUTE (.)
 0x000000003fca2e50 _btdm_bss_end = ABSOLUTE (.)	0 bytes`

Weijian-Espressif, 2022-10-21:
@stevegrau-ayla Thank you for the reminder, we will support this feature as soon as possible

Weijian-Espressif, 2023-05-12:
@stevegrau-ayla Sorry for the late reply, we have added new functions to support the release of bss and data segment memory for this question please refer to the function  `esp_bt_controller_mem_release()` and esp_bt_mem_release().

Alvin1Zhang, 2023-05-24:
Thanks for reporting, fix is available at https://github.com/espressif/esp-idf/commit/76d805f15a1c29095c0fa2a7f3cffc6ee52ad0a0, feel free to reopen.

closed 2023-05-24 by Alvin1Zhang