espressif/esp-idf #9932: ESP32S3 + auto light sleep + GPIO interrupt + WiFi power save issue

idea--list, 2022-10-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Based on the WiFi power save example i wrote a program, which logs sensor data to an SD card (both using the same SPI bus).
On an ESP32 Rev3 board (Sparkfun ESP32 Thing Plus C) code is running as expected.

After porting the code to an ESP32S3 board (Adafruit 5323) i observe following:

1. If WiFi is enabled + auto light sleep is enabled: power consumption is at expected level, the board stops triggering GPIO interrupts in just 3-4 seconds.
2. If WiFi is enabled + auto light sleep is disabled: power consumption is ~25mA, the board stops triggering GPIO interrupts after about 150 seconds (time varies somewhat)

3. If WiFi is disabled + auto light sleep is disabled: power consumption is ~25mA, but the board keeps triggering GPIO interrupts as expected
4. If WiFi is disabled + auto light sleep is enabled: the board keeps triggering GPIO interrupts as expected, but power consumption is ~25mA

On the ESP32S3 based board only the 3rd scenario behaves as expected.

The Sparkfun board has an on-board SD slot, while in case of the other board i am using bread-board wires, not sure though if that might cause the issue.

Am using ESP-IDF V4.4.2 with VS Code on a Windows 10 x64 machine.

idea--list, 2022-10-08:
Today i tried to narrow down the issue.
The auto light sleep function only works with level type GPIO interrupts not with edge type.

So today i tried the generic_gpio example boundled to IDF V4.4.2 on the ESP32S3 board.
The example uses edge interrupts and those work as expected.
However if i change line 78 to
`io_conf.intr_type = GPIO_INTR_HIGH_LEVEL;`

Then the board gets to an infinite reboot loop telling:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375698
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x1654
load:0x403c9700,len:0xbb8
load:0x403cc700,len:0x2f88
SHA-256 comparison failed:
Calculated: bd8ada184ecd6ab45cd1f2047d02f12ea2b06daefc2fa6b837c0679a8ebd7524
Expected: dff6a5613d09d5e2938d36377a71dda527c7d9d846b3618fc884df9fc018d31b
Attempting to boot anyway...
entry 0x403c9954
```
I know the ESP32S3 is relatively new, but i think i might be missing something.
Has anyone an idea how use level interrupts on GPIO pins of the ESP32S3?

esp-lis, 2022-10-08:
> Today i tried to narrow down the issue. The auto light sleep function only works with level type GPIO interrupts not with edge type.
>
> So today i tried the generic_gpio example boundled to IDF V4.4.2 [...]

Maybe you can refer to [here](https://github.com/espressif/esp-idf/issues/8887#issuecomment-1131467372).

idea--list, 2022-10-20:
Thanks for the tip!
Found a combination of settings in menuconfig which solves the issue, though not sure why.

closed 2022-10-20 by idea--list