espressif/esp-idf #9933: LR mode not being applied

kriegste, 2022-10-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4.2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### Development Kit.

TTGO

### Power Supply used.

USB

### What is the expected behavior?

I want the Station to be able to connect in all modes ("protocols"), i.e. B, G, N or LR.

https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-guides/wifi.html#wi-fi-protocol-mode

> Call esp_wifi_set_protocol(ifx, WIFI_PROTOCOL_11B| WIFI_PROTOCOL_11G|WIFI_PROTOCOL_11N|WIFI_PROTOCOL_LR) to set the station/AP to BGN and the LR mode.

### What is the actual behavior?

Whenever WIFI_PROTOCOL_LR is included in the call to esp_wifi_set_protocol(), it is ignored.

### Steps to reproduce.

According to esp_wifi_types.h:
```
#define WIFI_PROTOCOL_11B         1
#define WIFI_PROTOCOL_11G         2
#define WIFI_PROTOCOL_11N         4
#define WIFI_PROTOCOL_LR          8
```

When I call
```
esp_wifi_set_protocol(ESP_IF_WIFI_STA, 1);
esp_wifi_get_protocol(ESP_IF_WIFI_STA, &proto);
ESP_LOGI(TAG, "%d", proto);

esp_wifi_set_protocol(ESP_IF_WIFI_STA, 8);
esp_wifi_get_protocol(ESP_IF_WIFI_STA, &proto);
ESP_LOGI(TAG, "%d", proto);

esp_wifi_set_protocol(ESP_IF_WIFI_STA, 15);
esp_wifi_get_protocol(ESP_IF_WIFI_STA, &proto);
ESP_LOGI(TAG, "%d", proto);
```

the expected output is
1
8
15

However, the actual output is
1
0
7

zhangyanjiaoesp, 2022-10-12:
@kriegste I have tested using ESP32 core board, and the `WIFI_PROTOCOL_LR` mode can be set correctly. I will try to reproduce your problem using TTGO.

kriegste, 2022-10-12:
I will try to put together a minimum code example.

kriegste, 2022-10-12:
I accidentally called esp_wifi_set_protocol() BEFORE esp_wifi_start(). There was no warning, but this seems to be the problem. However, the doc only speaks about errors when calling before esp_wifi_init():
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#_CPPv421esp_wifi_set_protocol16wifi_interface_t7uint8_t
So either esp_wifi_set_protocol() still needs to be fixed or to display better warnings. And the documentation needs a fix.

kriegste, 2022-10-12:
Related observation:
If I call esp_wifi_get_country() BEFORE esp_wifi_start(), the returned max_tx_power is 2.
If I call esp_wifi_get_country() AFTER esp_wifi_start(), the returned max_tx_power is 20.
The other values seem to be the same.

Now some questions arise:
What APIs are allowed to be called BEFORE esp_wifi_start()?
May I call esp_wifi_set_country_code()?
esp_wifi_set_ps()?
esp_wifi_set_mode()???

zhangyanjiaoesp, 2022-10-13:
I will check the code

kriegste, 2022-10-14:
Thanks. Looking forward to your findings.

In the meanwhile I found another issue with the protocol setting that might or might not be related:
https://github.com/espressif/esp-idf/issues/9978

zhangyanjiaoesp, 2022-10-17:
@kriegste, I have checked the code, `esp_wifi_set_protocol` and `esp_wifi_set_country()`, `esp_wifi_set_country_code()` need to be called after `esp_wifi_start()`, will update the doc.

zhangyanjiaoesp, 2022-10-17:
@igrr @kriegste 
Checked again, `esp_wifi_set_country()` can be called after esp_wifi_init and before esp_wifi_start. But if the API is called before esp_wifi_start, the max tx power parameter can't works, because this parameter need to set  phy, and the phy can be operated after wifi start.
This API `esp_wifi_set_max_tx_power` can be called to set max tx power after wifi start. 
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html?highlight=set_max_tx#_CPPv425esp_wifi_set_max_tx_power6int8_t

kriegste, 2022-10-17:
This is good information. Thank you for clarifying! :)

zhangyanjiaoesp, 2022-10-21:
@kriegste 
When call `esp_wifi_set_protocol()` to set LR mode before wifi start, the LR mode need to set HW register, and HW register can be operated after wifi start, so if you call `esp_wifi_get_protocol()` immediately, the return value will show there is no LR mode. But the if you call `esp_wifi_get_protocol()` after wifi start, then the return value will be right, because the LR state has been saved in NVS, when wifi start, the phy and HW register will be updated.
Here is my test code and log basing station example:
[image]
[image]

zhangyanjiaoesp, 2022-11-07:
@kriegste Do you still have any problem on the API using? If not, I think this issue can be closed.

kriegste, 2022-11-07:
Hi, now that the order in which the APIs may be called is clear, my questions are answered. Others may still have problems: I did not check whether you corrected the documentation.

My most urging main problem is this:
https://github.com/espressif/esp-idf/issues/9978
This currently makes the LR feature unusable for us.

kriegste, 2022-11-15:
I noticed a new behavior in v4.4.3. All this is after esp_wifi_start():

When I call esp_wifi_get_country(&country) the parameter country.max_tx_power is always 2.
When I call esp_wifi_get_max_tx_power(&power) the resulting power value is 78.
What is the correct value? Is this correct behavior? Previously (v4.4.2), country.max_tx_power was reported as 20 (as mentioned above).

Alvin1Zhang, 2024-03-19:
Thanks for reporting, will close the ticket. For new issue(s), please help file another ticket, thanks.

closed 2024-03-19 by Alvin1Zhang