espressif/esp-idf #9934: emac_hal_set_phy_data, declared but neither defined nor used

crackwitz, 2022-10-07:
I'm wondering, what's with this declaration? There is no corresponding definition/implementation.

https://github.com/espressif/esp-idf/blob/c18a72134d649aa1985b9e43f8e98198c4eaebf8/components/hal/include/hal/emac_hal.h#L216

There *is* a `emac_ll_set_phy_data` (note the LL, not HAL), used in conjunction with `emac_hal_set_phy_cmd` inside of `emac_esp32_write_phy_reg`.

suda-morris, 2022-10-09:
nice catch, it's a leftover during emac HAL driver refactor. It should be a simple wrapper over `emac_ll_set_phy_data`.

crackwitz, 2022-10-09:
I'm only sniffing around in those parts because I need to use the SMI/SDIO individually, to set some registers in _two_ PHYs I've got attached to the bus. unfortunately, the LL+HAL+IDF interfaces don't give me clean access.

I think I instantiated some IDF-level mac object, but then couldn't call the init() on that because _that_ also assumes the existence of a "mediator", which ropes in a phy object and already *starts some task* that's only required for actual ethernet comms (which I don't need), so I copy-pasted the GPIO setup from that... I guess my use case (no actual ethernet comms) is unusual.

closed 2023-04-25 by espressif-bot