espressif/esp-idf #9935: TLS memory errors with external SPI RAM (mbedtls_ssl_setup returned -0x7F00 & mbedtls_ctr_drbg_seed returned -0x0001)

ddomnik, 2022-10-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

01d014c42def8d0c19e1ce55c07de6761e092ffa

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32S3 16R2

### Power Supply used.

External 3.3V

### What is the expected behavior?

External SPI RAM expands the current RAM and does not lead to memory allocation issues.

### What is the actual behavior?

After switching to a new board that has external SPI RAM installed, I face memory allocation issues after enabling it from the menu config.

First, I got the following error:
```
I (26719) MQTT: MQTT_EVENT_BEFORE_CONNECT
E (29299) esp-aes: Failed to allocate memory
E (29299) esp-tls-mbedtls: mbedtls_ctr_drbg_seed returned -0x0001
E (29299) esp-tls: create_ssl_handle failed
E (29299) esp-tls: Failed to open new connection
E (29309) TRANSPORT_BASE: Failed to open a new connection
E (29309) mqtt_client: Error transport connect
E (29319) MQTT: MQTT_EVENT_ERROR
E (29319) MQTT: Last error code reported from esp-tls: 0x8011: ESP_ERR_MBEDTLS_CTR_DRBG_SEED_FAILED
E (29329) MQTT: Last tls stack error number: 0x1: ERROR
E (29339) MQTT: Last captured errno : 0 (Success)
I (29349) MQTT: MQTT_EVENT_DISCONNECTED
```

After moving manually multiple large variables (around 32KB) into the external .bss via the EXT_RAM_BSS_ATTR statement, I got rid of the error above but now randomly get the following error (not every time)

```
I (16709) MQTT: MQTT_EVENT_BEFORE_CONNECT
E (22439) esp-tls-mbedtls: mbedtls_ssl_setup returned -0x7F00
E (22439) esp-tls: create_ssl_handle failed
E (22439) esp-tls: Failed to open new connection
E (22439) TRANSPORT_BASE: Failed to open a new connection
E (22449) mqtt_client: Error transport connect
E (22449) MQTT: MQTT_EVENT_ERROR
E (22459) MQTT: Last error code reported from esp-tls: 0x8017: ESP_ERR_MBEDTLS_SSL_SETUP_FAILED
E (22469) MQTT: Last tls stack error number: 0x7f00: ERROR
E (22469) MQTT: Last captured errno : 0 (Success)
I (22479) MQTT: MQTT_EVENT_DISCONNECTED
```

Menuconfig as follows:
[image]

Note, disabling the support for external SPI RAM in the Menuconfig again, leads to no runtime / memory errors.

### Steps to reproduce.

Use external SPI RAM with mbedTLS.

mahavirj, 2022-10-10:
@ddomnik 

Can you please try enabling following config option:

https://github.com/espressif/esp-idf/blob/1308ff3d211bd039995203e55a147b3329a87589/components/mbedtls/Kconfig#L31

Also please share complete console log if you still run into any issues

ddomnik, 2022-10-10:
> @ddomnik
> 
> Can you please try enabling following config option:
> 
> https://github.com/espressif/esp-idf/blob/1308ff3d211bd039995203e55a147b3329a87589/components/mbedtls/Kconfig#L31
> 
> Also please share complete console log if you still run into any issues

Thanks, it worked.

closed 2022-10-10 by ddomnik

HiteshHonmane, 2025-03-05:
I faced  same issue 
shifted
Memory allocation strategy (extern psram)
and it worked.