espressif/esp-idf #9936: [v4.4.2] NimBLE/Bluetooth not releasing as RAM than previous versions

sikpigs, 2022-10-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### Development Kit.

HUZZAH32

### Power Supply used.

USB

### What is the expected behavior?

The NimBLE stack to release at least the same amount of RAM to the heap as older versions when using esp_bt_mem_release(ESP_BT_MODE_BTDM).

### What is the actual behavior?

The NimBLE stack in ESP-IDF 4.4.2 doesn't release as much RAM as it does using 4.0.1. I would expect later versions to release at least the same amount of RAM. Using a simple test the NimBLE stack in 4.4.2 doesn't release as much memory as it does in 4.0.1

### Steps to reproduce.

Use the modified nimble perh. example attached using IDF versions 4.0.1 and 4.4.2.

[nimble_heap.tar.gz](https://github.com/espressif/esp-idf/files/9737739/nimble_heap.tar.gz)

### Debug Logs.

```plain
4.0.1
-----
I (76) boot: Chip Revision: 3
I (76) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (42) boot: ESP-IDF v4.0.1 2nd stage bootloader
I (42) boot: compile time 18:58:27
I (42) boot: Enabling RNG early entropy source...
I (46) boot: SPI Speed      : 40MHz
I (51) boot: SPI Mode       : DIO
I (54) boot: SPI Flash Size : 4MB
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) boot_comm: chip revision: 3, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0cbac ( 52140) map
I (131) esp_image: segment 1: paddr=0x0001cbd4 vaddr=0x3ffbdb60 size=0x032a4 ( 12964) load
I (136) esp_image: segment 2: paddr=0x0001fe80 vaddr=0x40080000 size=0x00190 (   400) load
0x40080000: _WindowOverflow4 at /home/sikpigs/esp/4.0.1/esp-idf/components/freertos/xtensa_vectors.S:1778

I (138) esp_image: segment 3: paddr=0x00020018 vaddr=0x400d0018 size=0x4ba08 (309768) map
0x400d0018: _stext at ??:?

I (257) esp_image: segment 4: paddr=0x0006ba28 vaddr=0x40080190 size=0x00270 (   624) load
0x40080190: _Level2Vector at ??:?

I (258) esp_image: segment 5: paddr=0x0006bca0 vaddr=0x40080400 size=0x12c14 ( 76820) load
I (308) boot: Loaded app from partition at offset 0x10000
I (308) boot: Disabling RNG early entropy source...
I (308) cpu_start: Pro cpu up.
I (312) cpu_start: Application information:
I (317) cpu_start: Project name:     nimble_heap
I (322) cpu_start: App version:      1
I (327) cpu_start: Compile time:     Oct  7 2022 18:58:22
I (333) cpu_start: ELF file SHA256:  4d6f81f6a012db64...
I (339) cpu_start: ESP-IDF:          v4.0.1
I (344) cpu_start: Starting app cpu, entry point is 0x400812b0
0x400812b0: call_start_cpu1 at /home/sikpigs/esp/4.0.1/esp-idf/components/esp32/cpu_start.c:271

I (0) cpu_start: App cpu up.
I (354) heap_init: Initializing. RAM available for dynamic allocation:
I (361) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (367) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (373) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (379) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (385) heap_init: At 3FFC39A8 len 0001C658 (113 KiB): DRAM
I (391) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (398) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (404) heap_init: At 40093014 len 0000CFEC (51 KiB): IRAM
I (410) cpu_start: Pro cpu start user code
I (429) spi_flash: detected chip: generic
I (429) spi_flash: flash io: dio
I (430) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (479) NimBLE_BLE_PRPH: Initial Heap: 244436
I (479) BTDM_INIT: BT controller compile version [c1cbe45]
I (479) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (589) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
GAP procedure initiated: stop advertising.
error enabling advertisement; rc=30
I (889) NimBLE_BLE_PRPH: Current Heap: 291028

4.4.2
-----
I (29) boot: ESP-IDF v4.4.2 2nd stage bootloader
I (29) boot: compile time 19:03:47
I (29) boot: chip revision: 3
I (32) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (96) boot_comm: chip revision: 3, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0d374h ( 54132) map
I (131) esp_image: segment 1: paddr=0001d39c vaddr=3ffbdb60 size=02c7ch ( 11388) load
I (136) esp_image: segment 2: paddr=00020020 vaddr=400d0020 size=526d8h (337624) map
I (259) esp_image: segment 3: paddr=00072700 vaddr=3ffc07dc size=01a18h (  6680) load
I (262) esp_image: segment 4: paddr=00074120 vaddr=40080000 size=16f24h ( 93988) load
I (303) esp_image: segment 5: paddr=0008b04c vaddr=50000000 size=00010h (    16) load
I (315) boot: Loaded app from partition at offset 0x10000
I (315) boot: Disabling RNG early entropy source...
I (326) cpu_start: Pro cpu up.
I (327) cpu_start: Starting app cpu, entry point is 0x400811b0
0x400811b0: call_start_cpu1 at /home/sikpigs/esp/4.4.2/esp-idf/components/esp_system/port/cpu_start.c:160

I (0) cpu_start: App cpu up.
I (341) cpu_start: Pro cpu start user code
I (341) cpu_start: cpu freq: 160000000
I (341) cpu_start: Application information:
I (345) cpu_start: Project name:     nimble_heap
I (351) cpu_start: App version:      1
I (355) cpu_start: Compile time:     Oct  7 2022 19:03:39
I (361) cpu_start: ELF file SHA256:  f7de98aff33f6549...
I (367) cpu_start: ESP-IDF:          v4.4.2
I (372) heap_init: Initializing. RAM available for dynamic allocation:
I (379) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (385) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (391) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (398) heap_init: At 3FFC4EF8 len 0001B108 (108 KiB): DRAM
I (404) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (410) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (417) heap_init: At 40096F24 len 000090DC (36 KiB): IRAM
I (424) spi_flash: detected chip: generic
I (428) spi_flash: flash io: dio
I (433) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (501) NimBLE_BLE_PRPH: Initial Heap: 234032
I (501) BTDM_INIT: BT controller compile version [bf9c077]
I (501) system_api: Base MAC address is not set
I (511) system_api: read default base MAC address from EFUSE
I (511) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
W (521) phy_init: failed to load RF calibration data (0xffffffff), falling back to full calibration
I (941) NimBLE: GAP procedure initiated: stop advertising.

E (981) NimBLE: error setting advertisement data; rc=30

W (991) Timer: Timer not stopped
W (991) Timer: Timer not stopped
W (991) Timer: Timer not stopped
I (991) NimBLE_BLE_PRPH: Current Heap: 268392
```

### More Information.

In my application I actually see a difference of about 25k-30k less memory released to heap from when using 4.4.2 compared to 4.0.1. This change is affecting our ability to upgrade to 4.4.2.

rahult-github, 2022-10-12:
Hi @sikpigs ,

Between v4.0 and v4.4 , there are multiple changes. The nimble branch version moved from nimble-1.0 to nimble-1.3 which implies more features and hence more data keeping information.  Also, there are controller level changes done. So, i think, it is not straight forward comparison for same amount of memory utilisation.  For example, there are more functions in IRAM now that would be making the controller code faster and hence i think will affect the numbers. 
 
However, i think, if across multiple init / deinit, in case , you are observing any memory leak, then that should be a problem that needs to be debugged. 

One suggestion i could provide is check if any feature ( that is added in v4.4 and was not used in your use case while using v4.0) can be disabled to see if it helps. 

Thanks,
 Rahul

sikpigs, 2022-10-12:
Thanks for your reply @rahult-github.

I do see an increase in IRAM usage, and that's okay for now. I'm concern with DRAM usage at the moment. I do have a couple of other data points to add.

In our application, when using 4.4.2 and invoking the de-init's and esp_bt_mem_release we have about 45kB of DRAM free. This same application runs with about 70kB of DRAM free after de-initing and esp_bt_mem_release is invoked when using 4.0.1. I have to disable NimBLE altogether in menuconfig to get back to about 70kB of DRAM free when using 4.4.2.

I ran the above attached test program with IDF versions 4.1.3, 4.2.4, and 4.3.4 and found these differences in DRAM after de-initializing and calling esp_bt_mem_release:
4.1.3: 46 960 bytes (close to 4.0.1's difference of 46 592 bytes)
4.2.4: 46 948 bytes
4.3.4: 37 104 bytes (closer to 4.4.2's difference of 34 360 bytes)

I can appreciate some level of approximation in the numbers from esp_get_free_heap_size() but I see a rather large amount of DRAM that I can only tell has been allocated/reserved by NimBLE or the controller that's not being made available for use. Is there anything that can be done to reclaim this memory?

Please let me know if I can get you more information.

Thanks again

rahult-github, 2024-09-10:
Release v4/4 is EOL. Closing issue.

closed 2024-09-10 by rahult-github