espressif/esp-idf #9937: Not able to perform WiFi Provisioning Manager with SoftAP and BLE at the same time

lawrenceko-intecular, 2022-10-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,

We are running the example [wifi_prov_mgr](https://github.com/espressif/esp-idf/tree/master/examples/provisioning/wifi_prov_mgr) and attempting to run softAP and BLE provisioning mode at the same time, but seems like it is limited by the [`scheme`](https://github.com/espressif/esp-idf/blob/02605f1a31be4233b97e83a0460bb458f277dfea/examples/provisioning/wifi_prov_mgr/main/app_main.c#L277) and [`scheme_event_handler`](https://github.com/espressif/esp-idf/blob/02605f1a31be4233b97e83a0460bb458f277dfea/examples/provisioning/wifi_prov_mgr/main/app_main.c#L292). 

Can softAP and BLE be running at the same time at idle and waiting to see which provisioning method is first being used then continue the pairing process?

Thank you.

kayzen9, 2022-10-09:
I do not think it is supported. Provisioning manager state machine is tied to specific transport that is being selected during its initialization. It could either be `SoftAP+HTTP` or `BLE`.

Reference: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/provisioning/provisioning.html#deciding-on-transport

AdityaHPatwardhan, 2022-10-10:
Hi @lawrenceko-intecular I think you will have to modify the `wifi_prov_mgr` example for provisioning with BLE and softAP simultaneously.

Have you tried creating two separate tasks that call `wifi_prov_mgr_start_provisioning` with different `scheme` ? I think the required support to make simultaneous provisioning through both modes work would be  a part of the application layer itself. The `wifi_provisioning` component should not require any changes.

closed 2022-10-18 by lawrenceko-intecular