espressif/esp-idf #9938: [v4.4.2] [Build] _Static_assert(sizeof(esp_chip_id_t) == 2, "esp_chip_id_t should be 16 bit");

chipweinberger, 2022-10-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2-378-g9269a536ac

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### What is the expected behavior?

I'm using 4.4.2, top of branch, stock, it should build.

### What is the actual behavior?

I should have brought up this issue a year ago. I've never taken the time to figure out what causes it,

For almost a year now, on `v4.3.x`, `v4.4.x`, I've always had to comment out these 3 lines in `esp_app_format.h` because of a build error.

```
_Static_assert(sizeof(esp_chip_id_t) == 2, "esp_chip_id_t should be 16 bit");
_Static_assert(sizeof(esp_image_header_t) == 24, "binary image header should be 24 bytes");
_Static_assert(sizeof(esp_app_desc_t) == 256, "esp_app_desc_t should be 256 bytes");
```

I'm not sure what it does not like about it.

Log:
```
/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bootloader_support/include/esp_app_format.h:24:15: error: expected constructor, destructor, or type conversion before '(' token
 _Static_assert(sizeof(esp_chip_id_t) == 2, "esp_chip_id_t should be 16 bit");
               ^
/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bootloader_support/include/esp_app_format.h:91:15: error: expected constructor, destructor, or type conversion before '(' token
 _Static_assert(sizeof(esp_image_header_t) == 24, "binary image header should be 24 bytes");
               ^
/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bootloader_support/include/esp_app_format.h:124:15: error: expected constructor, destructor, or type conversion before '(' token
 _Static_assert(sizeof(esp_app_desc_t) == 256, "esp_app_desc_t should be 256 bytes");
               ^
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

### Steps to reproduce.

No special steps. Building with idf.py.

kayzen9, 2022-10-09:
@chipweinberger 

Missing CPP guards in following header (?)

https://github.com/espressif/esp-idf/blob/9269a536ac123677b165ff531369dfd1338ef103/components/bootloader_support/include/esp_app_format.h#L24

For C++, it should be `static_assert`

chipweinberger, 2022-10-09:
My code is all CPP, so that's a good theory.

Yes the capitalization is a bit weird.

chipweinberger, 2022-11-19:
This is still broken on v4.4 branch (maybe master too)?

 I'd open a PR, but I think I should get the other PR's in first, hah.

o-marshmallow, 2022-11-21:
Hi @chipweinberger ,

An internal PR on-going to fix this issue, it will be backported to `v4.4` as soon as it is merged

chipweinberger, 2022-11-21:
appreciate the update! :)

lamaanrana-eaton, 2024-03-01:
I am using IDF v4.4.0, also tried V4.4.5 but I am getting this error while building the code. I am not sure what to do here or how to get around it. it is only happening with one repo. I can compile all the projects from the examples in the ESP-IDF

0xjakob, 2024-03-05:
> I am using IDF v4.4.0, also tried V4.4.5 but I am getting this error while building the code. I am not sure what to do here or how to get around it. it is only happening with one repo. I can compile all the projects from the examples in the ESP-IDF

@lamaanrana-eaton Do you have a minimal reproducing example?