espressif/esp-idf #9940: esp_hid_host trigger watchdog

iron-LONG1, 2022-10-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I burned the esp_hid_host example to the development esp32_devkit_V4 board, turned on the PS4 handle, and made a Bluetooth connection. After the connection was successful for dozens of seconds, the watchdog was triggered. I could not find the corresponding triggering reason, and I hope to get help

boblane1, 2022-10-09:
Hi @iron-LONG1,

Thanks for reporting! Have you ever modified the code in esp_hid_host example, especially the function `hidh_callback`? The delay in `hidh_callback` may cause task watch dog(e.g. Printing too much data). Would you mind providing more logs?

iron-LONG1, 2022-10-09:
Thank you for your suggestion. It should be that the source program printed too much data to trigger the task watchdog. After commenting out the program in the picture, it did not trigger it again.
[image]

closed 2022-10-10 by Alvin1Zhang