espressif/esp-idf #9941: Status: Requesting Composition Data gets an error when using ble-mesh/on-off-server example out of the box

gomez9656, 2022-10-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-C3-DevKitM1-v1.0

### Power Supply used.

USB

### What is the expected behavior?

When using ESP-BLE-MESH/on-off-server example, with two or more devices, the app should be able to get the Composition Data successfully for the 2nd device

### What is the actual behavior?

The Status: Requesting Composition Data for the 2nd device finishes into Error: request timed out

### Steps to reproduce.

1. Build ESP-BLE-MESH/on-off-server example
2. Flash two or more devices
3. Use the iOS nRF Mesh app to provision the devices
4. when provisioning the second device, the error appears
...

### Debug Logs.

```plain
1st device provisioned (this one works)
I (30) boot: ESP-IDF v4.4-dirty 2nd stage bootloader
I (30) boot: compile time 14:07:49
I (30) boot: chip revision: 3
I (32) boot.esp32c3: SPI Speed      : 80MHz
I (37) boot.esp32c3: SPI Mode       : DIO
I (42) boot.esp32c3: SPI Flash Size : 2MB
I (46) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (70) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (78) boot:  2 factory          factory app      00 00 00010000 00100000
I (85) boot: End of partition table
I (89) esp_image: segment 0: paddr=00010020 vaddr=3c0c0020 size=2a840h (174144) map
I (125) esp_image: segment 1: paddr=0003a868 vaddr=3fc8e400 size=0305ch ( 12380) load
I (127) esp_image: segment 2: paddr=0003d8cc vaddr=40380000 size=0274ch ( 10060) load
I (133) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=b218ch (729484) map
I (253) esp_image: segment 4: paddr=000f21b4 vaddr=4038274c size=0bc84h ( 48260) load
I (262) esp_image: segment 5: paddr=000fde40 vaddr=50000010 size=00010h (    16) load
I (267) boot: Loaded app from partition at offset 0x10000
[... 91 lines cut ...]
I (433) phy_init: phy_version 907,3369105-dirty,Dec  3 2021,14:55:12
W (443) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
I (493) system_api: Base MAC address is not set
I (493) system_api: read default base MAC address from EFUSE
I (493) BTDM_INIT: Bluetooth MAC: 84:f7:03:66:a4:76

I (593) EXAMPLE: ESP_BLE_MESH_PROV_REGISTER_COMP_EVT, err_code 0
I (593) EXAMPLE: ESP_BLE_MESH_NODE_PROV_ENABLE_COMP_EVT, err_code 0
I (593) EXAMPLE: BLE Mesh Node initialized
I (18863) EXAMPLE: ESP_BLE_MESH_NODE_PROV_LINK_OPEN_EVT, bearer PB-GATT
W (19193) BLE_MESH: No Health Server context provided
W (21083) BLE_MESH: No Health Server context provided
I (21093) EXAMPLE: ESP_BLE_MESH_NODE_PROV_LINK_CLOSE_EVT, bearer PB-GATT
I (21103) EXAMPLE: ESP_BLE_MESH_NODE_PROV_COMPLETE_EVT
I (21103) EXAMPLE: net_idx: 0x0000, addr: 0x01e1
I (21103) EXAMPLE: flags: 0x00, iv_index: 0x00000000
```

### More Information.

I want to be able to provision several devices with the same app. As far as I know, this should be possible using the example out of the box.

https://www.youtube.com/watch?v=cgqMMck3OSU You can see in 0:35 that the 2nd device is provisioned successfully, so I expect this to happen as well on my setup.

Thanks a lot in advance :)

forx157, 2022-10-09:
@gomez9656  hi, you can try to open `Automatic Connection` option in the App's Proxy tab before provisioning the node.

gomez9656, 2022-10-09:
@forx157 Hi there, thanks for the quick reply. 

That doesn't work :(. 

My goal is to create a group and control both devices. If I always need to disconnect the proxy and reconnect the other proxy, then I'm not going to be able to do it. 

Is there any way to do it as in the youtube example?

Thanks

forx157, 2022-10-10:
@gomez9656 hi, I've tested all the steps in the video, and two nodes all runs normally, and just need to keep `Automatic Connection` on. You can try it again.

gomez9656, 2022-10-10:
Hi @forx157 , I have tested this process a hundred times since 2 months ago, but it has never worked. I'm just posting it now. 

I have used an iPhone 11 and a Huawei P20 lite, but none of them works. 

I added some SS for you to see the dialog box on iPhone
[image]
[image]
[image]

I also tested it with a battery, didn't work. 

Do I need to test it with another OS? Another phone? Another branch? 

Thanks a lot in advance

forx157, 2022-10-11:
@gomez9656  hi,I have used an iPhone 11 to test again, and it also works well, here is my test video : [Test Video](https://drive.google.com/file/d/1WwqnGLoug7NC3iyBxQHQvzGLK1u2e0cB/view?usp=sharing).  And the version of nrf Mesh APP is 3.2.0, IOS version is 15.5, Is the software version the same? And have you changed the code or somthing is different from the default setting? Or any other tips could help me to reproduce the problem.

gomez9656, 2022-10-13:
Hi @forx157, I decided to install the ESP-IDF on a brand-new computer(Windows 11) but that didn't make the trick. 

I also tested it with an iPhone 13 Pro Max(IOS 15.6) and with a Samsung. The iPhone didn't work. Samsung phone was able to get the composition data, but once I wanted to add the application key, it didn't work. 

As I suspect, the problem is coming from the build or menuconfig side. The only thing I did with the new installation was to go to the example folder and type the commands:
1) idf.py fullclean
2) idf.py set-target esp32-c3 
3)idf.py build
4) idf.py -p comX flash monitor

I did those steps for both devices while they were plugged in via USB.

My version of the nrf Mesh app is the same. 

I also did a lot of esptool.py erase_flash in the past when I was developing an OTA feature, maybe this caused some bugs in the board?

Thanks a lot for the help so far

forx157, 2022-10-14:
@gomez9656 Hi, could you change the log print level and provide that log to me? 
you can type `idf.py menuconfig` command and follow the path of `Component config → ESP BLE Mesh Support → BLE Mesh STACK DEBUG LOG LEVEL` and `Component config → ESP BLE Mesh Support → BLE Mesh NET BUF DEBUG LOG LEVEL`, to change the log level from `WARNING` to `INFO`.
In addition, the function ` dev_comp_data_get` in the file `components/bt/esp_ble_mesh/mesh_core/cfg_srv.c`  could change to like this:
```
static void dev_comp_data_get(struct bt_mesh_model *model,
                              struct bt_mesh_msg_ctx *ctx,
                              struct net_buf_simple *buf)
{
    struct net_buf_simple *sdu = NULL;
    uint8_t page = 0U;

    BT_INFO("Comp data send net_idx 0x%04x app_idx 0x%04x src 0x%04x len %u: %s",
           ctx->net_idx, ctx->app_idx, ctx->addr, buf->len,
           bt_hex(buf->data, buf->len)); //changed line

    page = net_buf_simple_pull_u8(buf);
```
also the function ` proxy_connected` and `proxy_disconnected` in the file `components/bt/esp_ble_mesh/mesh_core/proxy_server.c` could change to like to this:
```
static void proxy_connected(struct bt_mesh_conn *conn, uint8_t err)
{
    struct bt_mesh_proxy_client *client = NULL;
    int i;

    BT_INFO("conn %p err 0x%02x", conn, err);  //changed line from BT_DBG to BT_INFO
```
```
static void proxy_disconnected(struct bt_mesh_conn *conn, uint8_t reason)
{
    int i;

    BT_INFO("conn %p reason 0x%02x", conn, reason); //changed line from BT_DBG to BT_INFO
```

gomez9656, 2022-10-17:
Hey @forx157, I decided to reinstall the ESP-IDF in an entirely new user account on my PC, to give it a try.

It didn't work. I added the changes you requested and here are the results:

Logs from the 1st device(this one is provisioned successfully)

I (30) boot: ESP-IDF v4.4.2-dirty 2nd stage bootloader
I (30) boot: compile time 09:41:35
I (30) boot: chip revision: 3
I (32) boot.esp32c3: SPI Speed      : 80MHz
I (37) boot.esp32c3: SPI Mode       : DIO
I (42) boot.esp32c3: SPI Flash Size : 2MB
I (47) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (70) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (78) boot:  2 factory          factory app      00 00 00010000 00100000
I (85) boot: End of partition table
I (89) esp_image: segment 0: paddr=00010020 vaddr=3c0c0020 size=2b188h (176520) map
I (125) esp_image: segment 1: paddr=0003b1b0 vaddr=3fc8ea00 size=02d30h ( 11568) load
I (128) esp_image: segment 2: paddr=0003dee8 vaddr=40380000 size=02130h (  8496) load
I (133) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=b3214h (733716) map
I (254) esp_image: segment 4: paddr=000f323c vaddr=40382130 size=0c750h ( 51024) load
I (264) esp_image: segment 5: paddr=000ff994 vaddr=50000010 size=00010h (    16) load
I (269) boot: Loaded app from partition at offset 0x10000
I (269) boot: Disabling RNG early entropy source...
I (284) cpu_start: Pro cpu up.
I (293) cpu_start: Pro cpu start user code
I (293) cpu_start: cpu freq: 160000000
I (293) cpu_start: Application information:
I (296) cpu_start: Project name:     onoff_server
I (301) cpu_start: App version:      v4.4.2-dirty
I (306) cpu_start: Compile time:     Oct 17 2022 09:40:41
I (313) cpu_start: ELF file SHA256:  4ed72cf1a8250338...
I (319) cpu_start: ESP-IDF:          v4.4.2-dirty
I (324) heap_init: Initializing. RAM available for dynamic allocation:
I (331) heap_init: At 3FC98850 len 000277B0 (157 KiB): DRAM
I (337) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (344) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (351) spi_flash: detected chip: generic
I (355) spi_flash: flash io: dio
W (359) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (373) sleep: Configure to isolate all GPIO pins in sleep state
I (379) sleep: Enable automatic switching of GPIO sleep configuration
I (386) coexist: coexist rom version 9387209
I (391) cpu_start: Starting scheduler.
I (396) EXAMPLE: Initializing...
I (396) gpio: GPIO[8]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (406) gpio: GPIO[8]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (416) gpio: GPIO[8]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
W (426) BTDM_INIT: esp_bt_controller_mem_release not implemented, return OK
I (426) BTDM_INIT: BT controller compile version [05195c9]
I (436) phy_init: phy_version 912,d001756,Jun  2 2022,16:28:07
I (476) system_api: Base MAC address is not set
I (476) system_api: read default base MAC address from EFUSE
I (476) BTDM_INIT: Bluetooth MAC: 84:f7:03:68:13:d6

I (556) EXAMPLE: ESP_BLE_MESH_PROV_REGISTER_COMP_EVT, err_code 0
I (556) EXAMPLE: ESP_BLE_MESH_NODE_PROV_ENABLE_COMP_EVT, err_code 0
I (566) EXAMPLE: BLE Mesh Node initialized
I (4426) BLE_MESH: conn 0x3fc94608 err 0x00
I (5566) EXAMPLE: ESP_BLE_MESH_NODE_PROV_LINK_OPEN_EVT, bearer PB-GATT
W (5896) BLE_MESH: No Health Server context provided
I (7996) BLE_MESH: Algorithm:   0x00
I (7996) BLE_MESH: Public Key:  0x00
I (7996) BLE_MESH: Auth Method: 0x00
I (7996) BLE_MESH: Auth Action: 0x00
I (7996) BLE_MESH: Auth Size:   0x00
I (8326) BLE_MESH: Primary Element: 0x0042
[... 103 lines cut ...]
I (54565) BLE_MESH: Primary address 0x0045, element count 3
I (54575) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (54585) BLE_MESH: Scan is already started
I (54585) EXAMPLE: ESP_BLE_MESH_NODE_PROV_LINK_CLOSE_EVT, bearer PB-GATT
I (54595) EXAMPLE: ESP_BLE_MESH_NODE_PROV_COMPLETE_EVT
I (54595) EXAMPLE: net_idx: 0x0000, addr: 0x0045
I (54605) EXAMPLE: flags: 0x00, iv_index: 0x00000000
W (54645) BT_HCI: hcif disc complete: hdl 0x2, rsn 0x13
I (54645) BLE_MESH: conn 0x3fc94608 reason 0x13
I (64575) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (74585) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (84585) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (94595) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (104595) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (114605) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (124605) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (134605) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (144615) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (154615) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (164625) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (174625) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (184635) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (194635) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00

Thanks in advance!

forx157, 2022-10-18:
Hey @gomez9656 ,could you apply the following patch to your progrom to help me get more infomation. I guess something went wrong there with the first node forwarding the `comp data get` msg to the second node. thanks :).
[001-for-test.zip](https://github.com/espressif/esp-idf/files/9810319/001-for-test.zip)

gomez9656, 2022-10-18:
Hi @forx157 , long time ago I changed the the TTL to 3 and it worked(just once), maybe it was pure random?

I didn't know how to change the patch using git, so I copy the lines directly. I added the files so you can double-check.

[cfg_srv.zip](https://github.com/espressif/esp-idf/files/9814255/cfg_srv.zip)

Logs from 1st device

I (30) boot: ESP-IDF v4.4.2-dirty 2nd stage bootloader
I (30) boot: compile time 09:41:35
I (30) boot: chip revision: 3
I (32) boot.esp32c3: SPI Speed      : 80MHz
I (37) boot.esp32c3: SPI Mode       : DIO
I (42) boot.esp32c3: SPI Flash Size : 2MB
I (47) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (70) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (78) boot:  2 factory          factory app      00 00 00010000 00100000
I (85) boot: End of partition table
I (89) esp_image: segment 0: paddr=00010020 vaddr=3c0c0020 size=2b448h (177224) map
I (126) esp_image: segment 1: paddr=0003b470 vaddr=3fc8ea00 size=02d30h ( 11568) load
I (128) esp_image: segment 2: paddr=0003e1a8 vaddr=40380000 size=01e70h (  7792) load
I (133) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=b350ch (734476) map
I (255) esp_image: segment 4: paddr=000f3534 vaddr=40381e70 size=0ca10h ( 51728) load
I (264) esp_image: segment 5: paddr=000fff4c vaddr=50000010 size=00010h (    16) load
I (269) boot: Loaded app from partition at offset 0x10000
I (269) boot: Disabling RNG early entropy source...
I (285) cpu_start: Pro cpu up.
I (293) cpu_start: Pro cpu start user code
I (293) cpu_start: cpu freq: 160000000
I (293) cpu_start: Application information:
I (295) cpu_start: Project name:     onoff_server
I (301) cpu_start: App version:      v4.4.2-dirty
I (306) cpu_start: Compile time:     Oct 17 2022 09:40:41
I (312) cpu_start: ELF file SHA256:  297357f7007f661f...
I (318) cpu_start: ESP-IDF:          v4.4.2-dirty
I (324) heap_init: Initializing. RAM available for dynamic allocation:
I (331) heap_init: At 3FC98850 len 000277B0 (157 KiB): DRAM
I (337) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (344) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (351) spi_flash: detected chip: generic
I (355) spi_flash: flash io: dio
W (359) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (372) sleep: Configure to isolate all GPIO pins in sleep state
I (379) sleep: Enable automatic switching of GPIO sleep configuration
I (386) coexist: coexist rom version 9387209
I (391) cpu_start: Starting scheduler.
I (395) EXAMPLE: Initializing...
I (395) gpio: GPIO[8]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (405) gpio: GPIO[8]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (415) gpio: GPIO[8]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
W (425) BTDM_INIT: esp_bt_controller_mem_release not implemented, return OK
I (425) BTDM_INIT: BT controller compile version [05195c9]
I (435) phy_init: phy_version 912,d001756,Jun  2 2022,16:28:07
I (465) system_api: Base MAC address is not set
I (465) system_api: read default base MAC address from EFUSE
I (465) BTDM_INIT: Bluetooth MAC: 84:f7:03:67:da:b2

I (495) BLE_MESH: relay init 0
I (555) EXAMPLE: ESP_BLE_MESH_PROV_REGISTER_COMP_EVT, err_code 0
I (555) EXAMPLE: ESP_BLE_MESH_NODE_PROV_ENABLE_COMP_EVT, err_code 0
I (555) EXAMPLE: BLE Mesh Node initialized
I (24315) BLE_MESH: conn 0x3fc94608 err 0x00
W (24315) BLE_MESH: conn 0 err 0x00
I (25305) EXAMPLE: ESP_BLE_MESH_NODE_PROV_LINK_OPEN_EVT, bearer PB-GATT
W (25635) BLE_MESH: No Health Server context provided
I (27405) BLE_MESH: Algorithm:   0x00
I (27405) BLE_MESH: Public Key:  0x00
[... 199 lines cut ...]
I (51746) EXAMPLE: net_idx: 0x0000, addr: 0x0051
I (51756) EXAMPLE: flags: 0x00, iv_index: 0x00000000
W (51796) BT_HCI: hcif disc complete: hdl 0x2, rsn 0x13
I (51796) BLE_MESH: conn 0x3fc94608 reason 0x13
W (51796) BLE_MESH: conn 0 reason 0x13
I (61736) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (71736) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (81746) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (91746) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (101746) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (111756) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (121756) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (131766) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (141766) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (151776) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (161776) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (171776) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (181786) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (191786) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (201796) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (211796) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (221806) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (231806) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00

Thanks again :)

gomez9656, 2022-10-21:
Hi @forx157 , have you been able to find a clue about the problem? 

If you have any direction you would like me to test, I will do it for sure

forx157, 2022-10-24:
hi, @gomez9656 , sorry for replay so late, could you disable the option with the following path ` Component config → Bluetooth → Bluetooth → Bluetooth controller`? And provide the log to me. Inaddition, do you have the sniffer of bluetooth, that data will help me a lot.

gomez9656, 2022-10-24:
Hi @forx157 , if I disable the Bluetooth controller then I get an error when building. 

[image]

I do not have the Bluetooth sniffer. What's the name? to check if I can buy it.

forx157, 2022-10-25:
hi,@gomez9656, so sorry for that,  i forget to say the suboption: BLE Scan Duplicate Options in the Bluetooth controller, you need to disable the suboption not the Bluetooth controller. And If you don't have that sniffer,just forget it :)

AntoineSX, 2022-10-25:
I'm having the same issue on ESP32-WROVER using the nRF iOS app. Tried on v4.3.3 and v.4.4.2 and got the same issue. Tried disabling BLE Scan Duplicate Options and it didn't fix the issue for me. 

If I power off the first ESP after the second one hangs at "Requesting Composition Data" I can successfully get passed that step, but obviously I lose the configuration of the first device which doesn't help. 

I figured it happened since the proxy switched to that new device. So then I tried to configure the mesh with Automatic Connection off, so by manually switching the proxy right after provisioning the second device. This lets me configure both devices but then I can only interact with the OnOff server of the proxy and not the other device, even if I switch the proxy again or turn Automatic Connection back on. 

I changed the GATT layer debug level to verbose and I see the following when the "Requesting Composition Data" prompt appears, until it fails: 
``` 
I (84020) BT_GATT: GATT_GetConnectionInfor conn_id=3
I (84050) BT_GATT: GATT_GetConnectionInfor conn_id=3
I (88000) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (88640) BT_GATT: GATT_GetConnectionInfor conn_id=3
I (88640) BT_GATT: GATT_GetConnectionInfor conn_id=3
I (97640) BT_GATT: GATT_GetConnectionInfor conn_id=3
I (97640) BT_GATT: GATT_GetConnectionInfor conn_id=3
I (98010) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
```

I have little to no background with Bluetooth Mesh, but it seems like the second device is not discoverable by the Proxy.

forx157, 2022-10-26:
hi, @gomez9656 @AntoineSX ,i want to reproduce that problem, so could you upload your project to github or any other ways which could share your project with me. Thanks a lot.

AntoineSX, 2022-10-26:
Hi @forx157 I used the onoff_server code as is. Luckily I stumbled across the source of my problem. My 2 boards were too far apart. My phone could see both boards,  but the two boards were unable to see each other. So after provisioning both boards, the composition data could not be relayed to the second board through the proxy since it could not see it.

I tested this by varying the distance between both boards and also by changing the tx power level without moving the boards using: 
```C++
esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_DEFAULT, ESP_PWR_LVL_XX);
esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_ADV, ESP_PWR_LVL_XX);
esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_SCAN, ESP_PWR_LVL_XX);
```
I was able to reproduce the observed behavior by doing so. I believe the issue is solely caused by a weak Bluetooth transmission power on the ESP32 boards I used.

gomez9656, 2022-10-26:
Hi @forx157 , I also used the onoff_server as it is. Nevertheless, I uploaded the ESP-IDF framework with the examples in my repo: https://github.com/gomez9656/Espressif-test/tree/master/examples/bluetooth/esp_ble_mesh/ble_mesh_node/onoff_server

It is the same as yours since I haven't changed anything.

@AntoineSX could you please tell me in which file did you change those TX functions? I would love to try that solution :)

forx157, 2022-10-27:
hi, @gomez9656 
I have tested with your project, but it still works well. Combined with @AntoineSX ‘s experience and the last log information(that log shows the first node forwarded the `composition data get` msg to the second node successfully but the second node didn't receive), I guess if that nodes on your side are also far apart or there are some obstacles that hinder the propagation of the signal, And how much nodes in your test net? could you provide me with your test video, so that I can replicate your test environment.

By the way, `esp_ble_tx_power_set` in the file `components\bt\controller\esp32c3\bt.c : 1367`. This function could change your transmitting power.

Thanks a lot.

gomez9656, 2022-10-27:
Hi @forx157 , the problem was indeed having both devices too far away. 

If I test the devices as in the first picture, everything goes fine and I can send messages to both devices at the same time without changing the proxy.

[image]

Now, if the devices are 3cm + away from each other like in the second picture, then only one of the devices receives the message(I suppose the proxy)

[image]

I added these functions after ble_mesh_init() but they don't seem to change anything. 
`    esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_DEFAULT, ESP_PWR_LVL_P9); 
    esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_ADV, ESP_PWR_LVL_P9);
    esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_SCAN ,ESP_PWR_LVL_P9);

    int pwrAdv  = esp_ble_tx_power_get(ESP_BLE_PWR_TYPE_ADV);
    ESP_LOGI(TAG, "%d", pwrAdv);
    int pwrScan = esp_ble_tx_power_get(ESP_BLE_PWR_TYPE_SCAN);
    ESP_LOGI(TAG, "%d, pwrScan);
    int pwrDef  = esp_ble_tx_power_get(ESP_BLE_PWR_TYPE_DEFAULT);
    ESP_LOGI(TAG, "%d", pwrDef);`

The log power is 12 for all of them. Should they go after ble_mesh_init() or somewhere else?

forx157, 2022-10-28:
Hi, @gomez9656 
Ohhhh...
I found that what the really problem is. Your nodes have no RF antenna. Look at the upper left corner of the chip. That's the antenna interface.
In our video, our nodes have the on-board antenna so that we don't need to plug in the antenna :).
And we have tested the performance of the antenna before. esp32c3 can communicate with each other in an open environment 70 meters apart and  when passing through two walls, the communication success rate is 97% when the distance is 13 meters in ESP_PWR_LVL_P9.
Btw, i test set the txpwr after ble_mesh_init, and it have successed. maybe you can try with other different value.

gomez9656, 2022-10-28:
Hi @forx157 , you're right.

I used an ESP32 in the past and used that same interface to connect an extern antenna, didn't notice on this one.

Can you point me to a development board that comes with the ESP32-C3 MCU and an onboard antenna?

Thanks a lot for all the help :)

forx157, 2022-10-31:
@Hi, @gomez9656 , sorry for replay so late.
You can try with ESP32-C3-DevKitM-1 Development Board. Here's a picture of it.
[image]
You can buy it from the official store. Here's the link.
[Offical store](https://www.amazon.com/Espressif-ESP32-C3-DevKitM-1-Development-Board/dp/B08W2J9B8J/ref=sr_1_1?crid=WKXOCPQBCC4F&keywords=espressif+esp32-c3-devkitm-1&qid=1667187983&qu=eyJxc2MiOiIxLjUyIiwicXNhIjoiMS41OSIsInFzcCI6IjAuMDAifQ%3D%3D&sprefix=espressif%2Caps%2C278&sr=8-1)

closed 2023-05-24 by Alvin1Zhang