espressif/esp-idf #9942: How to support IEC104 protocol

Kingwulin, 2022-10-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello，I want to use the protocol IEC104 ,but there isn't support this protocol in sdk,how can I make it to work ok in idf sdk
the protocol of IEC104 is like [this](https://github.com/airpig2011/IEC104)

david-cermak, 2022-10-24:
@Kingwulin Have you tried to build and run the library with IDF? I can see that it uses STM32 and supports linux port. Couldn't find STM32 related defines, though, do you have more information about an embedded project? Any examples or projects?
But looking at the linux port, it should work directly with ESP32 and IDF, since it uses socket API and a simple pthread interface which is supported.

Kingwulin, 2022-10-25:
I had achive it now,it jsut a protecol work on tcp client

david-cermak, 2022-10-26:
Could you close the issue then?

Moreover, I'd like to ask if it's not too much trouble to share your port layer (or related tips/ideas), if someone else have similar request? (or you might even consider publishing this in the espressif component registry: https://components.espressif.com/)

Alvin1Zhang, 2023-01-31:
Thanks for reporting, feel free to reopen with more updates.

closed 2023-01-31 by Alvin1Zhang