espressif/esp-idf #9945: external flash format watchodog triggered during formating

beginner140BB, 2022-10-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

wtachdog timer got triggered during chip erase in external flash.
[image]
code flow

external flash init
unmount & format

mounting again for the purpose of writing to fresh chip then its showing failed to open file for reading with some backtrace
[image]

what may be the isuue

beginner140BB, 2022-10-10:
```
	void ext_flash_begin()
{
	/**************************************************************************/
		// Set up SPI bus and initialize the external SPI Flash chip
	    flash = example_init_ext_flash();
	    if (flash == NULL) {
	        return;
	    }
	    // Add the entire external flash chip as a partition
	    const char *partition_label = "storage1";
	    example_add_partition(flash, partition_label);
	   // List the available partitions
	    example_list_data_partitions();

	    // Initialize FAT FS in the partition
	    if (!example_mount_fatfs(partition_label)) {
	        return;
	    }
	    // Print FAT FS size information
	    size_t bytes_total, bytes_free;
	    example_get_fatfs_usage(&bytes_total, &bytes_free);
	    ESP_LOGI(TAG, "FAT FS: %d kB total, %d kB free", bytes_total / 1024, bytes_free / 1024);

	    /**************************************************************************/
}

	void ext_flash_unmount()
{
	 ESP_LOGI(TAG, "Unmounting EXT FAT filesystem");
	 ESP_ERROR_CHECK( esp_vfs_fat_spiflash_unmount(base_path1, s_wl_handle));
	 ESP_LOGI(TAG, "Done");
	 return;
}

bool example_mount_fatfs(const char* partition_label)
{
    ESP_LOGI(TAG, "Mounting FAT filesystem");
    const esp_vfs_fat_mount_config_t mount_config = {
            .max_files = 4,
            .format_if_mount_failed = true,
            .allocation_unit_size = CONFIG_WL_SECTOR_SIZE
    };
    esp_err_t err = esp_vfs_fat_spiflash_mount(base_path1, partition_label, &mount_config, &s_wl_handle);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "Failed to mount FATFS (%s)", esp_err_to_name(err));
        return false;
    }
    return true;
}

	
	void ext_flash_format()
{
	esp_flash_erase_chip(flash);

}
```
	
	
	**function calls in order**
	
	ext_flash_begin();
	ext_flash_unmount();
	ext_flash_format();
	example_mount_fatfs("storage1");
	memset(rd_fdata,0,sizeof(rd_fdata));
	ext_flash_write("/extfile.txt",wt_data,32);
	printf("before extformat \n :%s\n\r",wt_data);
	ext_fat_read("/extfile.txt",rd_fdata,32);
	printf("before extformat1 \n :%s\n\r",rd_fdata);
	

the read write and begin function is working 
but  calling formating there is some isuue

chipweinberger, 2022-10-10:
I've hit this issue as well.

My non-ideal solution was to disable the Idle Task Watchdog during formatting, etc.

```
esp_task_wdt_delete(xTaskGetIdleTaskHandleForCPU(0));
esp_task_wdt_delete(xTaskGetIdleTaskHandleForCPU(1));

// call format code here

esp_task_wdt_add(xTaskGetIdleTaskHandleForCPU(0));
esp_task_wdt_add(xTaskGetIdleTaskHandleForCPU(1));
```

beginner140BB, 2022-10-11:
> My non-ideal solution was to disable the Idle Task Watchdog during formatting, mounting, etc.

okay will check on that @chipweinberger thanks😊

beginner140BB, 2022-10-11:
> This looks like a duplicate of #9212.

its been observed that there is significant change in heap size (xPortGetFreeHeapSize using this function) after calling the external flash using fat file system @igrr do you have any suggestion for this to use minimal heap area.

beginner140BB, 2022-11-04:
> its been said that it is by default 4096 sector size which is the minimum erase size

esp_flash_t ext;
        ext.chip_drv->sector_size=16;
        will this do the functionality as its been said on that **Do not call chip driver functions directly in other contexts**.
        is there any possible way to erase only specifc number of bytes from external flash instead of sectors of 4096 @igrr 
         `
/** @brief SPI flash chip driver definition structure.
 *
 * The chip driver structure contains chip-specific pointers to functions to perform SPI flash operations, and some
 * chip-specific numeric values.
 *
 * @note This is not a public API. These functions are called from the public API (declared in
 * esp_flash.h). They assume the caller has already validated arguments and enabled relevant protections
 * (disabling flash cache, prevent concurrent SPI access, etc.)
 *
 * Do not call chip driver functions directly in other contexts.
 *
 * A generic driver for generic chips and its related operations are defined in
 * spi_flash_chip_generic.h which can be used as building blocks for written
 * new/specific SPI flash chip drivers.
 *
 * @note All of these functions may be called with SPI flash cache disabled, so must only ever access IRAM/DRAM/ROM.
 */
struct spi_flash_chip_t {`

chipweinberger, 2022-11-04:
You need to erase the whole sector, and then rewrite any bytes you want to keep.

beginner140BB, 2022-11-04:
> I think there may be a helper function for this already, if you search ESP-IDF.

yes there is where multiples of sector size 4096 can be erased
here our need is that to erase only specific area is there any way to make it possible.

chipweinberger, 2022-11-04:
not possible. 

This is a limitation of NAND flash

closed 2023-01-13 by espressif-bot