espressif/esp-idf #9948: mbedtls do not have any process-level locks for cryptoauthlib

liliumjsn, 2022-10-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

### IDF version.
v4.4

### Operating System used.
freeRTOS

### Development Kit.
M5 Stack core2

### The Issue.
When the secure element (SE) is used by mbedtls, there are no process-level locks. Mbedtls is initializing and deinitializing the ATCA device in cryptoauthlib and uses the  implementation functions in "MicrochipTech/cryptoauthlib/main/lib/mbedtls" without using any lock mechanism, so you can not use the SE safely for your own procedures (read SN, read/sign certificates,  etc.). In our build, the application is handling the i2c layer using the i2c_manager and there are no collisions there. The problem is when you need to access the SE form a task while mqqt or other task is using mbedtls with SE. Other implementations in cryptoauthlib, like the pkcs11 implementation, use the supplied hal abstraction layer (MicrochipTech/cryptoauthlib/main/lib/hal/hal_freertos.c) to lock the ATCA device at process-level and handle the init-deinit. Even if we made our own implementation for mbedtls at cryptoauthlib, mbedtls will still init/deinit the ATCA device on its own, crashing our task.

AdityaHPatwardhan, 2022-10-27:
Hi @liliumjsn Sorry for the delayed reply.
I think I understand your question but I dont think it applies to esp-idf.
In esp-idf when esp-cryptoauthlib is used the initialisation for the `atca device` is done only once at the time of esp-tls handshake. ( [ref](https://github.com/espressif/esp-idf/blob/37c2b64f067921c8085f0c7d53fb71e6084d05dc/components/esp-tls/esp_tls_mbedtls.c#L882) )

When esp-cryptoauthlib is used, I dont think there are any init/deinit calls in the mbedtls code i.e. [here](https://github.com/espressif/esp-cryptoauthlib/tree/master/cryptoauthlib/lib/mbedtls)

If you are using amazon-freertos for your project then I think the interface is a little different. afr uses pkcs11 interface to communicate with `atecc608` chip.

Can you please correct me if there is any misunderstanding on my side?
Thanks,
Aditya

liliumjsn, 2022-10-31:
Hi @AdityaHPatwardhan,
You are correct about the calls. The problem is that there are scenarios where these calls are not controlled by the app layer. For example, if the  auto_reconnect in esp_mqtt_client_config is enabled, then these calls will happen at random time. If all the scenarios where covered by the app layer then we would implement our own locking, and communicate with the atca device when mbdetls is not making any handshake. If only the [mbedtls](https://github.com/MicrochipTech/cryptoauthlib/tree/main/lib/mbedtls) implementation was using **process-level** locks and was also handling the init/deinit of the atca device (and not being inside esp_tls_mbedtls.c) then all scenarios would be covered.

liliumjsn, 2022-12-29:
Any update please?

AdityaHPatwardhan, 2023-01-31:
Hi @liliumjsn Sorry for the delayed reply,
I think you are asking that the process level locks be implemented in the hal layer of the atca cryptoauthlib itself.
Again, I dont think that this can be handled with esp-idf. The changes needs to be added to the [mbedtls](https://github.com/MicrochipTech/cryptoauthlib/tree/main/lib/mbedtls) files as you have correctly mentioned.

I think the right place to raise this issue would be https://github.com/MicrochipTech/cryptoauthlib.
The component is as is mirrored in https://github.com/MicrochipTech/cryptoauthlib.

IIUC, I think all the atcab API would need some kind of locking mechanism for their synchronisation. I will follow it up internally as well.

AdityaHPatwardhan, 2023-02-13:
@liliumjsn  I am closing this issue since the responsibility to apply locks for individual tasks is the task of the library.
I am closing this issue, Please open a different issue at https://github.com/MicrochipTech/cryptoauthlib
Feel free to re-open this issue if you have any other thoughts.
Thanks,
Aditya

closed 2023-02-13 by AdityaHPatwardhan