espressif/esp-idf #9949: How to set the WiFi protocol?

podaen, 2022-10-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am trying to set the WiFi protocol and it works... Until I want to change it other than `WIFI_PROTOCOL_11B`. 

```
#include "esp_wifi.h"
#include "esp_log.h"

esp_err_t err;

wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
	err = esp_wifi_init(&cfg);
	if (err != ESP_OK) {
		printf("Could not set wifi power mode!");
		return false;
	}

	err = esp_wifi_set_mode(WIFI_MODE_STA);
	if (err != ESP_OK) {
		printf("Could not set sta!");
		return false;
	}
        //const uint8_t protocol = WIFI_PROTOCOL_11B | WIFI_PROTOCOL_11G;//works, but every time set to B
	//const uint8_t protocol = WIFI_PROTOCOL_11G;//doesn't work
	//const uint8_t protocol = WIFI_PROTOCOL_11B | 0 | 0;//works
        //const uint8_t protocol = WIFI_PROTOCOL_11G | 0 | 0;//doesn't work
        //const uint8_t protocol = WIFI_PROTOCOL_11G| WIFI_PROTOCOL_11B ;//works, but every time set to B
	//const uint8_t protocol = WIFI_PROTOCOL_11B;//works
        const uint8_t protocol = WIFI_PROTOCOL_11G;//doesn't work

	err = esp_wifi_set_protocol(WIFI_IF_STA, protocol);
	if (err != ESP_OK) {
		//258 = ESP_ERR_INVALID_ARG
		printf("Could not set protocol!");
		return false;
	}

	uint8_t getprotocol;
	//esp_err_t err;
	err = esp_wifi_get_protocol(WIFI_IF_STA, &getprotocol);
	if (err != ESP_OK) {
		printf("Could not get protocol!");
		//log_e("Could not get protocol! %d", err);
		return false;
	}
	if (getprotocol & WIFI_PROTOCOL_11N) {
		printf("WiFi_Protocol_11n");
	}
	if (getprotocol & WIFI_PROTOCOL_11G) {
		printf("WiFi_Protocol_11g");
	}
	if (getprotocol & WIFI_PROTOCOL_11B) {
		printf("WiFi_Protocol_11b");
	}
```

Anybody knows what I am doing wrong?

kriegste, 2022-10-12:
Call esp_wifi_set_protocol() after esp_wifi_start().
Same problem here due to poor documentation:
https://github.com/espressif/esp-idf/issues/9933

podaen, 2022-10-12:
What happens is if fails to set a desire protocole?

kriegste, 2022-10-12:
There is no error and no warning in the console and calling esp_wifi_set_protocol() before or after esp_wifi_start() returns ESP_OK in both cases.
The only way to see there is a problem is calling esp_wifi_get_protocol() afterwards.
And the documentation does not mention there is such a limitation.

kriegste, 2022-10-14:
The correct order is

esp_netif_create_default_wifi_sta
esp_wifi_init
esp_wifi_start
esp_wifi_set_protocol <-----
esp_wifi_connect

It is sad we have to figure this out by trial and error...

BTW, I think you cannot use G mode only. The options are B, BG, BGN.

podaen, 2022-10-14:
To be complete

ESP_ERROR_CHECK(esp_netif_init());
  ESP_ERROR_CHECK(esp_event_loop_create_default());
  wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
  ESP_ERROR_CHECK(esp_wifi_init(&cfg));
  ESP_ERROR_CHECK(esp_wifi_set_storage(WIFI_STORAGE_RAM));
  ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
  ESP_ERROR_CHECK(esp_wifi_start());

#if CONFIG_ESPNOW_ENABLE_LONG_RANGE
  ESP_ERROR_CHECK(esp_wifi_set_protocol(ESP_IF_WIFI_STA, WIFI_PROTOCOL_LR));
#endif
}

This a code that is using esp_now to transfer.

zhangyanjiaoesp, 2022-10-17:
@podaen If you want to set to 11G only mode, please refer to this https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#wi-fi-protocol-mode
[image]

podaen, 2022-10-21:
following this function after WiFi start my get protocol returns 

> WiFi_Protocol_11g
> WiFi_Protocol_11b

```
	uint8_t getprotocol;
	//esp_err_t err;
	err = esp_wifi_get_protocol(WIFI_IF_STA, &getprotocol);
	if (err != ESP_OK) {
		printf("Could not get protocol!");
		//log_e("Could not get protocol! %d", err);
		return false;
	}
	if (getprotocol & WIFI_PROTOCOL_11N) {
		printf("WiFi_Protocol_11n");
	}
	if (getprotocol & WIFI_PROTOCOL_11G) {
		printf("WiFi_Protocol_11g");
	}
	if (getprotocol & WIFI_PROTOCOL_11B) {
		printf("WiFi_Protocol_11b");
	}
```

1. Without setting the protocol there are two protocols been set... Is this normal? 
2. How do I know which one is chosen in the connection to my Client?

kriegste, 2022-10-21:
1. The chosen protocol is saved to flash, so it survives a reboot.
https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html#_CPPv416esp_wifi_restorev

I wonder if esp_wifi_set_protocol and the other APIs trigger flash writes every time or only if a change occurs. So if I call esp_wifi_set_protocol, esp_wifi_set_config  and esp_wifi_set_mode on every start, does it wear out the flash?

kriegste, 2022-10-21:
This is mostly trial and error for me, too.

That snippit is for initializing the NVS. But it erases it only if it is full or incompatible.

If you want to erase it, call nvs_flash_erase() in your code and restart. To erase only the WiFi info (station and ap data, protocol, bandwidth and mode; however, the country settings do not seem to be stored in the NVS), call esp_wifi_restore() and restart.
If you erase the flash using esptool and reflash your app, everything returns to default settings as well.

podaen, 2022-10-21:
Alot of things in esp_wifi.h are stored somewhere... it can be chosen where
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#_CPPv414wifi_storage_t

The country code is stored in to the flash. It is the inactive time that is not stored in to the flash.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#_CPPv425esp_wifi_set_country_codePKcb

 I tried once a long time ago to set the country code of my esp. And I couldn't get it right anymore. I had to throw my esp away. I think I ereased something to much. I tried the esptool, but it didn't work out for me. I am more expirienced than before and know that all default values that can be filled are in the APIs. But what else is stored in to the nvs? And is it erased every flash, depends on the toolchain you are using.

podaen, 2022-10-21:
I just did the test with an esp is closer to the AP and now the protocol N, G and B are selected.

zhangyanjiaoesp, 2022-10-24:
> following this function after WiFi start my get protocol returns
>
>
> ```
> 	uint8_t getprotocol;
> 	//esp_err_t err;
> 	err = esp_wifi_get_protocol(WIFI_IF_STA, &getprotocol);
> 	if (err != ESP_OK) { [...]

@podaen  
1. When call `esp_wifi_get_protocol()`, it will return the phy mode saved in NVS, and the return value is the bitmap which represents the supported phymode capability. So your test result is normal.
2.  When connecting to AP, the connected phy mode will be determined based on the set of rates supported by both parties. This can be get from the wifi log:

> I (871) wifi:state: init -> auth (b0)
I (871) wifi:state: auth -> assoc (0)
I (881) wifi:state: assoc -> run (10)
I (881) wifi:connected with test_zyj, aid = 3, channel 6, BW20, bssid = ac:84:c6:7e:31:2b
I (881) wifi:security: Open Auth, **phy: bgn**, rssi: -12

zhangyanjiaoesp, 2022-10-24:
> 1. The chosen protocol is saved to flash, so it survives a reboot.
>    https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html#_CPPv416esp_wifi_restorev
>    So what your app reads depends on what you changed earlier, The default, i.e. with a blank flash is BGN, iirc.
> 
> I wonder if esp_wifi_set_protocol and the other APIs trigger flash writes every time or only if a change occurs. So if I call esp_wifi_set_protocol, esp_wifi_set_config and esp_wifi_set_mode on every start, does it wear out the flash?

@kriegste When setting something into NVS, it will check whether the new value is same as the NVS value, so flash will be wirtten only if change occurs.

podaen, 2022-10-24:
If I understand it right. If there is no valid config been found during the esp_wifi_restore, it set the protocol and etc. This is to minimize the write cycles to the flash memory and speedup waking up from sleep mode. With the toolchain of ESP-IDF... Will the nvs be cleared when flashing a new application to the esp, without changing the partition scheme?

> 2\. When connecting to AP, the connected phy mode will be determined based on the set of rates supported by both parties. This can be get from the wifi log:

> I (871) wifi:state: init -> auth (b0)
> I (871) wifi:state: auth -> assoc (0)
> I (881) wifi:state: assoc -> run (10)
> I (881) wifi:connected with test, aid = 3, channel 6, BW20, bssid = ac:84:c6:7e:31:2b
> I (881) wifi:security: Open Auth, **phy: bgn**, rssi: -12

I looked everywhere... Where can I find that file?

zhangyanjiaoesp, 2022-10-24:
This link may help you https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#wi-fi-protocol-mode
[image]

zhangyanjiaoesp, 2022-10-24:
when call `esp_wifi_restore()`, the NVS will load the default value.

podaen, 2022-10-24:
I took a new esp fresh out of the box and like this documentation `esp_wifi `says it is set to bgn.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#_CPPv421esp_wifi_set_protocol16wifi_interface_t7uint8_t

I am closing this. My issue was indeed in my coding and that was why I only could set it to B because probably I did it before wifi start at that time. Thanks @kriegste and @zhangyanjiaoesp for helping me out.

closed 2022-10-24 by podaen