espressif/esp-idf #9952: XM25QH128AHIG flash startup fail

maplerian, 2022-10-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-1146-g1308ff3d21

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### Development Kit.

custom board use chip ESP32-D0WD-V3 & flash XM25QH128AHIG

### Power Supply used.

External 3.3V

### What is the expected behavior?

Normal boot

### What is the actual behavior?

It does not start properly

### Steps to reproduce.

1. Use the FLASH chip XM25QH128AHIG
2. build and flush app
3. setup

### Debug Logs.

```plain
rst:0x3 (SW_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7032
load:0x40078000,len:15380
load:0x40080400,len:3816
0x40080400: _init at ??:?

entry 0x40080648
I (52) boot: ESP-IDF v5.1-dev-1146-g1308ff3d21-dirty 2nd stage bootloader
I (53) boot: compile time Oct 11 2022 09:43:28
I (56) bootloader_flash: XM25QHxxC startup flow
E (63) bootloader_flash: XMC flash startup fail
E (66) boot.esp32: failed when running XMC startup flow, reboot!
ets Jul 29 2019 12:21:46

`
```

### More Information.

```
esptool.py flash_id   
esptool.py v4.3
Found 1 serial ports
Serial port /dev/ttyACM0
Connecting.....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: b4:8a:0a:4e:b3:e4
Uploading stub...
Running stub...
Stub running...
Manufacturer: 20
Device: 7018
Detected flash size: 16MB
Hard resetting via RTS pin...

ginkgm, 2022-10-11:
Will the chip success to run after the reboot?

ginkgm, 2022-10-11:
And could you provide the schematic of the flash chip? Is it powered by ESP32's VDD_SDIO, or always by the system LDO

maplerian, 2022-10-11:
> 重新启动后，芯片会成功运行吗？
没成功, 并且会不断重启, 重复报这个错误
It fails, and restarts repeatedly, repeating the error

> And could you provide the schematic of the flash chip? Is it powered by ESP32's VDD_SDIO, or always by the system LDO

FLASH是由外部电源供电, 与芯片公用一个电源
The FLASH is powered by an external power supply, which is shared with the chip

maplerian, 2022-10-11:
> And could you provide the schematic of the flash chip? Is it powered by ESP32's VDD_SDIO, or always by the system LDO

@ginkgm 
FLASH power supply can be supplied through resistance selection system LOD/VDD_SDIO

maplerian, 2022-10-11:
https://github.com/espressif/esp-idf/issues/7994
https://github.com/espressif/idf-flash-vendor-patches/tree/main/examples/lock_sr
@ginkgm 
This issue has been tested and does not allow the chip to start properly

maplerian, 2022-10-11:
原因是bootloader在对比xmc芯片ID的时候, 没有找到对应ID导致的, 修改以下内容后, 可以正常启动.
The cause is that the bootloader does not find the corresponding ID when comparing the xmc chip ID. The bootloader can start normally after modifying the following information.
```
file : esp-idf/components/bootloader_support/bootloader_flash/src/bootloader_flash.c
...
static IRAM_ATTR bool is_xmc_chip_strict(uint32_t rdid)
{
    uint32_t vendor_id = BYTESHIFT(rdid, 2);
    uint32_t mfid = BYTESHIFT(rdid, 1);
    uint32_t cpid = BYTESHIFT(rdid, 0);

    if (vendor_id != XMC_VENDOR_ID) {
        return false;
    }

    bool matched = false;
    if (mfid == 0x40) {
        if (cpid >= 0x13 && cpid <= 0x20) {
            matched = true;
        }
    } else if (mfid == 0x41) {
        if (cpid >= 0x17 && cpid <= 0x20) {
            matched = true;
        }
    } else if (mfid == 0x50) {
        if (cpid >= 0x15 && cpid <= 0x16) {
            matched =  true;
        }
    } 
    /* add */
    else if (mfid == 0x70) {
        if (cpid >= 0x17 && cpid <= 0x20) {
            matched =  true;
        }
    }
}
...

closed 2022-10-11 by maplerian