espressif/esp-idf #9953: [Security] [Flash Encryption] Why are host generated keys discouraged when each device does not have its own key?

chipweinberger, 2022-10-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The docs page, https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/security/flash-encryption.html, says:

> With flash encryption enabled, the following types of data are encrypted by default:
> 
> - Firmware bootloader
> - Partition Table
> - All “app” type partitions
> 
> **Using Host Generated Key**
> 
> "It is possible to pre-generate a flash encryption key on the host computer and burn it into the eFuse. This allows you to pre-encrypt data on the host and flash already encrypted data without needing a plaintext flash update."
> 
> **_This option is not recommended for production, unless a separate key is generated for each individual device._**

My Bootloader code, App code, and Partition code is the same for every device. So, why would there be an advantage of having a unique key for key device? If an attacker decrypts a single device, it's game over, they'll have all the app data. 

Furthermore, unique keys bring huge limitations:

**Disadvantages**:
- each device needs its own unique flash image
- OTA updates become a big pain, and feasibly impossible without a server to create the unique images

**Advantages**:
- ?

mahavirj, 2022-10-11:
@chipweinberger 

Unique keys helps in reducing scalability of the attacks. Flash encryption (through NVS encryption scheme) can also help to protect confidential data (e.g., WiFi credentials, TLS mutual auth keys) of the device. Hence, it is always recommended to have unique key per device.

> each device needs its own unique flash image

For flash encryption development mode scenario: `idf.py -p PORT encrypted-flash monitor` command can help to flash the project based on flash encryption settings on the device. This internally uses `esptool` with `--encrypt` option. (Please see `esptool.py write_flash --help` for details). 

Alternatively, for production mode scenario, device itself will generate the random unique key and encrypt the partitions on the flash.

> OTA updates become a big pain, and feasibly impossible without a server to create the unique images

Firmware images hosted on the server shall be plain-text. Encryption will happen when the firmware gets written to the flash. Please see section "Consideration III" from technical post at https://blog.espressif.com/ota-updates-framework-ab5438e30c12

mahavirj, 2022-10-11:
@chipweinberger 

Have you explored https://github.com/espressif/esp-idf/tree/master/examples/system/ota/pre_encrypted_ota example from IDF? This should allow you to have encrypted images for OTA updates. Please see relevant discussion from https://github.com/espressif/esp-idf/issues/6172#issuecomment-884777914

chipweinberger, 2022-10-11:
Thanks @igrr for the additional info.

That makes sense, and is thankfully not needed for my use case (I have no sensitive data stored on device).

Do you have a opinion on Option A vs Option B in terms of long term maintenance? & code complexity?

mahavirj, 2022-10-12:
@chipweinberger 

> Option A) single flash encryption key + store the binary in the mobile app with that key

Flash encryption scheme has dependency on flash offset where you will be flashing the firmware. With OTA updates, firmware offset shall change and hence you will have to maintain appropriate binary for corresponding slot in the mobile app. Please see above linked issue from my earlier comment on why this approach is not very feasible. 

> Option B) unique flash encryption keys + store the binary in the mobile app custom encrypted with a global AES key + decrypt AES binary on device + transparently re-encrypt binary using per-device flash encryption key

This approach could work. We already have similar scheme (slightly different way in which pre-encryption is achieved) supported in example I linked in my earlier comment.

> I am only interested in protecting the app code.

We also have a provision for [ESP32-PICO-V3 (SiP)](https://www.espressif.com/sites/default/files/documentation/esp32-pico-v3_datasheet_en.pdf), where flash is integrated and flash pins are terminated internally. This could also help, if you are concerned only about protecting app code from being read out externally through physical access.

chipweinberger, 2022-10-13:
Thanks for linking the example.  

At first glance, I thought the example was for Option A, but now I understand it is more similar to Option B. 

Thanks for the information!!!

closed 2022-10-13 by chipweinberger

mahavirj, 2022-10-13:
We do have section in docs https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/esp_https_ota.html#ota-upgrades-with-pre-encrypted-firmware. We will extend it to add more details, benefits and usage scenarios. Thanks for the feedback. CC @hmalpani