espressif/esp-idf #9954: esp_http_client skip_cert_common_name_check does not work when cn is ip address

fzboffice, 2022-10-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2-378-g9269a536ac

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom Board

### Power Supply used.

External 3.3V

### What is the expected behavior?

esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x2700
esp-tls-mbedtls: Failed to verify peer certificate!
esp-tls: Failed to open new connection
TRANSPORT_BASE: Failed to open a new connection
HTTP_CLIENT: Connection failed, sock < 0

### What is the actual behavior?

verify success

### Steps to reproduce.

Generate a certificate whose cn name is ip

fzboffice, 2022-10-13:
@igrr I hope the verification succeeds, but it fails.

ESP-YJM, 2022-10-13:
@fzboffice You can enable `.skip_cert_common_name_check = true` in http client config structure.

fzboffice, 2022-10-13:
@ESP-YJM I do not want enable that...  The http client will request different hostname.

AdityaHPatwardhan, 2022-10-13:
That is interesting @fzboffice !
I have locally tested out a scenario just to be sure about mbedTLS behaviour.
1) Create a openssl server with certificate in which server IP address is used as the common name
2) Connect to this local server using mbedtls on esp32.
For me the connection succeeds and I have verified that the hostname set by mbedtls internally is the IP address obtained from the server URI. 

Can you please share the certificate that you are using, I think that may give some additional details

fzboffice, 2022-10-13:
@AdityaHPatwardhan here is ca:
-----BEGIN CERTIFICATE-----
MIIDGzCCAgMCFFBYexUvqcZ7q25Sadx+TsNMYGnCMA0GCSqGSIb3DQEBCwUAMEkx
CzAJBgNVBAYTAkNOMQ0wCwYDVQQIDARUZXN0MQ0wCwYDVQQHDARUZXN0MQ0wCwYD
VQQKDARUZXN0MQ0wCwYDVQQDDARyb290MCAXDTIyMTAxMzA2NDczMloYDzIxMjIw
OTE5MDY0NzMyWjBJMQswCQYDVQQGEwJDTjENMAsGA1UECAwEVGVzdDENMAsGA1UE
BwwEVGVzdDENMAsGA1UECgwEVGVzdDENMAsGA1UEAwwEcm9vdDCCASIwDQYJKoZI
hvcNAQEBBQADggEPADCCAQoCggEBAKZs4D0i9euYapVsaFy2UgHMrTwKHWtXS4uU
tG69UVNeZ30yJfiUdTt8manrrK8CMqONnfIhhFbrO9774kpXh4fLa9KHzJ9kntKU
V3kAbUFT3Wpgdew5q5gAHCe7fBGPfEIlS/Ooh3jsFzaaEr71Lapna5699Zhm/OBX
AYQQP8cy0RgGN9HDN0FPZicOMztcUuZJl5GHKAjOtkXS+FtkLf4sSalzrZ+2boZO
TqYkY0Vf7kGtT0/raO8B8RlzIoV6qyyS5FdjYKTtqQadOeVBbum502dJEInG9sqp
7STNf3NgveqcFk9aLkiCAOWUHTwC/QcQ0DjJ9Qd8ePDjseHfG0MCAwEAATANBgkq
hkiG9w0BAQsFAAOCAQEAl1c5dDkwQe4unUkhs49Oakh9pruGTW5p9sc4J2FMqM31
dA+HwOPst5SBL8GoI35RvAr6SEMNGhjgmLT3XRoiZOQgphQuwOIu28f6tdQoLusS
tr5JYFOuiUgDeLHgDXo9PA43jIlxUdniLuMXzhZO6Fu/wZuyEQPhsJgw1prQzYzY
lUPy/geawBM8LNGVDbcGHLueEvsQt/QDS8JeBPvX8A6U10O4TeYOgKS0n/BmGFrJ
iDOFMPG37O374R2UBkrh26E3ngrFb7ZtVoYB5RuofMW/ZS6cUEjx6Iid4ByMYZmq
wLg3oUGHYMJci+18JrRT/NmcqPYpRAu3Xri3w0gPHg==
-----END CERTIFICATE-----
-----BEGIN RSA PRIVATE KEY-----
MIIEowIBAAKCAQEApmzgPSL165hqlWxoXLZSAcytPAoda1dLi5S0br1RU15nfTIl
+JR1O3yZqeusrwIyo42d8iGEVus73vviSleHh8tr0ofMn2Se0pRXeQBtQVPdamB1
7DmrmAAcJ7t8EY98QiVL86iHeOwXNpoSvvUtqmdrnr31mGb84FcBhBA/xzLRGAY3
0cM3QU9mJw4zO1xS5kmXkYcoCM62RdL4W2Qt/ixJqXOtn7Zuhk5OpiRjRV/uQa1P
T+to7wHxGXMihXqrLJLkV2NgpO2pBp055UFu6bnTZ0kQicb2yqntJM1/c2C96pwW
T1ouSIIA5ZQdPAL9BxDQOMn1B3x48OOx4d8bQwIDAQABAoIBADkX8tmma71WPYJj
fW4I8htB71BAnBUeDoGlEz1GX2b9ZP1zPcmIjKtKtKCxhD1JwLuRMtVFIFR9QL3f
paMhYhJ4+9Vu1wj7iOzvDQ8q0VBvB++dhfbzVeXAa+hJjmer1NIQX30InOkj/Vaz
BqlSDUd/UVFVE7Mj6a7AynlW5bpdK9nzXnBM8mBctiMQhK0wgjaGrva51ViifCCK
HS20UH90VHj7fIeUy+i74G8/bTle6EjKHuygvMtomyposFqaFQvPVNZ6+bJCMelQ
JnlYRJ+Zng9M9SkblRjvNrJSN5V+CtPKSidz8tvVYHxNBeb5jX4z2i33veuWO28l
DsxEuGECgYEA1oMOLeEuYLUkHdvrpsDW5arbp/5rVOA8MG4FbTAfTkwXOERAhQOF
PicSqviRHURcEllXcZ8JIfFYyCaDigY7F0KmgE+E+VAD+vDbnFLnCh00WFSjLtjc
orP2oGha5CjBq9opFin6QeaTKxMMM8XwI/i/YUOPfw3zxAwAUnNAHJECgYEAxpzy
wEj4MBv6YjM3UgkfQ3SCsvLqf4r/hwENkoypm1iHxnJULF3qk5BpxmqdEl55FfKI
cpIcQH+79rZLMMCfvWJaVIwilVGRAyiCTFu/OHNRWKzu6JoTwtxjHv2uZcdf88Ra
SvIxFm9X6qEEYrQUI/4Twqwyb0mFh5JvDNfNdJMCgYB+uRVbTTfUsa4QZack768E
Jpd0+vv1PDLSeWyDQHZ+Dr144/kpmxbPzM1ea1Fu9L9fp5/c45LFlopYU0hJ9bxa
1Wj/S/rGPdrcGAXkp6UNeIs/02BDgKF3DNGZmYx4VQt0toCnM01AOTKfs7T1sO1Y
ks6cawkOifOD0YMvMh6hwQKBgAI1krmy1hfq58EZ4r6l3VQpwp5s91DN4gPFbjo4
GFbjCDeb4/XOYaKck6CiZPbkWCJ3XS8xEuFgyl13L6TuL5iytGCGYogYrS1E0RY6
WwkstzXuToYF2LxtKSkaF0uIiTez8nEs+oDOYkL44YybhjHUTLH4qmL0hWi5p0SU
TfPDAoGBAJE/dQTCIR0HjVZlz2EM9MUEu0dR7+9DOlF/243iafR+9xMaE3UwiTPA
1xavP1QXFHUvLqdLCB5PCmE9GTWJB5aaXExeDP+699tcLALYzp+/cUDWSwAtkNDx
kTE35dAq3QR3/VRVmHCZAD9DpfydVEqo+Ner+myTNdXT3r9S1wsq
-----END RSA PRIVATE KEY-----

here is server cert:
-----BEGIN CERTIFICATE-----
MIIDPDCCAiSgAwIBAgIBAjANBgkqhkiG9w0BAQsFADBJMQswCQYDVQQGEwJDTjEN
MAsGA1UECAwEVGVzdDENMAsGA1UEBwwEVGVzdDENMAsGA1UECgwEVGVzdDENMAsG
A1UEAwwEcm9vdDAgFw0yMjEwMTMwNjQ3MzRaGA8yMTIyMDkxOTA2NDczNFowUjEL
MAkGA1UEBhMCQ04xDTALBgNVBAgTBFRlc3QxDTALBgNVBAcTBFRlc3QxDTALBgNV
BAoTBFRlc3QxFjAUBgNVBAMTDTE5Mi4xNjguMi4xNjMwggEiMA0GCSqGSIb3DQEB
AQUAA4IBDwAwggEKAoIBAQC3VeYd1NiO9AVCGD+CnPaMNifpejt9gCUd+xXZpKyr
hSDx/xYY7oI1B1qXwsHwLRxYhMDhaodB5BNy6uILWyu3cqgGMQS5Wd5hK2rdVWKc
CkaSP+TeRulclbA7i8WKOI8eLyTOUOb12+OHMPfD58VVsW7vOwwM8ddfjV3A0umw
CiCv/8QPmIkaJ6xeZLumvoN7CiU/pvyTNrGYYiFJYL7qM07gPufRtFRjy7zYrDDA
KMqG2Y6kaAKZVqzOFbovvLoCpO0Sg/dcQdgLhRRPiXVqHssQD1HfeOdMnQYXSvDV
hBuTt9eHRmogHddDmqI7k5Df2hETM+a/pQ4Zq7ZFveLXAgMBAAGjJDAiMCAGA1Ud
EQQZMBeCCWxvY2FsaG9zdIcEfwAAAYcEwKgCozANBgkqhkiG9w0BAQsFAAOCAQEA
FEOdjWQm06ZnYiU8++pvs5csAK8DVq7CHhr0f3IvHyx1ml8Pc7uluzxLkVJIrxOU
4WgnZwbSq4Kp+t3g7nq3bx93kOkV25iUwsgMCvAXMTwEYIqDBuG3xtIwBUY0iC0m
8itKYWh9fwbln+zDS/RSqVEBG6PC03bUArDsHazQcB+o1nezG4Gv/DAqdYtm3Nkf
haXghAyxet9JxIduBpd95MP53CzaZD8J80/rSf/AZy/mbHNHigEV5/u768//5+VV
vgl/FMlrx7O5IyQsUHS4ADsfi9wnd6HL5BWaRQnhv924R7yKzL6/aWUdbhfznt6B
+tuhPMp85lumirg/L3LR+w==
-----END CERTIFICATE-----

AdityaHPatwardhan, 2022-10-13:
@fzboffice Oh okay! so your server cert is not the root CA. 

Can you confirm that the session is successfully established when you set `.skip_common_name = true` in the config structure for the esp_tls connection ?

Additionally, is it possible for you to send Debug logs from your esp after enabling following ?

1) Component config > mbedTLS ->  Enable mbedTLS debugging
2) Set mbedTLS debugging level (Verbose) 
3) Component config > Log output > Default log verbosity (Debug)

fzboffice, 2022-10-14:
session is successfully established when I set in the skip_common_name = true.

Here is the log when skip_common_name = false

I (286910) mbedtls: ssl_tls.c:5904 => handshake

I (286911) mbedtls: ssl_cli.c:4483 client state: 0

I (286911) mbedtls: ssl_msg.c:2102 => flush output

I (286916) mbedtls: ssl_msg.c:2114 <= flush output

I (286921) mbedtls: ssl_cli.c:4483 client state: 1

I (286926) mbedtls: ssl_msg.c:2102 => flush output

I (286932) mbedtls: ssl_msg.c:2114 <= flush output

I (286937) mbedtls: ssl_cli.c:999 => write client hello

I (286946) mbedtls: ssl_msg.c:2542 => write handshake message

I (286951) mbedtls: ssl_msg.c:2701 => write record

I (286957) mbedtls: ssl_msg.c:2102 => flush output

I (286961) mbedtls: ssl_msg.c:2122 message length: 244, out_left: 244

I (286969) mbedtls: ssl_msg.c:2127 ssl->f_send() returned 244 (-0xffffff0c)

I (286976) mbedtls: ssl_msg.c:2155 <= flush output

I (286981) mbedtls: ssl_msg.c:2870 <= write record

I (286986) mbedtls: ssl_msg.c:2678 <= write handshake message

I (286993) mbedtls: ssl_cli.c:1467 <= write client hello

I (286999) mbedtls: ssl_cli.c:4483 client state: 2

I (287004) mbedtls: ssl_msg.c:2102 => flush output

I (287010) mbedtls: ssl_msg.c:2114 <= flush output

I (287015) mbedtls: ssl_cli.c:2082 => parse server hello

I (287021) mbedtls: ssl_msg.c:3941 => read record

I (287027) mbedtls: ssl_msg.c:1886 => fetch input

I (287032) mbedtls: ssl_msg.c:2043 in_left: 0, nb_want: 5

I (287038) mbedtls: ssl_msg.c:2068 in_left: 0, nb_want: 5

I (287044) mbedtls: ssl_msg.c:2069 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)

I (287052) mbedtls: ssl_msg.c:2089 <= fetch input

I (287058) mbedtls: ssl_msg.c:1886 => fetch input

I (287063) mbedtls: ssl_msg.c:2043 in_left: 5, nb_want: 74

I (287070) mbedtls: ssl_msg.c:2068 in_left: 5, nb_want: 74

I (287076) mbedtls: ssl_msg.c:2069 ssl->f_recv(_timeout)() returned 69 (-0xffffffbb)

I (287084) mbedtls: ssl_msg.c:2089 <= fetch input

I (287091) mbedtls: ssl_msg.c:4015 <= read record

I (287095) mbedtls: ssl_cli.c:2385 server hello, total extension length: 25

I (287103) mbedtls: ssl_cli.c:2627 <= parse server hello

I (287108) mbedtls: ssl_cli.c:4483 client state: 3

I (287114) mbedtls: ssl_msg.c:2102 => flush output

I (287119) mbedtls: ssl_msg.c:2114 <= flush output

I (287125) mbedtls: ssl_tls.c:2878 => parse certificate

I (287131) mbedtls: ssl_msg.c:3941 => read record

I (287136) mbedtls: ssl_msg.c:1886 => fetch input

I (287141) mbedtls: ssl_msg.c:2043 in_left: 0, nb_want: 5

I (287148) mbedtls: ssl_msg.c:2068 in_left: 0, nb_want: 5

I (287154) mbedtls: ssl_msg.c:2069 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)

I (287162) mbedtls: ssl_msg.c:2089 <= fetch input

I (287167) mbedtls: ssl_msg.c:1886 => fetch input

I (287173) mbedtls: ssl_msg.c:2043 in_left: 5, nb_want: 1649

I (287180) mbedtls: ssl_msg.c:2068 in_left: 5, nb_want: 1649

I (287185) mbedtls: ssl_msg.c:2069 ssl->f_recv(_timeout)() returned 1644 (-0xfffff994)

I (287194) mbedtls: ssl_msg.c:2089 <= fetch input

I (287212) mbedtls: ssl_msg.c:4015 <= read record

W (287220) mbedtls: ssl_tls.c:2702 x509_verify_cert() returned -9984 (-0x2700)

I (287220) mbedtls: ssl_msg.c:4990 => send alert message

I (287223) mbedtls: ssl_msg.c:2701 => write record

I (287229) mbedtls: ssl_msg.c:2102 => flush output

I (287234) mbedtls: ssl_msg.c:2122 message length: 7, out_left: 7

I (287241) mbedtls: ssl_msg.c:2127 ssl->f_send() returned 7 (-0xfffffff9)

I (287248) mbedtls: ssl_msg.c:2155 <= flush output

I (287254) mbedtls: ssl_msg.c:2870 <= write record

I (287259) mbedtls: ssl_msg.c:5003 <= send alert message

I (287265) mbedtls: ssl_tls.c:5915 <= handshake

E (287270) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x2700
E (287277) esp-tls: Failed to open new connection
E (287282) TRANSPORT_BASE: Failed to open a new connection
E (287292) HTTP_CLIENT: Connection failed, sock < 0

AdityaHPatwardhan, 2022-10-14:
Hi @fzboffice Can you please confirm the URL for your connection.
I think the problem is with the common_name that is being set.
internally esp-tls does following.
1. If common_name is set in esp_tls config structure, use it as the host_name
2. If common_name is not set it uses the URL sent to to obtain the host name.

This host name must match the CN in the certificate.
In release/v4.4 we dont have an option in esp_http_client to set the common_name for the esp_tls connection, that part was added later https://github.com/espressif/esp-idf/commit/4904d57fd98a787e41c268cfa19f590a0607b3c9 

I am providing you a patch that puts this feature over v4.4,
Can you please try setting the appropriate common_name in the esp_http_client configuration structure after you have applied this patch.
[common_name_change.patch.zip](https://github.com/espressif/esp-idf/files/9782844/common_name_change.patch.zip)

Thanks,
Aditya

fzboffice, 2022-10-14:
@AdityaHPatwardhan I apply the patch and set common_name = "192.1682.2.163"  and the url is "https://192.168.2.163"
Session connection failed.

AdityaHPatwardhan, 2022-10-14:
@fzboffice  The issue in your case is in the certificate extension.
Your x509 v3 certificate has added `IP Address field`, however mbedTLS currently does not support that particular type in the SAN section ( I found this note - [mbedtls note](https://github.com/Mbed-TLS/mbedtls/blob/b3c30907d61e896f2d88db92aaa1e38596b92f78/library/x509_crt.c#L639). There is already an issue/feature request for the same https://github.com/Mbed-TLS/mbedtls/issues/5082

In you case you should either follow the users approach i.e. add the IP address in the DNS field or remove the IP address field. With the patch you should be able to set the respective common name.

Thanks,
Aditya

fzboffice, 2022-10-17:
@AdityaHPatwardhan Thank you very much.

AdityaHPatwardhan, 2022-10-17:
@fzboffice Thanks for raising the issue, if it has been resolved can you please close the issue ?
Thanks

closed 2022-10-17 by fzboffice