espressif/esp-idf #9959: Can't force a lockup of the CPU

axos88, 2022-10-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I would like to be able to force a lockup of the CPU in some rare cases, even if the panic handling is set to reboot the ESP normally.
Currently I can't do that, because if I do:
```
            esp_task_wdt_deinit();
            vPortEnterCritical();
            loop {}
```
the IWDT will reset the CPU in a flash.

There is no API to disable the IWDT, and `disable_all_wdts` in `panic.c` is static inline, and cannot be accessed from outside of the source file.

Please consider either adding an API for forcing a system lockup, or to turn off the IWDT.

axos88, 2022-10-12:
@igrr, It is a dangerous thing to do, and I like the alternative to have an API call asking for a lockup better anyways, something like 

```
void esp_lockup() {
           disable_all_wdts();
           vPortEnterCritical();
           loop {}
}
```

axos88, 2022-10-12:
It would probably be with putting the CPU(s) to sleep instead of looping infinitely at full speed, also would need to consider how to lock the coprocessor as well

Dazza0, 2022-12-14:
@axos88 I'm closing this issue as I don't think we are going to an API to lockup the system since we don't want our users to ever do this (due to it's dangerous nature). I'm not sure what application scenario would require the CPUs to be locked up, but may there's a better way to achieve the same?

However, if you must, it's still possible for your own `esp_lockup()` function to lockup the CPUs by...

- Disabling all watchdogs manually (using the WDT HAL APIs)
- Clock gating the CPUs. The CPUs are clock gated during light/deep sleep, so there should be a RTC register bit to control that,

closed 2022-12-14 by Dazza0

axos88, 2022-12-14:
The use case is during development mostly. If something happens, and abort is called (after a panic in rust for example), the cpu is automatically reset, and it is hard to find the actual cause and message afterwards. So too avoid that, I'd like to lock up the system until it is manually reset.

Is it a dangerous thing to do? Probably, but same goes for a lot of things possible with embedded development, putting the wrong pin as output can have catastrophical consequences, but I'm really not sure that it is the operating system / esp-s responsibility to guard against that. That sounds a bit like when apple removes features because the average user would be to dumb to understand it, leaving the power users frustrated. Of course that is communicated differently, and it was just a rant, but I hope you get what I mean by it.

axos88, 2022-12-14:
But it can happen that something catastrophical happens in real life, and the only sane response is to bring the system in a safe state (say stop the car) and refuse to do anything else until manual intervention occurs

Dazza0, 2022-12-15:
@axos88 

> The use case is during development mostly. If something happens, and abort is called (after a panic in rust for example), the cpu is automatically reset, and it is hard to find the actual cause and message afterwards. So too avoid that, I'd like to lock up the system until it is manually reset.

We have the [`CONFIG_ESP_SYSTEM_PANIC_PRINT_HALT` option](https://docs.espressif.com/projects/esp-idf/en/v5.0/esp32/api-reference/kconfig.html#config-esp-system-panic) if that helps.

The main difficulty I see with locking up the system is that it's not as simple as locking up the CPUs. There are a couple of other things on the ESP that could be running even if the CPUs are locked up.

- ULP Coprocessors could also be running
- Any currently running peripherals will also need to be halted to stop them from interfering with the reset of the system (e.g., stopping the CAN peripheral from interacting with the car's CAN bus after a fatal error).
    - Most peripherals can be gated by the APB clock
    - But there are some peripherals that are can clocked separately (e.g., LEDC)

axos88, 2022-12-18:
Yes, that seems to be what I'm looking for. However it's documented as 

```
If FreeRTOS detects unexpected behaviour or an unhandled exception, the panic handler is invoked. Configure the panic handler’s action here.
```

Would calling `abort()` cause the panic handler to run?

Dazza0, 2023-01-04:
@axos88 

> Would calling abort() cause the panic handler to run?

It should, an `abort()` call will eventually use a null pointer access to trigger the panic handler ([line](https://github.com/espressif/esp-idf/blob/v5.0/components/esp_system/panic.c#L412)). However, because the the null pointer access, it is not treated as a run-time error (i.e., Guru meditation error) by the panic handler, thus the CPU registers are not dumped ([line](https://github.com/espressif/esp-idf/blob/v5.0/components/esp_system/port/panic_handler.c#L69)). Instead, only the backtrace is printed.

axos88, 2023-01-04:
I'll try that out. Seems to be good enough for my purposes paired with panic print halt