espressif/esp-idf #9966: Change LAN Priority (LAN should be higher then WLAN)

WatDavid, 2022-10-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hy there!

IDF: v.4.4.1

I want to change the priority of LAN/WLAN.
Currently when the ESP is connected to WLAN and LAN it uses the WLAN connection and not the LAN connection.
Is there a possiblility to change this?  I want that LAN is higher prior.
I tried to change it with netif_set_default()...
Is the variable [route_prio ](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_netif.html?highlight=route_prio#_CPPv4N25esp_netif_inherent_config10route_prioE)responsible for this?

KaeLL, 2022-10-13:
#8330

david-cermak, 2022-10-21:
@WatDavid Yes, setting the `route_prio` in the esp-netif configuration should do the job. But please note, that this would work only if both interfaces are set-up (usually link-up for ethernet, *connected* for wifi); if one is down, the other is set default. (won't work for some scenarios where e.g. ethernet link is up, but the connection is broken). In more complex situations, please use `esp_netif_set_default_netif()` to set the preferred netif the default gateway.

david-cermak, 2022-11-18:
@WatDavid Any update? Were you able to set the default interface?

Alvin1Zhang, 2022-12-12:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more updates. Thanks..

closed 2022-12-12 by Alvin1Zhang

david-cermak, 2022-12-16:
> No i was not able to set the default interface.

What have you tried, setting the `route_prio` didn't work?

Again, as noted in https://github.com/espressif/esp-idf/issues/9966#issuecomment-1287156705, the `route-prio` should work in most cases: if a specific interface goes down, the default netif is chosen from those that are up based on their priority.
If you need to apply more logic to it (i.e. checking connectivity instead of link up/down), you'd have to use `esp_netif_set_default_netif()`.

WatDavid, 2022-12-19:
currently I´m using IDF v.4.4.1.
esp_netif_set_default_netif() is not available in IDF v.4.4.1, or?
So i would have to uptade the IDF Version in my Project.

david-cermak, 2023-01-02:
> esp_netif_set_default_netif() is not available in IDF v.4.4.1, or?

That is correct, `esp_netif_set_default_netif()` has been added in f8907b78844a746125bb23c32bb65b55f7fb030e and is only available from `v5.0`.

> So i would have to uptade the IDF Version in my Project.

Yes, I'd suggest updating to [`v5.0`](https://docs.espressif.com/projects/esp-idf/en/v5.0/esp32/versions.html) since it's currently the *latest stable*.
Alternatively, you can use another public API (`esp_netif_get_netif_imp()`) to access lwip's netif pointer and set the default gateway directly in lwip. That's might be tricky if you use multiple interfaces, as the IDF *auto-route-prio* mechanism would "kick-in" and rewrite the default interface, but still manageable via event handlers.

```cpp
netif_set_default(esp_netif_get_netif_imp(esp_netif));
```