espressif/esp-idf #9967: to increase ESP32 OTA downloading speed

beginner140BB, 2022-10-13:
### Is your feature request related to a problem?

Issue is such that when updating esp32 firmware through ota. The downloading speed is observed that 10kbps to 12kbps.
The bin size is approx 1.5mb.So thereby it takes more time at this speed. 

### Describe the solution you'd like.

hence the update is through wifi need to increase wifi reception speed i.e the router provides more bandwidth.Expecting atleast 50 to 100 kbps speed in OTA updation through wifi thereby the waiting time for ota updation will reduce.

### Describe alternatives you've considered.

using arduino 1.05rc4  version there is observed some difference in the ota updation speed.

Need to know about how to **configure the sdk config using espidf platform for faster wifi reception  hence faster ota download** hence ESP32 can be programmed in more streched out way.

kriegste, 2022-10-15:
I am seeing this as well and I wonder what can be done to speed this up. Updating by other means (http server upload for example) is way faster, so neither flash writing speed nor WiFi speed are the problems here.
I have yet to try http instead of https.

beginner140BB, 2022-10-20:
will increase the wifi throughput will enable more ota download speed.
[](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#esp32-wi-fi-throughput)
does anyone specialised in how to configure the sdk config file to maximise the wifi throughput.

hmalpani, 2022-10-20:
Hello @beginner140BB 
This might help you increase the download speed: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/speed.html#improving-network-speed

beginner140BB, 2022-10-20:
> okay will check on that but i have observed a change from 10kbps to 54kbps after some edit on the sdk config based on
(https://github.com/espressif/arduino-esp32/issues/4529) using idf

kriegste, 2022-10-20:
That sounds normal. 23 seconds for ~1 megabyte is a typical OTA API update duration here.

Uploading and writing the firmware via the HTTP server on my ESP32 modules only takes 6 seconds, though!

Both goes through WiFi and both involves writing the flash, so the bottleneck for the OTA API must be somewhere else. Could be the encryption, of course...

beginner140BB, 2022-10-21:
> Uploading and writing the firmware via the HTTP server on my ESP32 modules only takes 6 seconds, though!

only 6 seconds what have you done for that...?❓🙋‍♂️
if the bin size is ~1.5mb the speed will be approx ~250kbps right🙄

hmalpani, 2022-10-21:
@beginner140BB 
We have `buffer_size` field in `ota_config->http_config`. By default its value is 1024. You can set it to a custom value. Bigger buffer size might also help in increasing the upgrade speed as client have to make fewer requests to download the binary image from server.

> Uploading and writing the firmware via the HTTP server on my ESP32 modules only takes 6 seconds, though!

@kriegste  Are you writing whole binary to server at once or sending it in chunks?

kriegste, 2022-10-21:
Nothing special. This is about it (omitted the http header and json response parts):

```
int remaining = req->content_len;
if (remaining == 0) return ESP_FAIL;

const esp_partition_t *partition = esp_ota_get_next_update_partition(NULL);
esp_ota_handle_t handle;
esp_err_t err = esp_ota_begin(partition, remaining, &handle);
if (err) return ESP_FAIL;

while (remaining > 0) {
	char buf[2048];
	const int received = httpd_req_recv(req, buf, MIN(remaining, sizeof(buf)));
	if (received <= 0) {
		if (received == HTTPD_SOCK_ERR_TIMEOUT) continue;
		esp_ota_abort(handle);
		return ESP_FAIL;
	}
	err = esp_ota_write(handle, buf, received);
	if (err) {
		esp_ota_abort(handle);
		return ESP_FAIL;
	}
	remaining -= received;
}

esp_ota_end(handle);
if (err) return ESP_FAIL;

err = esp_ota_set_boot_partition(partition);
if (err) return ESP_FAIL;

esp_restart();
return ESP_OK;
```

There is a JavaScript using XMLHttpRequest on the browser side. This way you can display a progress bar.

mahavirj, 2022-11-04:
I hope suggestions from comments here and from https://github.com/espressif/arduino-esp32/issues/4529 helped to fix the issue. Please feel free to re-open if you are still facing the issue.

closed 2022-11-04 by mahavirj

mbastida123, 2023-08-17:
Hi, I have made all the improvement on the Wifi side to improve its performance. However, now I've come to the escenario where the write to flash is the slowest by far.

My OTA loop is basically:
```
while(otaWriting){
  esp_http_client_read();
  esp_ota_write();
}
```

And now I observe that the esp_http_client read is extremely fast in comparasion to esp_ota_write().

My question is, is there a faster way to write to flash? I'm currently downloading a 1,8MB .bin in aproximately 1 minute.

kriegste, 2024-08-07:
Setting bulk_flash_erase to true (in esp_https_ota_config_t) cuts my update time in half.

kunj2308, 2025-01-20:
i am using the ble , in which the OTA speed is too low for 1MB file it takes more than 15 minutes is there any way to reduce this time ?