espressif/esp-idf #9968: status=16 in ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT after esp_ble_gap_update_conn_params()

jacek12345, 2022-10-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I see status=16 in ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT and after 1sec. see
 BT_HCI: DiscCmpl evt: hdl=0, rsn=0x8 
It is only if connecting with MSI notebook and it is 100% repeatable.
Whole context is:
1. congestion while sending data
2. updating connection parameters (to slower)
3. receiving status=16 in ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT
4. if my application makes disconnection after that status=16, receive BT_HCI: DiscCmpl evt: hdl=0, rsn=0x22
4a. if my application makes nothing (waiting) after status=16, receive BT_HCI: DiscCmpl evt: hdl=0, rsn=0x8

`I (1278924) GATTS_SPP_DEMO: event = 5

I (1278934) GATTS_SPP_DEMO: EVT 5, gatts if 3

I (1278934) GATTS_SPP_DEMO: event = 5

I (1278944) GATTS_SPP_DEMO: EVT 5, gatts if 3

I (1278944) GATTS_SPP_DEMO: event = 5

I (1278954) GATTS_SPP_DEMO: EVT 20, gatts if 3

I (1278954) GATTS_SPP_DEMO: event = 14

I (1278954) GATTS_SPP_DEMO: EVT 5, gatts if 3

I (1278964) GATTS_SPP_DEMO: event = 5

W (1281794) TESTC: 1a. creating connection params with: addr=58FB847169B, min_int=88, max_int=100, latency=0, timeout=200

W (1281794) TESTC: 2a. connection params negotiation started OK (second, slow)

 Time: 5946

 Congested events: 1
E (1282794) GATTS_SPP_DEMO: GAP_EVT, event 20

W (1282794) GATTS_SPP_DEMO: 3. Update conn params response: status =16, min_int=88, max_int=100, latency=0, current_int=12, timeout=200

W (1283774) BT_HCI: DiscCmpl evt: hdl=0, rsn=0x8
I (1283784) GATTS_SPP_DEMO: EVT 15, gatts if 3

I (1283784) GATTS_SPP_DEMO: event = f

W (1283814) BT_L2CAP: L2CAP - rcvd segment complete, unknown handle: 0

E (1283814) GATTS_SPP_DEMO: GAP_EVT, event 6`

esp-cjh, 2022-10-14:
Could you please capture the air interface packets and provide the capture file? You can use ellisys , nrf sniffer or other capture tools. I don't have MSI notebook to reproduce the problem.

jacek12345, 2022-10-17:
Now i don't have access to this MSI notebook. But i discovered that there is one way problem. My device (server) can't send data due to no free buffers. Checked with  esp_ble_get_cur_sendable_packets_num(). Data from MSI notebook (client) comes.