espressif/esp-idf #9969: SD card and ENC28J60 Ethernet Interface with ESP32

priya2212, 2022-10-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am facing some problem in integrating SD card and Ethernet both on ESP32 over SPI interface. 
Hardware setup - 
SD card/Etherent - Common Pins 
MOSI - 23
MISO - 19
SCK - 18

SD Card - CS - 5, Ethernet CS - 4
DMA Channel - SD Card - 1, Ethernet - 2

Software - 
First SD card is initialised successfully and fetch the data from it and then Ethernet is initialised and got the IP address successfully. Now when I try to open the file with fopen command to write some data, get error - 
`E (936695) diskio_sdmmc: Check status failed (0x107)`

I am stuck in the problem and not able to proceed further in developments. If anyone could help me urgently in giving some insights, that would be highly appreciated. 

Please let me know if need more details. 
Thank you.

nopnop2002, 2022-10-15:
SD-SPI uses HSPI(SPI2) Host Bus by default.

ENC28J60 may be using the same bus.

Try isolating the SPI bus.

priya2212, 2022-10-17:
@nopnop2002, Thank you for your support. 
I checked in my code and already isolating the SPI Bus. 
For ENC28J60 -    ` ESP_ERROR_CHECK(spi_bus_initialize(2, &buscfg, 2));`
And for SD Card - it is 1 by default. 
Still facing the same error.

priya2212, 2022-10-17:
SD card Initialisation function - 
```
void sdCard_init()
{
    esp_vfs_fat_sdmmc_mount_config_t mount_config = {
#ifdef CONFIG_EXAMPLE_FORMAT_IF_MOUNT_FAILED
        .format_if_mount_failed = true,
#else
        .format_if_mount_failed = false,
#endif // EXAMPLE_FORMAT_IF_MOUNT_FAILED
        .max_files = 5,
        .allocation_unit_size = 16 * 1024};
    sdmmc_card_t *card;
    const char mount_point[] = MOUNT_POINT;
    ESP_LOGI(TAG, "Initializing SD card");

    ESP_LOGI(TAG, "Using SPI peripheral");

    sdmmc_host_t host = SDSPI_HOST_DEFAULT();
    spi_bus_config_t bus_cfg = {
        .mosi_io_num = PIN_NUM_MOSI,
        .miso_io_num = PIN_NUM_MISO,
        .sclk_io_num = PIN_NUM_CLK,
        .quadwp_io_num = -1,
        .quadhd_io_num = -1,
        .max_transfer_sz = 10000,
    };

    printf("Default SPI Host is %d \n", host.slot);

    esp_err_t ret = spi_bus_initialize(host.slot, &bus_cfg, SPI_DMA_CHAN);
    if (ret != ESP_OK)
    {
        ESP_LOGE(TAG, "Failed to initialize bus.");
        return;
    }
    // This initializes the slot without card detect (CD) and write protect (WP) signals.
    // Modify slot_config.gpio_cd and slot_config.gpio_wp if your board has these signals.
    sdspi_device_config_t slot_config = SDSPI_DEVICE_CONFIG_DEFAULT();
    slot_config.gpio_cs = PIN_NUM_CS;
    slot_config.host_id = host.slot;

    ESP_LOGI(TAG, "Mounting filesystem");
    ret = esp_vfs_fat_sdspi_mount(mount_point, &host, &slot_config, &mount_config, &card);

    if (ret != ESP_OK)
    {
        if (ret == ESP_FAIL)
        {
            ESP_LOGE(TAG, "Failed to mount filesystem. "
                          "If you want the card to be formatted, set the EXAMPLE_FORMAT_IF_MOUNT_FAILED menuconfig option.");
        }
        else
        {
            ESP_LOGE(TAG, "Failed to initialize the card (%s). "
                          "Make sure SD card lines have pull-up resistors in place.",
                     esp_err_to_name(ret));
        }
        return;
    }
    ESP_LOGI(TAG, "Filesystem mounted");

    // Card has been initialized, print its properties
    sdmmc_card_print_info(stdout, card);
}

```
And ENC Initialisation function - 

```
void ethernet_enc28j60_connection(void){
    ESP_ERROR_CHECK(gpio_install_isr_service(0));
    ESP_ERROR_CHECK(esp_netif_init());
    ESP_ERROR_CHECK(esp_event_loop_create_default());
    esp_netif_config_t netif_cfg = ESP_NETIF_DEFAULT_ETH();
    esp_netif_t *eth_netif = esp_netif_new(&netif_cfg);
    ESP_ERROR_CHECK(esp_eth_set_default_handlers(eth_netif));
    // Register user defined event handers
    ESP_ERROR_CHECK(esp_event_handler_register(ETH_EVENT, ESP_EVENT_ANY_ID, &eth_event_handler, NULL));
    ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_ETH_GOT_IP, &got_ip_event_handler, NULL));

    spi_bus_config_t buscfg = {
        .miso_io_num = 19,
        .mosi_io_num = 23,
        .sclk_io_num = 18,
        .quadwp_io_num = -1,
        .quadhd_io_num = -1,
        .max_transfer_sz = 10000,
    };
    ESP_ERROR_CHECK(spi_bus_initialize(2, &buscfg, 2));

    /* ENC28J60 ethernet driver is based on spi driver */
    spi_device_interface_config_t devcfg = {
        .command_bits = 3,
        .address_bits = 5,
        .mode = 0,
        .clock_speed_hz = 5 * 1000 * 1000,
        .spics_io_num = 4,
        .queue_size = 20,
        
    };

    spi_device_handle_t spi_handle = NULL;
    ESP_ERROR_CHECK(spi_bus_add_device(2, &devcfg, &spi_handle));

    eth_enc28j60_config_t enc28j60_config = ETH_ENC28J60_DEFAULT_CONFIG(spi_handle);
    enc28j60_config.int_gpio_num = 22;

    eth_mac_config_t mac_config = ETH_MAC_DEFAULT_CONFIG();
    mac_config.smi_mdc_gpio_num = -1; // ENC28J60 doesn't have SMI interface
    mac_config.smi_mdio_gpio_num = -1;
    mac = esp_eth_mac_new_enc28j60(&enc28j60_config, &mac_config);

    eth_phy_config_t phy_config = ETH_PHY_DEFAULT_CONFIG();
    phy_config.autonego_timeout_ms = 0; // ENC28J60 doesn't support auto-negotiation
    phy_config.reset_gpio_num = -1;     // ENC28J60 doesn't have a pin to reset internal PHY
    esp_eth_phy_t *phy = esp_eth_phy_new_enc28j60(&phy_config);

    esp_eth_config_t eth_config = ETH_DEFAULT_CONFIG(mac, phy);
    esp_eth_handle_t eth_handle = NULL;
    ESP_ERROR_CHECK(esp_eth_driver_install(&eth_config, &eth_handle));

    mac->set_addr(mac, (uint8_t[]){
                           0x02, 0x00, 0x00, 0x12, 0x34, 0x56});

    /* attach Ethernet driver to TCP/IP stack */
    ESP_ERROR_CHECK(esp_netif_attach(eth_netif, esp_eth_new_netif_glue(eth_handle)));

    /* start Ethernet driver state machine */
    ESP_ERROR_CHECK(esp_eth_start(eth_handle));
}
```

nopnop2002, 2022-10-17:
ESP-IDF also supports W5500.

If you change ENC28J60 to W5500, you may find some cause.

[image]

priya2212, 2022-10-17:
I don't have this here. I will place an order for this but this will take another a week to be delivered to us. 
Meanwhile, can we check what can be the issue using ENC28J60. I don't see hardware fault as they (SD card and ENC) both are working perfectly fine independently. 
Facing problem when working together.

nopnop2002, 2022-10-17:
```
        .mosi_io_num = PIN_NUM_MOSI,
        .miso_io_num = PIN_NUM_MISO,
        .sclk_io_num = PIN_NUM_CLK,
    slot_config.gpio_cs = PIN_NUM_CS;
```

Are these GPIOs used by SDSPI overlapping with the GPIOs below?

```
        .miso_io_num = 19,
        .mosi_io_num = 23,
        .sclk_io_num = 18,
        .spics_io_num = 4,
    enc28j60_config.int_gpio_num = 22;
```

priya2212, 2022-10-17:
MISO, MOSI, CLK are same for both. 

CS pin is different - for SDSPI it is 5, and for ENC it is 4.

nopnop2002, 2022-10-17:
>MISO, MOSI, CLK are same for both.

I don't know if the same pins can be used on different SPI-BUSes.

Maksons, 2022-10-17:
You do not need to initialize SPI  Bus twice, probably it is a problem (you over init previos settings)!
Comment out 
` ESP_ERROR_CHECK(spi_bus_initialize(2, &buscfg, 2));`
SPI bus could have multiple device connected. 
Just use same SPU Bus / Host for Ethernet.

priya2212, 2022-10-17:
@Maksons , Thank you very much for your response. This absolutely solves the problem. I am now able to use both on same bus. 
@nopnop2002 , thanks for your help too.

closed 2022-10-17 by priya2212