espressif/esp-idf #9973: CORRUPT HEAP after WiFi reconnect if previous connection loss has not been recognized by ESP32

marclee5, 2022-10-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0-beta1-641-gc321739074

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### Development Kit.

ESP32-S2-WROVER and ESP32-S3-WROOM-1-N8R8

### Power Supply used.

USB

### What is the expected behavior?

If an external device reconnects to the ESP32 AP, the connection should be handled correctly, even if the previous connection loss has not been recognized by ESP32.

### What is the actual behavior?

If an external device reconnects to the ESP32 AP and the previous connection loss has not been recognized by ESP32, following erros occurs:

CORRUPT HEAP: Invalid data at 0x3ffdd4a0. Expected 0xfefefefe got 0x00000000

### Steps to reproduce.

**Configuration:**
ESP32 is in WiFi mode WIFI_MODE_APSTA
Heap corruption detection = Comprehensive
authmode of ESP-AP = WIFI_AUTH_WPA2_PSK

**Action:**
1. Step: Smartphone connects to ESP32 AP (channel 1)
2. Step: ESP32 STA connects to external accesspoint (channel 3 .. 11)
- ESP32 has to switch from channel 1 to 3 .. 11 (STA and AP)
- Smartphone is not capable to switch channel during a connection, so the connection gets lost
- The ESP32 does not recognize the connection loss to the smartphone
3. Step: The smartphone connects again to the ESP32 AP
-> CORRUPT HEAP: Invalid data at 0x3ffdd4a0. Expected 0xfefefefe got 0x00000000

**Code:**
```
#include "esp_log.h"
#include "esp_wifi.h"
#include "nvs_flash.h"
#include "lwip/ip_addr.h"

static const char* TAG = "main.c";
static uint16_t connect_timer = 0;

static void wifi_event_handler(void* arg, esp_event_base_t event_base, int32_t event_id, void* event_data)
{
  ESP_LOGI(TAG, "event_base = %i, event_id = %i", (int16_t) *event_base, (int16_t) event_id);

  if ((event_base == WIFI_EVENT) && (event_id == WIFI_EVENT_STA_START))
  { connect_timer = 1; }

  else if ((event_base == WIFI_EVENT) && (event_id == WIFI_EVENT_STA_DISCONNECTED))
[... 243 lines cut ...]
I (56066) wifi:station: a0:10:81:c2:d9:36 leave, AID = 1, bss_flags is 658531, bss:0x3ffdd1a0
I (56076) wifi:new:<4,0>, old:<4,0>, ap:<4,1>, sta:<4,0>, prof:1
[0;32mI (56076) main.c: event_base = 87, event_id = 15[0m
I (56556) wifi:new:<4,0>, old:<4,0>, ap:<4,1>, sta:<4,0>, prof:1
I (56556) wifi:station: a0:10:81:c2:d9:36 join, AID=1, bgn, 20
CORRUPT HEAP: Invalid data at 0x3ffdd4a0. Expected 0xfefefefe got 0x00000000

assert failed: multi_heap_malloc multi_heap_poisoning.c:256 (ret)

Backtrace: 0x40023912:0x3ffd51c0 0x4002a045:0x3ffd51e0 0x400309ad:0x3ffd5200 0x4002e96e:0x3ffd5320 0x40023a7d:0x3ffd5340 0x40023adb:0x3ffd5360 0x40026535:0x3ffd5380 0x400efd81:0x3ffd53a0 0x400330e5:0x3ffd53d0 0x400ac729:0x3ffd5400 0x400ac24d:0x3ffd5420 0x400a042a:0x3ffd5440 0x400a19d6:0x3ffd5480 0x400ab312:0x3ffd54a0 0x40098596:0x3ffd5510 0x40098cd1:0x3ffd5620 0x400973f3:0x3ffd5680 0x40033cf9:0x3ffd56a0 0x400320ac:0x3ffd56c0 0x4002c9c1:0x3ffd56f0

ELF file SHA256: d73d438bf07a0c2a

Rebooting...
```

### More Information.

The problem occurs if authmode of ESP32-AP is WIFI_AUTH_WPA2_PSK.
The problem doesn't occur if authmode of ESP32-AP is WIFI_AUTH_OPEN.

I tested with ESP32-S2-WROVER and ESP32-S3-WROOM-1-N8R8 and got similar results.
I tested with PSRAM enabled and disabled and got similar results.

The problem doesn't occur always.

boarchuz, 2022-10-22:
Same issue here.

Could it be significant that the heap is _always_ corrupted at exactly 0x300 bytes offset from the logged bss pointer a few lines prior?
eg. From @marclee5's log above (0x3ffdd4a0 - 0x3ffdd1a0):

```
I (56066) wifi:station: a0:10:81:c2:d9:36 leave, AID = 1, bss_flags is 658531, bss:0x3ffdd1a0
CORRUPT HEAP: Invalid data at 0x3ffdd4a0.
```

nishanth-radja, 2022-11-04:
@marclee5 can you pls mention the commit ID of the IDF you are using?

marclee5, 2022-11-04:
> @marclee5 can you pls mention the commit ID of the IDF you are using?

What is a "commit ID" and how can I get it?

nishanth-radja, 2022-11-04:
@marclee5 in the IDF folder, pls enter the command git log,and paste the first line of the output

marclee5, 2022-11-04:
> @marclee5 in the IDF folder, pls enter the command "git log",and paste the first line of the output

commit c32173907486c6fbbc506fc25df344a032ea4317 (HEAD -> release/v5.0, origin/release/v5.0)

gspatankar, 2022-11-07:
@marclee5 Can you please share the .elf file?

marclee5, 2022-11-09:
> @marclee5 Can you please share the .elf file?

#
[wifi_test.zip](https://github.com/espressif/esp-idf/files/9972358/wifi_test.zip)

marclee5, 2022-11-09:
> @marclee5 Can you please share the .elf file?

[wifi_test.zip](https://github.com/espressif/esp-idf/files/9973167/wifi_test.zip)

marclee5, 2022-11-24:
I described a similar problem here: [https://www.esp32.com/viewtopic.php?f=13&t=30747](https://www.esp32.com/viewtopic.php?f=13&t=30747)

AxelLin, 2022-12-18:
@gspatankar Any update? (Any finding in https://github.com/espressif/esp-idf/issues/9973#issuecomment-1309032831 ?)

nicksenva, 2023-02-10:
Hello @nachiketkukade, we noticed this issue has been marked as resolved, could you confirm if there will be a patch available soon?

nicksenva, 2023-02-15:
Looking through the commit history for esp_wifi, it appears a patch was applied in [this commit](https://github.com/espressif/esp-idf/commit/a24062d080a031135d92ff1da19a5d62e38b85b2) from @sarveshb14, and it seems to work for us after doing a checkout and some quick testing. I'm not sure what range of commits are necessary to perform a cherry pick, however. Could any dev provide a patch file or a cherry pick range for this issue? It is occurring quite frequently for us and we would like to implement a fix ASAP.

AxelLin, 2023-02-19:
> Hi @nicksenva
> 
> > could you confirm if there will be a patch available soon?
> 
> Yes. Work is in progress to make this fix available on 5.0 branch.

@sarveshb14 Could you confirm if v4.x branches need this fix as well?

sarveshb14, 2023-02-19:
Hi @AxelLin 
> Could you confirm if v4.x branches need this fix as well?

Issue is not there on v4.x branches.

nicksenva, 2023-03-09:
Hi @sarveshb14, any updates on when the patch will be made available?

nachiketkukade, 2023-03-13:
Fix will be available on v5.0 branch in couple of days

closed 2023-03-13 by nachiketkukade

nicksenva, 2023-03-27:
Is the patch available yet? If so, can you please provide the commit it was made in?

sarveshb14, 2023-03-28:
Yes. You can check [89bb920c](https://github.com/espressif/esp-idf/commit/89bb920c86d65fe2694c0d9b533b170d4a1c7dd2) available on release/v5.0 branch.