espressif/esp-idf #9978: esp_wifi_set_protocol() on ESP_IF_WIFI_STA also changes protocol on ESP_IF_WIFI_AP

kriegste, 2022-10-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4.2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### Development Kit.

TTGO

### Power Supply used.

USB

### What is the expected behavior?

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#_CPPv421esp_wifi_set_protocol16wifi_interface_t7uint8_t

It says

> Set protocol type of specified interface

as if the protocol is set for either the STA or the AP interface, i.e. **independently**.

### What is the actual behavior?

When I use esp_wifi_set_protocol() on the ESP_IF_WIFI_STA interface, it also changes the protocol for the ESP_IF_WIFI_AP interface.

I noticed this when my laptop/phone lost the connection to the ESP32 AP when I changed the protocol for the STA interface to 15 (BGN+LR). The AP disappeared from the scans as well. My guess is that the AP instantly switched to LR mode (which is incompatible to standard hardware of course).

I need to use BGN+LR for STA and BGN for AP. How do I do that?

### Steps to reproduce.

Minimum code example

```
#include "esp_log.h"
#include "esp_wifi.h"
#include "nvs_flash.h"

static const char *TAG = "main";

void app_main(void) {
	nvs_flash_init();
	esp_netif_init();
	esp_event_loop_create_default();

	esp_netif_create_default_wifi_sta();
	esp_netif_create_default_wifi_ap();

	wifi_init_config_t wifi_init_config = WIFI_INIT_CONFIG_DEFAULT();
	esp_wifi_init(&wifi_init_config);

	esp_wifi_start();

	uint8_t proto;

	esp_wifi_set_protocol(ESP_IF_WIFI_STA, 7);
	esp_wifi_get_protocol(ESP_IF_WIFI_STA, &proto);
	ESP_LOGI(TAG, "sta proto=%d (should be 7)", proto);

	esp_wifi_set_protocol(ESP_IF_WIFI_AP, 7);
	esp_wifi_get_protocol(ESP_IF_WIFI_AP, &proto);
	ESP_LOGI(TAG, "ap proto=%d (should be 7)", proto);

	esp_wifi_set_protocol(ESP_IF_WIFI_STA, 15);
	esp_wifi_get_protocol(ESP_IF_WIFI_STA, &proto);
	ESP_LOGI(TAG, "sta proto=%d (should be 15)", proto);

	esp_wifi_get_protocol(ESP_IF_WIFI_AP, &proto);
	ESP_LOGI(TAG, "ap proto=%d (should be 7)", proto);
}
```

### Debug Logs.

```plain
Output is

I 00:00:00.347 main: sta proto=7 (should be 7)
I 00:00:00.348 main: ap proto=7 (should be 7)
I 00:00:00.351 main: sta proto=15 (should be 15)
I 00:00:00.353 main: ap proto=15 (should be 7)
```

podaen, 2022-10-14:
Did you do the FTM? It's to fine tune IEEE 802.11-2016 standaard. I think it is related for your subject. It is about finding the right parameters, for that distance you want to retrieve. Those parameters contain package size(smaller) and bandwidth(lower) for longer distances.

zhangyanjiaoesp, 2022-10-17:
Currently, the STA and the softAP use the same LR bit, so when you set the LR mode, both STA and softAP will   take effect.

kriegste, 2022-10-17:
Also the API guide is written as if the LR modes for STA/AP can be set independently. There is even a table showing this.
https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-guides/wifi.html#lr-compatibility

So will this bug be fixed? Is there a workaround?

This is what breaks my application:
> For LR enabled in ESP32 AP, it’s incompatible with traditional 802.11 mode because the beacon is sent in LR mode.

I use the AP interface to provide the user the list with scan results to pick a WiFi (including those with LR protocol) to connect to.

zhangyanjiaoesp, 2022-10-18:
we will discuss it and reply to you.

kriegste, 2022-12-02:
Any news here?

zhangyanjiaoesp, 2022-12-05:
Sorry to tell you it will need a long time to support the STA(bgn+lr) and softAP(bgn) mode.

kriegste, 2022-12-13:
That is bad news. But it is nice to see that you are working on fixing this.

term-est, 2023-02-23:
What is a possible workaround for this issue for the time being?

Perhaps I can somehow tether over serial?

kriegste, 2023-04-01:
Any progress here?

zhangyanjiaoesp, 2024-03-05:
@kriegste I'm sorry that we have higher priority things to do now. As for the function that LR modes for STA/AP can be set independently, I will reply you as soon as there is any progress.

acrixl, 2024-03-10:
look like LR mode completely incompatible with BGN, update docs please

closed 2025-03-27 by espressif-bot

kriegste, 2025-04-19:
Cool. Will you backport this (to 5.3 or 5.4) or is it too big a change?

zhangyanjiaoesp, 2025-04-21:
> Cool. Will you backport this (to 5.3 or 5.4) or is it too big a change?

yes, it will backport to v5.4 ~ v5.1