espressif/esp-idf #9980: pip install -r

Gamecock, 2022-10-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

I expected to install dependencies.

Initial problem was corporate SSL inspection caused requirements not to install.
As part of troubleshooting I activated the virtual environment and tried `pip install requirements.txt`

I expected the requirements to install and then proceed to next problem.

### What is the actual behavior?

Collecting gdbgui==0.13.2.0
  Using cached gdbgui-0.13.2.0-py3-none-any.whl (878 kB)
Requirement already satisfied: pygdbmi<=0.9.0.2 in c:\users\finchmik\.espressif\python_env\idf4.4_py3.10_env\lib\site-packages 
(from -r requirements.txt (line 25)) (0.9.0.2)
Collecting python-socketio<5
  Using cached python_socketio-4.6.1-py2.py3-none-any.whl (51 kB)
Requirement already satisfied: jinja2<3.1 in c:\users\finchmik\.espressif\python_env\idf4.4_py3.10_env\lib\site-packages (from 
-r requirements.txt (line 29)) (3.0.1)
Requirement already satisfied: itsdangerous<2.1 in c:\users\finchmik\.espressif\python_env\idf4.4_py3.10_env\lib\site-packages 
(from -r requirements.txt (line 30)) (2.0.1)
Collecting kconfiglib==13.7.1
  Using cached kconfiglib-13.7.1-py2.py3-none-any.whl (145 kB)
Collecting reedsolo<=1.5.4,>=1.5.3
  Using cached reedsolo-1.5.4-py3-none-any.whl
Collecting bitstring>=3.1.6
  Using cached bitstring-3.1.9-py3-none-any.whl (38 kB)
Collecting ecdsa>=0.16.0
  Using cached ecdsa-0.18.0-py2.py3-none-any.whl (142 kB)
Collecting construct==2.10.54
  Using cached construct-2.10.54-py3-none-any.whl
Collecting gevent<2.0,>=1.2.2
  Using cached gevent-1.5.0.tar.gz (5.3 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting Flask<1.0,>=0.12.2
  Using cached Flask-0.12.5-py2.py3-none-any.whl (81 kB)
Requirement already satisfied: Pygments<3.0,>=2.2.0 in c:\users\finchmik\.espressif\python_env\idf4.4_py3.10_env\lib\site-packages (from gdbgui==0.13.2.0->-r requirements.txt (line 22)) (2.10.0)
Requirement already satisfied: Flask-Compress<2.0,>=1.4.0 in c:\users\finchmik\.espressif\python_env\idf4.4_py3.10_env\lib\site-packages (from gdbgui==0.13.2.0->-r requirements.txt (line 22)) (1.10.1)
Collecting Flask-SocketIO<3.0,>=2.9
  Using cached Flask_SocketIO-2.9.6-py2.py3-none-any.whl (16 kB)
Requirement already satisfied: colorama in c:\users\finchmik\.espressif\python_env\idf4.4_py3.10_env\lib\site-packages (from click>=7.0->-r requirements.txt (line 8)) (0.4.5)
Requirement already satisfied: cffi>=1.12 in c:\users\finchmik\.espressif\python_env\idf4.4_py3.10_env\lib\site-packages (from 
cryptography>=2.1.4->-r requirements.txt (line 12)) (1.15.1)
Requirement already satisfied: requests<3 in c:\users\finchmik\.espressif\python_env\idf4.4_py3.10_env\lib\site-packages (from 
idf-component-manager~=1.0->-r requirements.txt (line 20)) (2.28.1)
Requirement already satisfied: six in c:\users\finchmik\.espressif\python_env\idf4.4_py3.10_env\lib\site-packages (from idf-component-manager~=1.0->-r requirements.txt (line 20)) (1.16.0)
[... 321 lines cut ...]
Microsoft (R) C/C++ Optimizing Compiler Version 19.29.30146 for x86
Copyright (C) Microsoft Corporation.  All rights reserved.

usage: cl [ option... ] filename... [ /link linkoption... ]
(idf4.4_py3.10_env)``` 

After commenting out gdbgui there is annother version conflict which I have not traced down yet.

```ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
flask-socketio 5.1.1 requires python-socketio>=5.0.2, but you have python-socketio 4.6.1 which is incompatible.
Successfully installed UNKNOWN-0.0.0 python-engineio-3.14.2 python-socketio-4.6.1
(idf4.4_py3.10_env) '''

### Steps to reproduce.

1. Install behind firewall
2. Activate virtual env
3. `pip install -r requirements.txt`
...

### Build or installation Logs.

```plain
Where do I get build log from?
```

dobairoland, 2022-10-15:
Please use the Windows tool installer. This is addressed there. You are not supposed to run these commands by yourself.

dobairoland, 2022-10-15:
https://dl.espressif.com/dl/esp-idf

closed 2022-10-24 by dobairoland

yoyojacky, 2024-03-20:
I have met this error on my Raspberry Pi 4B, 8GB
ESP-IDF： v4.4.0 
when i execute the ./install.sh ，this error will pop up. How can I solve this problem?
```
pi@RPi8GA01:~/esp-idf $ ./install.sh
Detecting the Python interpreter
Checking "python" ...
Python 3.11.2
"python" has been detected
Installing ESP-IDF tools
Selected targets are: esp32s3, esp32, esp32c3, esp32s2
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, xtensa-esp32s3-elf, riscv32-esp-elf, openocd-esp32
Skipping xtensa-esp32-elf@esp-2021r2-8.4.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2021r2-8.4.0 (already installed)
Skipping xtensa-esp32s3-elf@esp-2021r2-8.4.0 (already installed)
Skipping riscv32-esp-elf@esp-2021r2-8.4.0 (already installed)
Skipping openocd-esp32@v0.11.0-esp32-20211220 (already installed)
Installing Python environment and packages
Python 3.11.2
pip 24.0 from /home/pi/.espressif/python_env/idf4.4_py3.11_env/lib/python3.11/site-packages/pip (python 3.11)
Installing Python packages from /home/pi/esp-idf/requirements.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple, https://dl.espressif.com/pypi
Requirement already satisfied: setuptools>=21 in /home/pi/.espressif/python_env/idf4.4_py3.11_env/lib/python3.11/site-packages (from -r /home/pi/esp-idf/requirements.txt (line 4)) (66.1.1)
Collecting click>=7.0 (from -r /home/pi/esp-idf/requirements.txt (line 8))
  Using cached https://dl.espressif.cn/pypi/click/click-8.1.7-py3-none-any.whl (97 kB)
Collecting pyserial>=3.3 (from -r /home/pi/esp-idf/requirements.txt (line 9))
  Using cached https://dl.espressif.cn/pypi/pyserial/pyserial-3.5-py2.py3-none-any.whl (90 kB)
Collecting future>=0.15.2 (from -r /home/pi/esp-idf/requirements.txt (line 10))
  Using cached https://dl.espressif.cn/pypi/future/future-1.0.0-py3-none-any.whl (491 kB)
Collecting cryptography>=2.1.4 (from -r /home/pi/esp-idf/requirements.txt (line 12))
  Using cached https://dl.espressif.cn/pypi/cryptography/cryptography-42.0.5-cp39-abi3-manylinux_2_28_aarch64.whl (4.4 MB)
Collecting pyparsing<2.4.0,>=2.0.3 (from -r /home/pi/esp-idf/requirements.txt (line 18))
  Using cached https://dl.espressif.cn/pypi/pyparsing/pyparsing-2.3.1-py2.py3-none-any.whl (61 kB)
Collecting pyelftools>=0.22 (from -r /home/pi/esp-idf/requirements.txt (line 19))
  Using cached https://dl.espressif.cn/pypi/pyelftools/pyelftools-0.31-py3-none-any.whl (180 kB)
Collecting idf-component-manager>=0.2.99-beta (from -r /home/pi/esp-idf/requirements.txt (line 20))
  Using cached https://dl.espressif.cn/pypi/idf-component-manager/idf_component_manager-1.5.2-py2.py3-none-any.whl (160 kB)
Collecting gdbgui==0.13.2.0 (from -r /home/pi/esp-idf/requirements.txt (line 22))
  Using cached https://dl.espressif.cn/pypi/gdbgui/gdbgui-0.13.2.0-py3-none-any.whl (878 kB)
Collecting pygdbmi<=0.9.0.2 (from -r /home/pi/esp-idf/requirements.txt (line 25))
  Using cached https://dl.espressif.cn/pypi/pygdbmi/pygdbmi-0.9.0.2-py3-none-any.whl (16 kB)
Collecting python-socketio<5 (from -r /home/pi/esp-idf/requirements.txt (line 28))
  Using cached https://dl.espressif.cn/pypi/python-socketio/python_socketio-4.6.1-py2.py3-none-any.whl (51 kB)
Collecting kconfiglib==13.7.1 (from -r /home/pi/esp-idf/requirements.txt (line 30))
  Using cached https://dl.espressif.cn/pypi/kconfiglib/kconfiglib-13.7.1-py2.py3-none-any.whl (145 kB)
Collecting reedsolo<=1.5.4,>=1.5.3 (from -r /home/pi/esp-idf/requirements.txt (line 33))
  Using cached https://dl.espressif.cn/pypi/reedsolo/reedsolo-1.5.4-py3-none-any.whl (28 kB)
Collecting bitstring>=3.1.6 (from -r /home/pi/esp-idf/requirements.txt (line 34))
  Using cached https://dl.espressif.cn/pypi/bitstring/bitstring-4.1.4-py3-none-any.whl (60 kB)
Collecting ecdsa>=0.16.0 (from -r /home/pi/esp-idf/requirements.txt (line 35))
  Using cached https://dl.espressif.cn/pypi/ecdsa/ecdsa-0.18.0-py2.py3-none-any.whl (142 kB)
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/construct/
Collecting construct==2.10.54 (from -r /home/pi/esp-idf/requirements.txt (line 39))
  Using cached https://dl.espressif.cn/pypi/construct/construct-2.10.54-py3-none-any.whl (57 kB)
Collecting Flask<1.0,>=0.12.2 (from gdbgui==0.13.2.0->-r /home/pi/esp-idf/requirements.txt (line 22))
  Using cached https://dl.espressif.cn/pypi/flask/Flask-0.12.5-py2.py3-none-any.whl (81 kB)
Collecting Flask-Compress<2.0,>=1.4.0 (from gdbgui==0.13.2.0->-r /home/pi/esp-idf/requirements.txt (line 22))
  Using cached https://dl.espressif.cn/pypi/flask-compress/Flask_Compress-1.14-py3-none-any.whl (8.4 kB)
Collecting Flask-SocketIO<3.0,>=2.9 (from gdbgui==0.13.2.0->-r /home/pi/esp-idf/requirements.txt (line 22))
  Using cached https://dl.espressif.cn/pypi/flask-socketio/Flask_SocketIO-2.9.6-py2.py3-none-any.whl (16 kB)
Collecting gevent<2.0,>=1.2.2 (from gdbgui==0.13.2.0->-r /home/pi/esp-idf/requirements.txt (line 22))
  Using cached gevent-1.5.0.tar.gz (5.3 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error

  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [319 lines of output]
      Compiling src/gevent/resolver/cares.pyx because it changed.
[... 316 lines cut ...]
          raise CompileError(None, pyx_file)
[... cut ...]
    subprocess.check_call(run_args, stdout=sys.stdout, stderr=sys.stderr, env=env_copy)
  File "/usr/lib/python3.11/subprocess.py", line 413, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/home/pi/.espressif/python_env/idf4.4_py3.11_env/bin/python', '-m', 'pip', 'install', '--no-warn-script-location', '-r', '/home/pi/esp-idf/requirements.txt', '--extra-index-url', 'https://dl.espressif.com/pypi']' returned non-zero exit status 1.
pi@RPi8GA01:~/esp-idf $

```

dobairoland, 2024-03-20:
> v4.4.0

You need fixes from later bugfix releases. It is recommended to always use the latest bugfix release which is v4.4.7 at this moment. Python 3.11 did not exist at the time v4.4.0 was released.

yoyojacky, 2024-03-21:
> > v4.4.0
> 
> You need fixes from later bugfix releases. It is recommended to always use the latest bugfix release which is v4.4.7 at this moment. Python 3.11 did not exist at the time v4.4.0 was released.

OK，thanks bro, i will try to use the latest one and try again..