espressif/esp-idf #9982: idf.py command returns "env: python: No such file or directory"

khray, 2022-10-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

**Intro:** This isn't the first time I have had this issue, but its becoming a bit frustrating. I was building without a problem until I shut my computer down. After booting it back up, I have been running into this error consistently. 

**Background:** My current environment is: Mac intel 2019 16", it's currently running on Monterey, and I am using the ESP32 sparkfun board.

**Issue:** When I run the command "idf.py", "idf.py menuconfig", or "idf.py build", pretty much anything that starts with the idf.py command I receive the following error.

**env: python: No such file or directory**

Since running into this error, I have searched for a solution. I have looked at similar solutions with that recreates this error, however the solutions that were listed did not work. I believe the issue is related to my PATH or environment variable both of which have me confused. 

**Action:** Please let me know if anyone has any feedback or a potential solution. In addition, I'd like to learn about environment variables and PATHS so any relative information would be helpful. Thank you in advance.

dobairoland, 2022-10-15:
You are not supposed to change the PATH. 

Please follow the guide from here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-macos-setup.html

Running `. export.sh` in each terminal will take care of setting up PATH.

khray, 2022-10-16:
Thank you!! Following these steps resolved the issue, however, when I log off, restart, or shut my computer down I receive the same error. 

What do I have to do in order to make these changes permanent?

dobairoland, 2022-10-17:
The only thing you need to repeat from those steps is to use `. export.sh` in every shell you want to use ESP-IDF.

I don't recommend to make this "permanent". This is done so the ESP-IDF environment would be separated from the rest of your system. If you permanently join them then they one part of your system could broke the other one.

closed 2022-10-17 by dobairoland