espressif/esp-idf #9985: Methods to disable internet access but remain intranet

lawrenceko-intecular, 2022-10-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,

We are trying to disable internet access on ESP32 but remain the local network connection (including mDNS). Is the solution to change the DNS settings?

Thank you.

gabsuren, 2022-10-21:
Hello,

Can you please provide more details?
You want to disable network access for income and outcome traffic?

Regards,
Suren

law-ko, 2022-10-22:
Hi @gabsuren ,

That is correct, so intranet access only. For example, pinging `google.com` will not work but pinging `myserver.local` or `192.168.1.1` will work.

Thank you

espressif-abhikroy, 2022-11-03:
Hi, 
I am assuming you are using the esp32 as an access point, and you want to disable internet access for its clients while allowing the esp32 AP itself to be able to access the internet.
One way to achieve this is by disabling “IP forwarding”. 
```idf.py menuconfig```
Component config → LWIP -> Enable IP forwarding
and disable it.

twlawrenceko, 2022-11-23:
> Hi,
> I am assuming you are using the esp32 as an access point, and you want to disable internet access for its clients while allowing the esp32 AP itself to be able to access the internet.
> One way to [...]

Hi @espressif-abhikroy , this is not quite what I mean. I mean by when esp32 is on STA mode, how to set the ESP32 for it to not have internet access but intranet access.

For example: Ping `google.com` or `8.8.8.8` will fail; where ping `192.168.2.1` will success

Thank you

law-ko, 2022-11-23:
> Hi,
> I am assuming you are using the esp32 as an access point, and you want to disable internet access for its clients while allowing the esp32 AP itself to be able to access the internet.
> One way to [...]

Hi @espressif-abhikroy , this is not quite what I mean. I mean by when esp32 is on STA mode, how to set the ESP32 for it to not have internet access but intranet access.

For example: Ping `google.com` or `8.8.8.8` will fail; where ping `192.168.2.1` will success

Thank you

espressif-abhikroy, 2023-03-22:
Hi,
Please attempt the following solution: 
Incorporate this function into your project and invoke it after the network setup is complete.
``` C
void block_internet(void)
{     
    esp_netif_t *esp_netif;
    esp_netif_config_teth_cfg = ESP_NETIF_DEFAULT_ETH();
    esp_netif = esp_netif_new(&eth_cfg)
    esp_netif_set_default_netif(esp_netif);
    esp_netif_dhcps_stop(esp_netif); 
}
``` 
Explanation:
This function creates a dummy interface and designates it as the default interface. 
Consequently, any traffic intended for outside the network is routed to this default interface and is consequently dropped. 
However, traffic destined for an IP within the LAN functions properly since it is not routed to the default interface.

law-ko, 2023-03-31:
Hello @espressif-abhikroy ,

Thanks for the answer, is this only possible on ESP IDF v5.0 and above? We are currently on v4.4

Also, what are the commands to unblock the internet after calling `block_internet()`?

Thank you

espressif-abhikroy, 2023-03-31:
This function should also work on v4.4.
To restore access to the internet, you need to reset the default interface to its original setting by using ```esp_netif_set_default_netif(esp_netif);```.
You can use ```esp_netif_get_default_netif()``` to get the default interface and store it before calling ```block_internet()```.
Although ```esp_netif_get_default_netif()``` may not be available on v4.4.
In that case, you'll need to identify the interface that's connected to the internet and set that as the default interface.

law-ko, 2023-03-31:
> This function should also work on v4.4.
> To restore access to the internet, you need to reset the default interface to its original setting by using `esp_netif_set_default_netif(esp_netif);`.
> You can [...]

@espressif-abhikroy Yes, `esp_netif_set_default_netif(esp_netif);` is not available on v4.4. What are the alternative commands to handle this on v4.4?

espressif-abhikroy, 2023-04-04:
Hi @law-ko,
Please try the following solution:
```c
esp_netif_t* block_internet(esp_netif_t *esp_netif)
{
    esp_netif_t *esp_netif_block = esp_netif;

    if (esp_netif_block == NULL) {
        esp_netif_config_t eth_cfg = ESP_NETIF_DEFAULT_ETH();

        esp_netif_inherent_config_t _l_esp_netif_inherent_eth_config = { 
            .flags = (esp_netif_flags_t)(ESP_NETIF_FLAG_IS_BRIDGE),
            .if_key = "ETH_BLK",
            .if_desc = "eth404",
            .route_prio = 150,
        };

        eth_cfg.base = &_l_esp_netif_inherent_eth_config;
        esp_netif_block = esp_netif_new(&eth_cfg);
    }

    esp_netif_action_start(esp_netif_block, 0, 0, 0);
    esp_netif_action_connected(esp_netif_block, 0, 0, 0);

    return esp_netif_block;
}

void unblock_internet(esp_netif_t *esp_netif)
{
    esp_netif_action_stop(esp_netif, 0, 0, 0);
}
```

The usage will be as follows:
```c
esp_netif_t *esp_netif_block = block_internet(NULL);
run_ping();

unblock_internet(esp_netif_block);
run_ping();

block_internet(esp_netif_block);
run_ping();
```

This solution should work for both v4.4 and v5.0.

Alvin1Zhang, 2023-06-14:
Thanks for reporting, will close due to short of feedback, feel free to reopen. Thanks.

closed 2023-06-14 by Alvin1Zhang