espressif/esp-idf #9986: [Console] Proposal: Console should be refactored for more flexibility

chipweinberger, 2022-10-15:
### Is your feature request related to a problem?

 # Console Refactor & Use Case #

Hi, I wanted to share my vision for the Console. These are pretty big changes, so it's not worth starting a PR without discussion first. Let me know your thoughts!

**Necessary Features**
- Support all outputs simultaneously
- Support multiple Console Inputs simultaneously (ideal)
- Support enabling & disabling inputs & outputs dynamically

I have slowly working towards this with with a couple PRs so far:

**PRs:**
- fix init/deinit: https://github.com/espressif/esp-idf/pull/9983
- add more secondary outputs: https://github.com/espressif/esp-idf/pull/9979

However, it is still not flexible enough for my use case.

**Use case:** My product has a _sometimes occupied_ USB port, and an _always available but physically internal_ UART port. I want to be able to do everything from either port, at any time for maximum convenience. 

 ## Necessary Changes ##

### KConfig: ###

✅ == my use case

> **CONSOLE INPUT (STDIN):**
> (choice) UART0 ✅  
> (choice) UART1
> (choice) USB CDC
> (choice) USB Serial JTAG 
> 
> **CONSOLE OUTPUT (STDOUT):**
> (bool) UART0 ✅
> (bool) UART1
> (bool) USB CDC
> (bool) USB Serial JTAG ✅

 ### ESP-IDF: ###
- **kConfig** - needs booleans for console output
- **panic.c** - needs to print to all console outputs, not just the primary one
- **linenoise.c** - should support multiple instances, each with their own  FDs

# Multiple Console Inputs #

But how do we support input from either the **USB Serial JTAG** *or* the **UART port** simultaneously?

### Option 1: Multiple Threads / Multiple linenoise ### 

The simplest approach is to instantiate multiple linenoise's, each with their own repl thread. 

e.g.

```
void start_consoles() {
     xTaskCreate(usb_serial_jtag_console_loop...)
     xTaskCreate(uart0_console_loop...)
}

void usb_serial_jtag_console_loop() {
   linenoise_handle_t lnHandle = linenoiseCreateWithInput("/dev/usbserjtag");

    while(!should_stop){
        char* line = linenoise(lnHandle)
    }
}

void uart0_console_loop() {
   linenoise_handle_t lnHandle = linenoiseCreateWithInput("/dev/uart0");

    while(!should_stop){
        char* line = linenoise(lnHandle)
    }
}
```

### Option 2: Muxing STDIN ### 

This may or may not be the best idea, but it would be sufficient in most use cases. 

Combine input from all console ports into a single stdin stream.

**KConfig:**

> (bool) Mux Multiple Inputs into STDIN ✅

If selected, the "input choice" would become "bools", like this:

> **CONSOLE INPUT (STDIN):**
> (bool) UART0 ✅  
> (bool) UART1 
> (bool) USB CDC
> (bool) USB Serial JTAG ✅

**How:**
- `vfs_console.h` would expose a single "RX Available" semaphore
-  the RX ISR's would "give" to that semaphore whenever input becomes available. 
- `console_read()` would start by taking that semaphore, then sum how many bytes are available on each input
- if not enough bytes are available, it will take the semaphore again
- **Advantage**: users only need to use a single console thread 
- **Disadvantage**: if multiple inputs are actually plugged in at the same time, linenoiseProbe() should return an error. Instead, users should probe a specific path, i.e. "/dev/uart0/", and use that as the default.

chipweinberger, 2022-10-17:
Or perhaps it would be better to keep the Primary and Secondary theme.

We would support up to , say, 3 consoles ports, which should hopefully be enough for anybody.

Something like this:

> 
> PRIMARY CONSOLE:
> (choice) NONE
> (choice) UART0
> (choice) UART1
> (choice) USB CDC
> (choice) USB Serial JTAG
> 
> (bool) Use Custom UART Settings
> depends on PRIMARY CONSOLE UART0/UART1
> 
> SECONDARY CONSOLE:
> (choice) NONE
> (choice) UART0
> (choice) UART1
> (choice) USB CDC
> (choice) USB Serial JTAG
> 
> (bool) Support Secondary Console Input (stdin)
> 
> (bool) Use Custom UART Settings
> depends on SECONDARY CONSOLE UART0/UART1
> 
> TERTIARY CONSOLE:
> (choice) NONE
> (choice) UART0
> (choice) UART1
> (choice) USB CDC
> (choice) USB Serial JTAG
> 
> (bool) Use Custom UART Settings
> depends on TERTIARY CONSOLE UART0/UART1
> 
> (bool) Support Tertiary Console Input (stdin)

chipweinberger, 2022-10-18:
I have basic "runtime console choice" working in my project, using a GPIO to determine which console to use after boot. 

The minimal configuration for me is:

- kconfig: primary: UART, secondary: USB Serial/JTAG
- linenoise 2 https://github.com/espressif/esp-idf/pull/9998 to swap stdin/stdout for /dev/usbserjtag after boot)
- `esp_vfs_usb_serial_jtag_use_driver_for_rx()` https://github.com/espressif/esp-idf/pull/9983 to prevent the app from hanging after the terminal is unplugged

chipweinberger, 2022-10-19:
I have my basic use case running now. (i.e. "bootime console choice"), using this kConfig and my 3 PRs.

> **kconfig:** 
> primary: UART
> secondary: USB Serial/JTAG

With my basic use case is now running, I wanted to comment on the proposed features again:

## Proposed Features

### Support all outputs simultaneously 

I've mostly covered with with https://github.com/espressif/esp-idf/pull/9979. 
It would be great if enabling all outputs was trivial, but it is kind of an annoying code change.
Having good 1st, 2nd, (3rd...), etc consoles, should be good enough for most people.

### Support multiple Console Inputs simultaneously

In my code, I went with **Option 1** (multiple threads). The required changes were:

https://github.com/espressif/esp-idf/pull/9998 : linenoise2
in order to choose the vfs to use for input. i.e. fopen("/dev/usbserjtag")

https://github.com/espressif/esp-idf/pull/9983  : `esp_vfs_usb_serial_jtag_use_driver_for_rx()` 
to prevent the app from hanging after the terminal is unplugged

It would be great if **Option 2** was made a kConfig option for the user, IMO.

### Support enabling & disabling inputs & outputs dynamically

I chose not to support this in my code. Instead, I have both outputs always enabled. For input, I choose the linenoise2 input based on if a USB Terminal is detected at boot time. The required changes were:

https://github.com/espressif/esp-idf/pull/9998 : linenoise2
in order to choose the vfs to use for input. i.e. fopen("/dev/usbserjtag")

https://github.com/espressif/esp-idf/pull/9983  : `esp_vfs_usb_serial_jtag_use_driver_for_rx()` 
to prevent the app from hanging after the terminal is unplugged

chipweinberger, 2022-11-02:
@igrr Not sure if you saw this, but it helps give context into what I have been working on.

In short, I'd like `USB Serial/JTAG'  w/ Console to be usable by default, without needing to disable UART Console.

chipweinberger, 2023-06-09:
I will close this issue when 

https://github.com/espressif/esp-idf/pull/10526 & https://github.com/espressif/esp-idf/pull/10132 

land

closed 2024-12-17 by chipweinberger

SoucheSouche, 2025-06-24:
hi @chipweinberger,

I am working on refactoring the existing console component in ESP-IDF at the moment.

The new design will handle multiple instances that should allow the user to use all available IO simultaneously. It will allow to use other IOs than the one already provided (UART, USB Serial JTAG and USB CDC).

The concept of "console" component will disappear and will be replaced by a more modular approach consisting of a series of standalone components available from the [component registry](https://components.espressif.com/). Mainly, you will be able to find:
- an updated version of linenoise supporting instantiation.
- a component handling command registration and execution
- a component providing the REPL implementation 

I see that you closed this ticket a while ago but since you contributed to the component with your ideas, reported bugs and fixes, I wanted to let you know that the work continues :)

chipweinberger, 2025-06-24:
I appreciate you updating me on this! Cool to see work is being done on it!

The idea of allowing any IO is a great feature!